m-labs/artiq #847: Sayma won't boot from flash

sbourdeauducq, 2017-11-03:
Bitstream is not loaded after ``artiq_flash.py -t sayma proxy gateware start`` or after a power-cycle.
@gkasprow Are there switches or anything that need to be set on the boards you sent us?

jordens, 2017-11-03:
M[2:0] settings?

sbourdeauducq, 2017-11-23:
According to @gkasprow it boots when the bitstream is generated and flashed with Vivado.

gkasprow, 2017-11-23:
Yes, switches need to be set to boot from FLASH. I tested all boards loading via JTGAG, some of them had FLASH tested as well. Without automated test-bench it's hard to test every functionality of every board. Creotech guys are working on it, Sayma AMC part is already done.

jbqubit, 2017-11-24:
What is Vivado doing that ARTIQ is not? @gkasprow please spell out exactly what the switches are that need to be set for booting from FLASH. Have you seen booting-from-flash failure that @sbourdeauducq reports?

gkasprow, 2017-11-24:
The switch position does not matter if you load both fpgas via jtag.

gkasprow, 2017-11-24:
So far the FLASH configures only AMC side.

jbqubit, 2017-11-24:
ARTIQ loading: 1. flash FPGA proxy .bit using JTAG, 2. send flash data to FPGA which writes to flash, 3. FPGA resets and expects to load .bit from flash. Please detail what the switch configuration needs to be for step 3.

jordens, 2017-11-30:
@gkasprow could you give me the `bitgen` options you used to test loading the AMC bitstream from flash?

gkasprow, 2017-11-30:
@jordens is it defined in some text file?

jordens, 2017-11-30:
I forgot that `bitgen` is ISE.
If you can give me your Vivado sources for that test project or a screenshot of the bitstream properties/configuration dialog, that would be good. Things like frequency, SPI width...

gkasprow, 2017-11-30:
hope this helps
the project is [here](https://www.dropbox.com/sh/b5gnf815xk6mzds/AABjPOywleaJw29mKkxbrfZCa?dl=0):

[image]
[image]

[image]

[image]

[image]

[image]

[image]

[image]

jbqubit, 2017-12-11:
@jordens What's the status of this?

sbourdeauducq, 2017-12-13:
> M[2:0] settings?

They are all at 0, which is incorrect, on the HKG boards.

sbourdeauducq, 2017-12-13:
Er, no all at 1 (but still incorrect) the switches are enabling pull-downs.
This should be the correct position:
[image]

sbourdeauducq, 2017-12-13:
After setting the switches into that position, the board boots from the bitstream that @jordens flashed, but not the one I built from master and flashed with artiq_flash.

jordens, 2017-12-13:
2917208d

closed 2017-12-13 by jordens

jbqubit, 2017-12-13:
Confirmed that I can flash using ```artiq_flash -t sayma```.  Thank you @jordens !