m-labs/artiq #848: Label masters with a name to distinguish dashboards

cjbe, 2017-11-03:
It would be useful for Artiq masters to have a configurable name, and have the dashboards show the logical name of the master they are connected to, rather than just an IP address. 

At the moment is it rather confusing when connecting to multiple masters with similar addresses. This is particularly necessary if one is wants to run multiple masters on one host (as we are planning).

jordens, 2017-11-04:
Using the hostname (instead of the IP address) should work just fine.

sbourdeauducq, 2017-11-04:
He wants to run multiple masters one one host. You can have multiple hostnames mapped to one IP, and different ports open for each master, but then there can be confusion if you connect with an unplanned hostname/port combination.

I suppose the dashboard should also store one set of configuration data per master.

cjbe, 2017-12-03:
Implemented in PR #855

closed 2017-12-03 by cjbe