m-labs/artiq #849: Core device panics (infrequently) on loading kernel function

cjbe, 2017-11-14:
Using Artiq 3 on a KC705, connected to a 100 MB/s router, the following causes the core device to panic roughly 1 every 1e4 times. I have also seen similar panics for kernel functions that are running continuously, but calling the master via RPCs. I guess this may be related to #837 

Experiment pseudocode:
```python
def run(self):
  data = 0xff
  while True:
    self.krun(data)

@kernel
    def krun(self, data):
        self.core.break_realtime()
        self.spi.write(data<<24)
```

Error in core log (the indices are different every panic):
```
panic at src/libcore/slice/mod.rs:741: slice index starts at 20 but ends at 0
halting.
```

sbourdeauducq, 2017-11-14:
Can we get the device to print backtraces on panic?

whitequark, 2017-11-14:
Yes. But this is likely not necessary in this particular case.

I don't think #837 is relevant here, the TCP stack must not panic no matter what sort of garbage it is fed with. I fuzz it regularly to achieve that.

sbourdeauducq, 2017-11-14:
How do you know this is in the TCP stack?

whitequark, 2017-11-14:
I don't *know*, but that's where almost all of the slicing lives; most of the rest is in the RPC code and we haven't touched that.

whitequark, 2017-12-06:
@sbourdeauducq I have thought about it and making the core device print backtraces is a good idea and is easy, because that code is already in libstd (and we only need to tweak it slightly for our embedded environment). This will however bloat the runtime even further (which I personally don't consider a problem whatsoever).

sbourdeauducq, 2017-12-06:
Sure go ahead, we have 1+GB RAM. As long as it can be easily disabled for running on the block-RAM of the RTM FPGA...

whitequark, 2017-12-08:
How much BRAM do you have there and what do you intend to fit into it?

sbourdeauducq, 2017-12-08:
Not much, it's a XC7A15T. Just satman (drive si5324, process drtio aux packets), no smoltcp or anything particularly big or complex.

whitequark, 2017-12-27:
No panics with the following experiment at current master over 20000 runs:
```python
from artiq.experiment import *

class TestExperiment(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("spi0")

    def run(self):
        data = 0xff
        runs = 0
        while True:
            self.krun(data)
            runs += 1
            if runs % 100 == 0:
                print(runs, " runs")

    @kernel
    def krun(self, data):
        self.core.break_realtime()
        self.spi0.write(data << 24)
```

jordens, 2018-01-05:
@whitequark what are you waiting for? more data?

@cjbe it might be useful to check whether this still occurs after all the smoltcp changes.

whitequark, 2018-01-05:
@jordens A way to reproduce and/or a backtrace.

cjbe, 2018-01-05:
@jordens @whitequark I did try to reproduce this with the master branch as of a few days ago, and I could not. 

I am going to upgrade a running experiment to a current version of Artiq next week - I will report if I see this again.

whitequark, 2018-01-13:
@cjbe Did you see anything?

cjbe, 2018-01-13:
@whitequark I have not been able to reproduce this problem with the current version of Artiq.

closed 2018-01-13 by whitequark