m-labs/artiq #850: Log messages not printed in scheduled experiment

philipkent, 2017-11-16:
We're using python logging to print info, warning, debug, etc. messages.  If we use the scheduler to submit an experiment that uses python logging the messages are not printed.  Is there a way to specify the log level when submitting an experiment through the scheduler?  Seems like that's what's needed but there is no argument for log level for the scheduler.submit() method.

sbourdeauducq, 2017-11-17:
It's not built into the scheduler, but you can implement that yourself in the experiment using ARTIQ's [EnumerationValue](http://m-labs.hk/artiq/manual-release-3/core_language_reference.html#artiq.language.environment.EnumerationValue) and Python's [Logger.setLevel](https://docs.python.org/3/library/logging.html#logging.Logger.setLevel).

sbourdeauducq, 2017-11-17:
The log level is in ``expid["log_level"]``. It expects an integer with the same conventions as the Python logging module, e.g. ``logging.WARNING``.

sbourdeauducq, 2017-11-17:
And the current ``expid``, if you need that, is ``scheduler.expid``.

sbourdeauducq, 2017-11-29:
@philipkent Does this solve your problem?

philipkent, 2017-11-29:
Thanks, I will try that.

philipkent, 2017-11-29:
Using scheduler.expid['log_level'] worked.  Thanks.

closed 2017-11-29 by philipkent