m-labs/artiq #851: searching in log messages

sbourdeauducq, 2017-11-17:
Currently, the filters only applies to the log messages as they arrive. There is currently no way in the dashboard to search for a previously received log message by keyword (and, with lesser importance, by priority). I see two ways of doing this:
1. also applying the filters to the log messages already in the panel's buffer.
2. adding another search feature.

I'm more leaning towards option 1.

jbqubit, 2017-11-17:
I like (1). It's helpful when focusing on a specific logged event type to see both its history and new events of that type as they happen.

dnadlinger, 2019-02-17:
cc @ljstephenson

sbourdeauducq, 2022-08-10:
Should the incoming message filtering be removed?

PR #1950 (2022-08-29): GUI log: Apply level and text filter to the log messages in panel

# ARTIQ Pull Request

## Description of Changes
Apply level and text filter to the log messages in panel
Incoming message filtering is removed

### Related Issue #851 

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :sparkles: New feature |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [ ] Update [RELEASE_NOTES.rst](../RELEASE_NOTES.rst) if there are noteworthy changes, especially if there are changes to existing APIs [... cut]
closed 2022-08-29 by sbourdeauducq