m-labs/artiq #852: flashing Sayma_AMC

jbqubit, 2017-11-21:
I've built .bit for sayma_amc (without sawg). But artiq_flash is having trouble with the jtag chain. 

```
$ python3 -m artiq.gateware.targets.sayma_rtm
$ python3 -m artiq.gateware.targets.sayma_amc_standalone
$ mkdir binaries
$ wget https://raw.githubusercontent.com/jordens/bscan_spi_bitstreams/master/bscan_spi_xcku040-sayma.bit binaries
$ cp misoc_saymaamcstandalone_sayma_amc/gateware/top.bit binaries
$ cp binaries/top.bit binaries/top.bin   
$ cp misoc_saymaamcstandalone_sayma_amc/bios/bios.bin binaries
$ cp misoc_saymaamcstandalone_sayma_amc/software/bios/bios.bin binaries
$  cp misoc_saymaamcstandalone_sayma_amc/software/bios/bios.elf binaries
$  cp misoc_saymaamcstandalone_sayma_amc/software/runtime/runtime.bin binaries
$  cp misoc_saymaamcstandalone_sayma_amc/software/runtime/runtime.elf binaries
```

```
(artiq-dev) britton@britton1:~/a4t2/artiq$ python3 -m serial.tools.list_ports -v
/dev/ttyS0          
    desc: ttyS0
    hwid: PNP0501
/dev/ttyUSB0        
    desc: CP2103 USB to UART Bridge Controller
    hwid: USB VID:PID=10C4:EA60 SER=0001 LOCATION=1-4.3
/dev/ttyUSB1        
    desc: Quad RS232-HS
    hwid: USB VID:PID=0403:6011 LOCATION=1-2:1.0
/dev/ttyUSB2        
    desc: Quad RS232-HS
    hwid: USB VID:PID=0403:6011 LOCATION=1-2:1.1
/dev/ttyUSB3        
    desc: Quad RS232-HS
    hwid: USB VID:PID=0403:6011 LOCATION=1-2:1.2
/dev/ttyUSB4        
    desc: Quad RS232-HS
    hwid: USB VID:PID=0403:6011 LOCATION=1-2:1.3
6 ports found
```

```
$ sudo apt-get install openocd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
openocd is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
(artiq-dev) britton@britton1:~/a4t2/artiq$ python artiq/frontend/artiq_flash.py -t sayma -d binaries
here
Open On-Chip Debugger 0.10.0-00007-g977d9e4 (2017-10-23-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
loaded file binaries/bscan_spi_xcku040-sayma.bit to pld device 1 in 3s 858890us
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unknown flash device (ID 0x00000000)

Traceback (most recent call last):
  File "artiq/frontend/artiq_flash.py", line 268, in <module>
    main()
  File "artiq/frontend/artiq_flash.py", line 260, in main
    programmer.do()
  File "artiq/frontend/artiq_flash.py", line 103, in do
    subprocess.check_call(cmdline)
  File "/home/britton/anaconda3/envs/artiq-dev/lib/python3.5/subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openocd', '-s', '/home/britton/anaconda3/envs/artiq-dev/share/openocd/scripts', '-c', 'interface ftdi; ftdi_device_desc "Quad RS232-HS"; ftdi_vid_pid 0x0403 0x6011; ftdi_channel 0; ftdi_layout_init 0x0098 0x008b; reset_config none; adapter_khz 5000; transport select jtag; source [find cpld/xilinx-xc7.cfg]; set CHIP XCKU040; source [find cpld/xilinx-xcu.cfg]; set XILINX_USER1 0x02; set XILINX_USER2 0x03; set JTAGSPI_IR $XILINX_USER1; source [find cpld/jtagspi.cfg]; flash bank xcu.spi0 jtagspi 0 0 0 0 xcu.proxy $XILINX_USER1; flash bank xcu.spi1 jtagspi 0 0 0 0 xcu.proxy $XILINX_USER2; init; pld load 1 {binaries/bscan_spi_xcku040-sayma.bit}; reset halt; flash probe xcu.spi0; irscan xcu.tap $XILINX_USER1; flash write_bank 0 binaries/top.bin 0x0; flash probe xcu.spi1; irscan xcu.tap $XILINX_USER2; flash write_bank 1 binaries/bios.bin 0x0; flash probe xcu.spi1; irscan xcu.tap $XILINX_USER2; flash write_bank 1 binaries/runtime.fbi 0x10000; xcu_program xcu.tap; exit']' returned non-zero exit status 1
```

sbourdeauducq, 2017-11-21:
Looks like a hardware bug that we also see here. Make sure that the RTM is installed (there is a MMC firmware bug that intermittently causes this issue if the RTM is not there) and power-cycle the board. Cc @gkasprow

jbqubit, 2017-11-21:
I've tried power cycling. The RTM is connected. No luck.

sbourdeauducq, 2017-11-21:
Does your board have the 1V8 supply working (corresponding LED on)?

gkasprow, 2017-11-21:
@jbqubit you can flash your MMC easily - all you need is Flashmagic and USB cable

sbourdeauducq, 2017-11-21:
I suppose this also works with OpenOCD and most JTAG cables with the ARM pinout, but I have not tried (there is a risk of bricking the board and of more MMC frustration). In ionpak there is an example OpenOCD script to program a ELF binary into a TM4C MCU.

jbqubit, 2017-11-21:
I've switched to using a bench top power supply, no back plane. All power LEDS on Sayma_AMC are illuminated. Current draw on +12V is 363 mA. A new variety of flashing errors appear.

```
$ python artiq/frontend/artiq_flash.py -t sayma -d binaries
here
Open On-Chip Debugger 0.10.0-00007-g977d9e4 (2017-10-23-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Warn : JTAG tap: xc7.tap       UNEXPECTED: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Error: JTAG tap: xc7.tap expected 1 of 27: 0x0362e093 (mfg: 0x049 (Xilinx), part: 0x362e, ver: 0x0)
Error: JTAG tap: xc7.tap expected 2 of 27: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Error: JTAG tap: xc7.tap expected 3 of 27: 0x0362c093 (mfg: 0x049 (Xilinx), part: 0x362c, ver: 0x0)
Error: JTAG tap: xc7.tap expected 4 of 27: 0x03632093 (mfg: 0x049 (Xilinx), part: 0x3632, ver: 0x0)
Error: JTAG tap: xc7.tap expected 5 of 27: 0x03631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x0)
Error: JTAG tap: xc7.tap expected 6 of 27: 0x03636093 (mfg: 0x049 (Xilinx), part: 0x3636, ver: 0x0)
Error: JTAG tap: xc7.tap expected 7 of 27: 0x03647093 (mfg: 0x049 (Xilinx), part: 0x3647, ver: 0x0)
Error: JTAG tap: xc7.tap expected 8 of 27: 0x0364c093 (mfg: 0x049 (Xilinx), part: 0x364c, ver: 0x0)
Error: JTAG tap: xc7.tap expected 9 of 27: 0x03651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x0)
Error: JTAG tap: xc7.tap expected 10 of 2: 0x03747093 (mfg: 0x049 (Xilinx), part: 0x3747, ver: 0x0)
Error: JTAG tap: xc7.tap expected 11 of 2: 0x03656093 (mfg: 0x049 (Xilinx), part: 0x3656, ver: 0x0)
Error: JTAG tap: xc7.tap expected 12 of 2: 0x03752093 (mfg: 0x049 (Xilinx), part: 0x3752, ver: 0x0)
Error: JTAG tap: xc7.tap expected 13 of 2: 0x03751093 (mfg: 0x049 (Xilinx), part: 0x3751, ver: 0x0)
Error: JTAG tap: xc7.tap expected 14 of 2: 0x03671093 (mfg: 0x049 (Xilinx), part: 0x3671, ver: 0x0)
Error: JTAG tap: xc7.tap expected 15 of 2: 0x036b3093 (mfg: 0x049 (Xilinx), part: 0x36b3, ver: 0x0)
Error: JTAG tap: xc7.tap expected 16 of 2: 0x036b7093 (mfg: 0x049 (Xilinx), part: 0x36b7, ver: 0x0)
Error: JTAG tap: xc7.tap expected 17 of 2: 0x036bb093 (mfg: 0x049 (Xilinx), part: 0x36bb, ver: 0x0)
Error: JTAG tap: xc7.tap expected 18 of 2: 0x036bf093 (mfg: 0x049 (Xilinx), part: 0x36bf, ver: 0x0)
Error: JTAG tap: xc7.tap expected 19 of 2: 0x03667093 (mfg: 0x049 (Xilinx), part: 0x3667, ver: 0x0)
Error: JTAG tap: xc7.tap expected 20 of 2: 0x03682093 (mfg: 0x049 (Xilinx), part: 0x3682, ver: 0x0)
Error: JTAG tap: xc7.tap expected 21 of 2: 0x03687093 (mfg: 0x049 (Xilinx), part: 0x3687, ver: 0x0)
Error: JTAG tap: xc7.tap expected 22 of 2: 0x03692093 (mfg: 0x049 (Xilinx), part: 0x3692, ver: 0x0)
Error: JTAG tap: xc7.tap expected 23 of 2: 0x03691093 (mfg: 0x049 (Xilinx), part: 0x3691, ver: 0x0)
Error: JTAG tap: xc7.tap expected 24 of 2: 0x03696093 (mfg: 0x049 (Xilinx), part: 0x3696, ver: 0x0)
Error: JTAG tap: xc7.tap expected 25 of 2: 0x036d5093 (mfg: 0x049 (Xilinx), part: 0x36d5, ver: 0x0)
Error: JTAG tap: xc7.tap expected 26 of 2: 0x036d9093 (mfg: 0x049 (Xilinx), part: 0x36d9, ver: 0x0)
Error: JTAG tap: xc7.tap expected 27 of 2: 0x036db093 (mfg: 0x049 (Xilinx), part: 0x36db, ver: 0x0)
Info : JTAG tap: xcu.tap tap/device found: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Warn : JTAG tap: xcu.tap       UNEXPECTED: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Error: JTAG tap: xcu.tap  expected 1 of 1: 0x03822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x0)
Error: Trying to use configured scan chain anyway...
Error: xcu.tap: IR capture error; saw 0x3f not 0x01
Warn : Bypassing JTAG setup events due to errors
loaded file binaries/bscan_spi_xcku040-sayma.bit to pld device 1 in 3s 854550us
Info : JTAG tap: xc7.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Warn : JTAG tap: xc7.tap       UNEXPECTED: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Error: JTAG tap: xc7.tap expected 1 of 27: 0x0362e093 (mfg: 0x049 (Xilinx), part: 0x362e, ver: 0x0)
Error: JTAG tap: xc7.tap expected 2 of 27: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Error: JTAG tap: xc7.tap expected 3 of 27: 0x0362c093 (mfg: 0x049 (Xilinx), part: 0x362c, ver: 0x0)
Error: JTAG tap: xc7.tap expected 4 of 27: 0x03632093 (mfg: 0x049 (Xilinx), part: 0x3632, ver: 0x0)
Error: JTAG tap: xc7.tap expected 5 of 27: 0x03631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x0)
Error: JTAG tap: xc7.tap expected 6 of 27: 0x03636093 (mfg: 0x049 (Xilinx), part: 0x3636, ver: 0x0)
Error: JTAG tap: xc7.tap expected 7 of 27: 0x03647093 (mfg: 0x049 (Xilinx), part: 0x3647, ver: 0x0)
Error: JTAG tap: xc7.tap expected 8 of 27: 0x0364c093 (mfg: 0x049 (Xilinx), part: 0x364c, ver: 0x0)
Error: JTAG tap: xc7.tap expected 9 of 27: 0x03651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x0)
Error: JTAG tap: xc7.tap expected 10 of 2: 0x03747093 (mfg: 0x049 (Xilinx), part: 0x3747, ver: 0x0)
Error: JTAG tap: xc7.tap expected 11 of 2: 0x03656093 (mfg: 0x049 (Xilinx), part: 0x3656, ver: 0x0)
Error: JTAG tap: xc7.tap expected 12 of 2: 0x03752093 (mfg: 0x049 (Xilinx), part: 0x3752, ver: 0x0)
Error: JTAG tap: xc7.tap expected 13 of 2: 0x03751093 (mfg: 0x049 (Xilinx), part: 0x3751, ver: 0x0)
[... cut ...] xy $XILINX_USER1; flash bank xcu.spi1 jtagspi 0 0 0 0 xcu.proxy $XILINX_USER2; init; pld load 1 {binaries/bscan_spi_xcku040-sayma.bit}; reset halt; flash probe xcu.spi0; irscan xcu.tap $XILINX_USER1; flash write_bank 0 binaries/top.bin 0x0; flash probe xcu.spi1; irscan xcu.tap $XILINX_USER2; flash write_bank 1 binaries/bios.bin 0x0; flash probe xcu.spi1; irscan xcu.tap $XILINX_USER2; flash write_bank 1 binaries/runtime.fbi 0x10000; xcu_program xcu.tap; exit']' returned non-zero exit status 1
```

jbqubit, 2017-11-21:
@gkasprow I'd rather not mess with the MMC right now. I just want to get ARTIQ running on Sayma_AMC.

gkasprow, 2017-11-21:
@jbqubit did you modify the loading script? @jordens posted some time ago that to make JTAG over USB running, one needs to setup right clock edge.

jbqubit, 2017-11-22:
I am using artiq_flash.py in master. @jordens are there diffs I should know about?

jordens, 2017-11-22:
I am not aware of any.

jbqubit, 2017-11-22:
@gkasprow I have two copies of Sayma_AMC. Their current draw is markedly different. That doesn't seem right.
- TS190717_2 draws 385 mA
- TS190717_7 draws 475 mA

jordens, 2017-11-22:
Please describe your setup.

jbqubit, 2017-11-22:
- Sayma_AMC alone (no back plane, no RTM)
- bench top power supply
- all power LEDS are green on PCB 
- USB cable connected to front microUSB port on AMC
- running artiq_flash.py from master

sbourdeauducq, 2017-11-22:
artiq_flash doesn't support Sayma without RTM. You need to edit the OpenOCD script manually if you want to do anything with Sayma AMC alone.

jbqubit, 2017-11-22:
That's not what a comment in the source says... I started without the RTM due to this comment.

https://github.com/m-labs/artiq/blob/280392708d42afe12cd660ddb6408f0b32e752dc/artiq/frontend/artiq_flash.py#L130

gkasprow, 2017-11-22:
Anyway, in case of troubles with JTAG, try with Xilinx cable first.

sbourdeauducq, 2017-11-22:
That comment simply means that writing the RTM bitstream into the flash is not yet supported. By the way, you still need to make a decision regarding the way the RTM FPGA will be loaded.

sbourdeauducq, 2017-11-22:
Judging from the last log, it seems there are no JTAG troubles other than OpenOCD attempting to access the non-existent RTM FPGA...

jbqubit, 2017-11-22:
Success!

````
$ python artiq/frontend/artiq_flash.py -t sayma -d binaries
here
Open On-Chip Debugger 0.10.0-00007-g977d9e4 (2017-10-23-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362e093 (mfg: 0x049 (Xilinx), part: 0x362e, ver: 0x0)
Info : JTAG tap: xcu.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
loaded file binaries/bscan_spi_xcku040-sayma.bit to pld device 1 in 3s 857648us
Info : JTAG tap: xc7.tap tap/device found: 0x0362e093 (mfg: 0x049 (Xilinx), part: 0x362e, ver: 0x0)
Info : JTAG tap: xcu.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
flash 'jtagspi' found at 0x00000000
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
wrote 5190378 bytes from file binaries/top.bin to flash bank 0 at offset 0x00000000 in 35.231434s (143.869 KiB/s)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
flash 'jtagspi' found at 0x00000000
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
wrote 34952 bytes from file binaries/bios.bin to flash bank 1 at offset 0x00000000 in 0.238553s (143.083 KiB/s)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
flash 'jtagspi' found at 0x00000000
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
wrote 455012 bytes from file binaries/runtime.fbi to flash bank 1 at offset 0x00010000 in 3.092959s (143.664 KiB/s)
```

closed 2017-11-22 by sbourdeauducq

jbqubit, 2017-11-22:
I don't see ARTIQ startup messages on any of the four UARTs exposed via the front panel USB on Sayma_AMC. I do see some MMC messages. 

```
$ flterm /dev/ttyUSB4
Started. CPU clock = 80 MHz, SDRAM clock = 40 Mhz
LWIP HTTP Web Server FreeRTOS Demo...
Waiting for TCPIP thread to initialize...
Starting LWIP HTTP server...
Net up
```