m-labs/artiq #853: Sayma_AMC: interacting with ARTIQ

jbqubit, 2017-11-22:
How do I load ARTIQ Python programs? Ethernet or PPP?

sbourdeauducq, 2017-11-22:
It doesn't work, see my emails about RGMII. PPP is no longer supported.

jbqubit, 2017-11-22:
I didn't appreciate that this is an implication of the RGMII problem you mentioned. The RGMII discussion should be an Issue so everyone involved can contribute to the discussion. Shall I create a new Issue by cutting and pasting from your email to me? Or would you like to create the Issue?

sbourdeauducq, 2017-11-23:
> I didn't appreciate that this is an implication of the RGMII problem you mentioned

I don't know how many times we told you that there was a serious problem with Sayma Ethernet...

> The RGMII discussion should be an Issue so everyone involved can contribute to the discussion

Unless someone wants to fix that bug or provide funding, there is nothing to discuss.

closed 2017-11-23 by sbourdeauducq