m-labs/artiq #857: ARP storm fix for Artiq 3

jonaskeller, 2017-11-28:
As far as I can tell, there is currently no artiq 3 build which implements the ARP storm fix in smoltcp. (Sorry if I missed anything, but I only see dev versions of artiq 4 which include it).

Since we can't have any of our FPGAs on the network without this fix, we are currently stuck with 4.0.dev versions if we want anything newer than 2.5.

When do you plan to integrate it in an artiq 3 release version?

Alternatively, is it possible to have a dev version of artiq 3 which includes this fix?

sbourdeauducq, 2017-11-29:
There will be a 3.1 release soon that includes the fix. Do you confirm that the problem is fixed in 4.0.dev?

jonaskeller, 2017-11-30:
Thanks.
So far, I can only confirm that it didn't happen in >20h of operation. I can't tell whether there was an event which would have triggered it during that time, though.
At the moment, the FPGA can't be on the network for separate reasons - once that is sorted out, I'll monitor it for a longer period of time and will try to also look for the events which used to trigger the ARP storm.

closed 2017-12-07 by sbourdeauducq