m-labs/artiq #858: Gateware bitstream fail on artiq 2.5, recipe for target 'libunwind.o' failed

SoporKiss, 2017-11-29:
I'm trying to build gateware bitstream from artiq 2.5 following [manual-release-2/installing_from_source](https://m-labs.hk/artiq/manual-release-2/installing_from_source.html) and got these errors when I'm running `python3.5 -m artiq.gateware.targets.kc705 -H nist_clock`

```bash
make: Entering directory '/home/iofs/ARTIQ/artiq-dev/misoc_nist_clock_kc705/software/libunwind'
 CX       libunwind.o
In file included from /home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/unwinder/src/libunwind.cpp:27:
In file included from /home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/unwinder/src/UnwindCursor.hpp:30:
In file included from /home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/unwinder/src/AddressSpace.hpp:330:
/home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/include/dyld/link.h:20:12: error: conflicting

      types for 'dl_iterate_phdr'
extern int dl_iterate_phdr (int (*__callback) (struct dl_phdr_info *,
           ^
/home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/include/dyld/dlfcn.h:16:12: note: previous
      declaration is here
extern int dl_iterate_phdr (int (*__callback) (struct dl_phdr_info *,
           ^
In file included from /home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/unwinder/src/libunwind.cpp:27:
In file included from /home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/unwinder/src/UnwindCursor.hpp:30:
/home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/unwinder/src/AddressSpace.hpp:367:15: error: 
      no matching function for call to 'dl_iterate_phdr'
  int found = dl_iterate_phdr(
              ^~~~~~~~~~~~~~~
/home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/include/dyld/dlfcn.h:16:12: note: candidate
      function not viable: no known conversion from '(lambda at
      /home/iofs/ARTIQ/artiq-dev/misoc/misoc/software/unwinder/src/AddressSpace.hpp:368:7)' to
      'int (*)(struct dl_phdr_info *, size_t, void *)' (aka 'int (*)(dl_phdr_info *, unsigned
      long, void *)') for 1st argument
extern int dl_iterate_phdr (int (*__callback) (struct dl_phdr_info *,
           ^
2 errors generated.
Makefile:24: recipe for target 'libunwind.o' failed
make: *** [libunwind.o] Error 1
make: Leaving directory '/home/iofs/ARTIQ/artiq-dev/misoc_nist_clock_kc705/software/libunwind'
Traceback (most recent call last):
  File "/home/iofs/.conda/envs/artiq_2_dev/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/iofs/.conda/envs/artiq_2_dev/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/iofs/ARTIQ/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 418, in <module>
    main()
  File "/home/iofs/ARTIQ/artiq-dev/artiq/artiq/gateware/targets/kc705.py", line 414, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/iofs/ARTIQ/artiq-dev/artiq/artiq/gateware/soc.py", line 61, in build_artiq_soc
    builder.build()
  File "/home/iofs/ARTIQ/artiq-dev/misoc/misoc/integration/builder.py", line 149, in build
    self._generate_software()
  File "/home/iofs/ARTIQ/artiq-dev/misoc/misoc/integration/builder.py", line 126, in _generate_software
    subprocess.check_call(cmd)
  File "/home/iofs/.conda/envs/artiq_2_dev/lib/python3.5/subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/iofs/ARTIQ/artiq-dev/misoc_nist_clock_kc705/software/libunwind']' returned non-zero exit status 2
```

It looks like something is wrong with misoc.

Any suggestion?

sbourdeauducq, 2017-11-29:
Please follow closely the versions specified in the conda recipe in 2.5, and address your python packaging/pip problems.

whitequark, 2017-11-29:
This seems like the wrong compiler is used to build the firmware. Are you sure your `PATH` is correct?

KaifengC, 2017-11-29:
The versions of the packaging is checked by comparing `/artiq/conda/artiq-dev/meta.yaml` and `pip list`.
Only misoc is version 0.1 while the requirement is 0.3.
But I found in misoc/tag/0.4 the version in `setup.py` changed directly from 0.1 to 0.4, so this seems not the problem.
 
I checked the `$PATH` using `echo $PATH` and the `/usr/local/llvm-or1k/bin` is the first one.
Perhaps "the wrong compiler" means that Vivado is not installed correctly?

sbourdeauducq, 2017-11-29:
> Perhaps "the wrong compiler" means that Vivado is not installed correctly?

Vivado is not involved in compiling the part of the firmware where you have the error.

sbourdeauducq, 2017-12-10:
@KaifengC Is that resolved?

closed 2017-12-28 by sbourdeauducq