m-labs/artiq #860: HMC830 fails to lock

sbourdeauducq, 2017-11-29:


sbourdeauducq, 2017-11-30:
It was due to incorrect settings of my 100MHz source. The HMC830 locks fine and then both AD9154 DACs usually work, including JESD lane startup (but intermittenly I get a "bad SYNC" error).

enjoy-digital, 2017-12-01:
It would be interesting to look at the output of the HMC7043 to be sure the clocks are clean. (from the first tests i remember it was not the case, so this could maybe explain the intermittents JESD failures).

sbourdeauducq, 2017-12-18:
The lock is not reliable.
@enjoy-digital Do you see it on your board too?
@gkasprow Can you do thorough tests of the HMC830 including running the ARTIQ code? Can you have a look at our code and see if you spot anything suspicious?
https://github.com/m-labs/artiq/blob/master/artiq/firmware/libboard/hmc830_7043.rs

sbourdeauducq, 2017-12-22:
@jorden iirc @a-shafir found some SPI core bugs. Could that be the source of the problem?

jordens, 2017-12-22:
Could be. Try the workaround in his tree with slow clock.

jordens, 2017-12-22:
https://github.com/m-labs/misoc/issues/65

sbourdeauducq, 2017-12-23:
@jbqubit What happens on your boards?

a-shafir, 2017-12-23:
There is a visible on a scope trailing (extra) short pulse on SCK. With a simple shift register as in CPLD it makes all received bits shifted etc. Probably not all SPI IC are sensitive to it because some of them have "de-glitching" filter. CPLD certainly need a very clean clock. HMC830 possible too. 

The combinatoric logic is not part of MiSoC but in artic/gateware/spi.py 
The proper fix needs latches on all SPI signals that will make the spi transaction 1 sysclk longer. In this case will need to adjust the read phase as well or use higher dividers. It can be done more efficient way: generate the "internal CS" one clock in advance so the latch will not delay the SCK. For this the changes in MiSoC the must.

jordens, 2017-12-28:
m-labs/misoc#65 is fixed

hartytp, 2017-12-30:
Did this fix the HMC830 problem?

If not, @sbourdeauducq did you check the obvious things with a scope? e.g.
- Does the SPI going into the HMC830 look good?
- Does the 150MHz clock going into it look good?
- What does the output look like?

hartytp, 2017-12-30:
Note to self, cf https://github.com/m-labs/sinara/issues/216#issuecomment-314706532

enjoy-digital, 2017-12-30:
Not sure it could be related, but the say i was testing hmc803/hmc7043 i also did this:
https://github.com/enjoy-digital/litex/commit/c02de1632b9cbe269aeb76ef248f2a1869bcb7f9
I remember having an issue with Vivado that was not generating the last bit correctly (observed with litescope), issue that was solved by the way i reorganized part of the code.

hartytp, 2017-12-30:
A scope trace showing the SPI lines would allow us to eliminate this kind of thing very quickly.

@gkasprow I don't have PDFs of the layout. What is the best place to probe the SPI lines for the HMC830?

hartytp, 2017-12-30:
@sbourdeauducq Remind me, what reference frequency are you using for the HMC830? Is it still 100MHz?

sbourdeauducq, 2017-12-30:
Yes.

hartytp, 2017-12-30:
https://github.com/m-labs/artiq/blob/fcc438524c38d1ac336aa8d2b857e4c2338705f3/artiq/firmware/libboard_artiq/hmc830_7043.rs#L34

What is this for? AFAICT, we're running in HMC mode, so the address register is write only, right?

hartytp, 2017-12-30:
https://github.com/m-labs/artiq/blob/fcc438524c38d1ac336aa8d2b857e4c2338705f3/artiq/firmware/libboard_artiq/hmc830_7043.rs#L36

Why divide the 100MHz reference? The PFD is fully specified to run up to 100MHz.

hartytp, 2017-12-30:
> What is this for? AFAICT, we're running in HMC mode, so the address register is write only, right?

Ignore that comment, mis-read the datasheet.

But, I'm still not sure this is correct. @enjoy-digital AFAICT, this sets the soft-reset to 1. The data sheet says that it must be set to 0 for correct operation, but I don't see where you do that. Am I missing something?

enjoy-digital, 2017-12-30:
@hartytp: i'm reusing an initialization sequence from the vendor that is adapted for our case. But yes we can try to set it to 0.

hartytp, 2017-12-30:
@enjoy-digital Good to know. Where exactly did you get the initialisation sequence from? Copied from eval software?

hartytp, 2017-12-30:
Weida is double checking the loop filter, charge pump settings etc.

enjoy-digital, 2017-12-30:
@hartytp: yes from the evaluation software (but don't have the software with me). Review of the settings is welcome!

hartytp, 2017-12-30:
I see. Did that spit the SPI transactions out into a text file? Do you have a copy of the original (don't have access to a windows PC atm to do this myself).

hartytp, 2017-12-30:
@sbourdeauducq @enjoy-digital Have you verified that transitions high before SCLK on startup (must not be any glitches during turn on)? This is needed for proper SPI operation?

I know @gkasprow checked this using the gateware he was using, but did anyone check for glitches on the gateware you're using?

enjoy-digital, 2017-12-30:
@hartytp: what happens on startup should be fine since we are able to communicate with it then. But that would be good yes to do a capture of a spi transaction. (to see if we have the same issue i had with vivado). The spi transactions generated by the gui are described in a text file, but i don't have access to it now.

hartytp, 2017-12-30:
hmm...these chips are just full of undocumented test features. Note in register 6 how you program all the default values, but the data sheet instructs one to program values different to the default in, without any explanation.

hartytp, 2017-12-30:
CP should be 1.6mA with this loop filter not 2.54mA. Probably not enough to prevent it from locking, but won't help.

I'll create a PR later this eve with some slight modifications to the HMC830 regs.

If that doesn't lock, the next step would be looking on a scope. If you guys can't do that, can you send @gkasprow a binary so that he can, please?

sbourdeauducq, 2017-12-31:
> If you guys can't do that, can you send @gkasprow a binary so that he can, please?

This is all packaged on conda with the usual infrastructure.

sbourdeauducq, 2017-12-31:
> Why divide the 100MHz reference? The PFD is fully specified to run up to 100MHz.

> CP should be 1.6mA with this loop filter not 2.54mA. Probably not enough to prevent it from locking, but won't help.

Then why does the ADI software suggest different values? Is the software buggy, was it used with incorrect input, is the documentation inaccurate, does the software contain workarounds for silicon bugs?

hartytp, 2017-12-31:
Not sure. PFD has a lower rating if used in fractions mode IIRC.

Anyway, if you change the PFD frequency you change the loop gain, which will cause problems. Did it lock with my PR?

hartytp, 2017-12-31:
I wouldn't take those recommendations too seriously. They're just giving you something that should work all other things being equal (e.g. loop filter etc). Not a guarantee that it's the best option for you case.

Note also that halving the reference frequency also degrades the phase noise, so we don't want to do it unless forced to.

sbourdeauducq, 2017-12-31:
Made a total of 4 tests (those tests are quite tedious due to slow bitstream loading times, 1V8 bugs, and serwb bugs). Failed twice with your patch, worked twice without.

hartytp, 2017-12-31:
:(

Well, so surprising given how ambiguous some parts of that data sheet are. Anyway, the next step is definitely to verify the inputs with a scope. Once that's done we can have a more systematic look at the register values.

hartytp, 2018-01-01:
The other thing that would be extremely helpful is a scope trace at one of the loop filter components. This would tell us if the PLL has railed (implying an incorrect register setting) or if it's oscillating (something up wit the loop filter).

sbourdeauducq, 2018-01-03:
@gkasprow @jbqubit Can you install ARTIQ, try this on your boards, report whether your HMC830 works well, and do some oscilloscope testing of the kind @hartytp is suggesting?
In case of problems with the ARTIQ installation itself, please use the mailing list or open other issues for support, do not post non-HMC830 content here.

gkasprow, 2018-01-25:
I delivered 100MHz to HMC, loaded 👍 0 $0 $20 spi_hmc830 .
```
0 $1 $2 spi_hmc830 .  \ chip enable by SPI 
0 $2 $2 spi_hmc830 .  \ divider = 2 
0 $6 $303CA spi_hmc830 . 
0 $5 $1628 spi_hmc830 . 
0 $5 $60A0 spi_hmc830 . 
0 $5 $E090 spi_hmc830 . 
0 $5 $2818 spi_hmc830 . 
0 $5 $F88  spi_hmc830 . \ magic word enables PLL out
0 $5 $0 spi_hmc830 . 
0 $7 $14d spi_hmc830 . 
0 $8 $c1beff spi_hmc830 . 
0 $a $2046 spi_hmc830 . 
0 $b $7c061 spi_hmc830 . 
0 $f $81 spi_hmc830 . 
```

and I get 2.9726GHz at the output
The loop voltage is 0.16V

then I load
 0 $3 38 spi_hmc830 . 
 0 $4 3851ec spi_hmc830 . 
and I get 2.8000 Mhz and the voltage is 2.95V

gkasprow, 2018-01-25:
when I read 0x12H I get 
```
>1 $12 0 spi_hmc830 .
 3  ok
```

which means the PLL is locked.

sbourdeauducq, 2018-01-25:
Yes, sometimes it works here, too. Have you tried with artiq?

gkasprow, 2018-01-25:
not yet

hartytp, 2018-01-25:
@gkasprow how many bits are you transferring for each of those writes?

gkasprow, 2018-01-25:
I use 32 bits - this is bit-bang SPI but the code is executed really fast. I get a few MHz SPI clock
```

: spi_hmc830 ( a b c )
	rot 31 lshift rot 25 lshift rot 1 lshift or or
	1 OUT0_SET_REG io! \ SCK hi
	1 2 lshift OUT0_SET_REG io! \ deactivate hmc7043
	1 16 lshift OUT0_SET_REG io! \ OE buff active
	1 8 lshift OUT0_SET_REG io! \ SEN high
    32 0 do
        dup
        1 31 lshift and
        if
           1 1 lshift OUT0_SET_REG io!
        else
           1 1 lshift OUT0_CLR_REG io!
        then
		1 OUT0_CLR_REG io! \ SCK low
		2*
		INP0_REG io@ 4 rshift 1 and +
        1 OUT0_SET_REG io! \ SCK hi	
    loop
	 1 8 lshift OUT0_CLR_REG io! \ SEN low
	 $ffffff and  \ return only valid bits
;
```

jbqubit, 2018-03-29:
@sbourdeauducq @hartytp What's next step on this? It's one of the last bits to nail down before moving on to next revision of Sayma.

hartytp, 2018-03-29:
@jbqubit I'll look at it again after Easter.

hartytp, 2018-03-29:
Next step is to run Greg's code on my board and see if it works.

gkasprow, 2018-03-29:
I ordered the HMC chip long time ago but still didn't get it to my hands.
WIll probably get it next week and do a try.

sbourdeauducq, 2018-04-03:
@hartytp As I understand, this is blocked by you not having a board currently? @enjoy-digital do you still need it?

hartytp, 2018-04-03:
@sbourdeauducq atm there are a few things blocking this from my end:
- my board hadn't arrived yet because our department has been shut for Easter.
- last time I worked on the hmc830 frequent serwb crashes wasted a lot of my time. As it seems that serwb should be relatively quick to fix id rather wait for that to be done first 
- Greg had some problems with the hmc830 on one of his boards, which he plans to look into using the forth coy. Id like him to do that first to check there isn't a hw issue here.

gkasprow, 2018-04-03:
@hartytp I got the HMC chip on Friday. I will solder it tomorrow and let you know what is going on.

sbourdeauducq, 2018-04-18:
@gkasprow Any update?

gkasprow, 2018-04-18:
I did tests with Tom board where I replaced the HMC chip and now it loks.

sbourdeauducq, 2018-04-18:
Yes, please do.

hartytp, 2018-04-18:
@sbourdeauducq Is not having the HMC830 causing you problems right now? Is this something you need fixed asap, or can you wait a bit? I've posted my AMC+RTM to Greg. One thing he will do with it is check that the HMC830 works with his code. Once that's done I'm happy to make sure it works with ARTIQ.

sbourdeauducq, 2018-04-18:
OK.

sbourdeauducq, 2018-05-02:
Just gave Sayma another dozen restarts with the clock through the HMC830 - no bug.

closed 2018-05-02 by sbourdeauducq

marmeladapk, 2018-05-10:
Setup: AMC + RTM, 100MHz 10 dBm signal supplied to SMA input on RTM.

```
Gateware ident 4.0.dev+922.gda1a61b8
Initializing SDRAM...
(...)
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000006s]  INFO(runtime): ARTIQ runtime starting...
[     0.003890s]  INFO(runtime): software version 4.0.dev+922.gda1a61b8
[     0.010153s]  INFO(runtime): gateware version 4.0.dev+922.gda1a61b8
[     0.016434s]  INFO(runtime): log level set to INFO by default
[     0.022141s]  INFO(runtime): UART log level set to INFO by default
[     0.028289s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to
be ready...
[     0.467323s]  INFO(board_artiq::serwb): done.
(...)
[     2.581851s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+922.gda
1a61b8
[     2.589369s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     3.589007s]  INFO(runtime): continuing boot
[     3.591976s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 found
[     3.598123s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 configuration..
.
[     3.605350s]  INFO(board_artiq::hmc830_7043::hmc830): waiting for lock...
[     5.606007s] ERROR(board_artiq::hmc830_7043::hmc830): HMC830 lock timeout. R
egister dump:
[     5.612902s] ERROR(board_artiq::hmc830_7043::hmc830): [0x00] = 0xa7975
[     5.619412s] ERROR(board_artiq::hmc830_7043::hmc830): [0x01] = 0x0002
[     5.625837s] ERROR(board_artiq::hmc830_7043::hmc830): [0x02] = 0x0002
[     5.632263s] ERROR(board_artiq::hmc830_7043::hmc830): [0x03] = 0x0030
[     5.638688s] ERROR(board_artiq::hmc830_7043::hmc830): [0x04] = 0x0000
[     5.645114s] ERROR(board_artiq::hmc830_7043::hmc830): [0x05] = 0x0000
[     5.651539s] ERROR(board_artiq::hmc830_7043::hmc830): [0x06] = 0x303ca
[     5.658052s] ERROR(board_artiq::hmc830_7043::hmc830): [0x07] = 0x014d
[     5.664477s] ERROR(board_artiq::hmc830_7043::hmc830): [0x08] = 0xc1beff
[     5.671077s] ERROR(board_artiq::hmc830_7043::hmc830): [0x09] = 0x153fff
[     5.677676s] ERROR(board_artiq::hmc830_7043::hmc830): [0x0a] = 0x2046
[     5.684102s] ERROR(board_artiq::hmc830_7043::hmc830): [0x0b] = 0x7c061
[     5.690614s] ERROR(board_artiq::hmc830_7043::hmc830): [0x0c] = 0x0000
[     5.697040s] ERROR(board_artiq::hmc830_7043::hmc830): [0x0f] = 0x0081
[     5.703465s] ERROR(board_artiq::hmc830_7043::hmc830): [0x10] = 0x00f0
[     5.709891s] ERROR(board_artiq::hmc830_7043::hmc830): [0x11] = 0x0a2e
[     5.716316s] ERROR(board_artiq::hmc830_7043::hmc830): [0x12] = 0x0000
[     5.722742s] ERROR(board_artiq::hmc830_7043::hmc830): [0x13] = 0x1259
panic at src/libcore/result.rs:945: cannot initialize HMC830/7043: "HMC830 lock
timeout"
```

gkasprow, 2018-05-10:
@marmeladapk did you use version which @hartytp prepared with settings identical to my FORTH code?

hartytp, 2018-05-11:
@marmeladapk I'm seeing the same as you. One thing I don't understand is why 0x04 reads 0x0000...