m-labs/artiq #863: deprecate old manual trees

jordens, 2017-12-05:
From looking at an old manual it not clear that those are actually obsolete and that newer releases are available. We should deprecate those manuals (especially 2.x) using a [banner](https://github.com/ckan/ckan/commit/93d7e768b7c15ca1372f8d32e9f213259b9d4334) or remove them.

jordens, 2017-12-05:
/cc @a-shafir

sbourdeauducq, 2017-12-07:
Seems to me this is done. Reopen if not.

closed 2017-12-07 by sbourdeauducq

jordens, 2017-12-07:
Yes. Done.

sbourdeauducq, 2019-06-14:
Along with the new simplified release process (https://github.com/m-labs/artiq/issues/1326) I propose we simply maintain ``/manual/`` and ``/manual-beta/``.

We also have a PDF manual now, so it's easy for users to keep a local copy of old documentation, and in any case, it can be rebuilt from the Git history.

So I don't think it's worth the hassle to maintain all the ``/manual-release-X/`` manuals.

commit 44c78a5e (2017-12-06): manual: deprecate old release

closes #863
commit 22ba2858 (2017-12-06): manual: deprecate old release

closes #863
commit f2eafa89 (2018-11-21): manual: deprecate old release

closes #863