m-labs/artiq #865: speed drops / packet losses

sbourdeauducq, 2017-12-07:
@cjbe reports (https://github.com/m-labs/artiq/issues/837#issuecomment-349800480):

"Focusing on the JGS516, I now typically get transfer rates of 1.8 - 2.0 MB/s. However for a while (~20 runs of the transfer benchmark, ~10 minutes wall time) this dropped down to 50 kB/s. When this occurred I saw the following in the log:
```
[  4425.733542s]  INFO(runtime::mgmt): new connection from 10.255.6.32:54076
[  4433.832440s]  INFO(runtime::session): new connection from 10.255.6.32:50724
[  4434.239606s]  WARN(runtime): ethernet mac: rx dropped: 1
[  4434.385361s]  WARN(runtime): network error: truncated packet
[  4434.389762s]  WARN(runtime): ethernet mac: rx dropped: 4
[  4434.687018s]  WARN(runtime): ethernet mac: rx dropped: 5
[  4435.285560s]  WARN(runtime): ethernet mac: rx dropped: 6
[  4435.697234s]  WARN(runtime): ethernet mac: rx dropped: 7
[  4435.844028s]  WARN(runtime): ethernet mac: rx dropped: 8
[  4435.992076s]  WARN(runtime): ethernet mac: rx dropped: 9
[  4436.139521s]  WARN(runtime): ethernet mac: rx dropped: 10
[  4436.287049s]  WARN(runtime): ethernet mac: rx dropped: 12
[  4436.434579s]  WARN(runtime): ethernet mac: rx dropped: 13
[  4436.582112s]  WARN(runtime): ethernet mac: rx dropped: 15
[  4436.729639s]  WARN(runtime): ethernet mac: rx dropped: 16
[  4436.877173s]  WARN(runtime): ethernet mac: rx dropped: 19
[  4437.024700s]  WARN(runtime): ethernet mac: rx dropped: 23
[  4437.172228s]  WARN(runtime): ethernet mac: rx dropped: 27
```
This continued up to the message `... mac: rx dropped: 523`. Unfortunately the transfer rate went back to normal before I got a packet dump. With the transfer rate back to normal I did not see any dropped packets."

jordens, 2017-12-07:
From the logs, on the build it that may be occurring occasionally in the windows VM as well.

whitequark, 2017-12-08:
These log messages mean that the runtime is processing packets slower than they arrive, sometimes by quite large chunks. This most likely means that the runtime was busy doing something else.

whitequark, 2017-12-08:
Can you follow the steps here and report? https://github.com/m-labs/artiq/issues/845#issuecomment-346276437

jordens, 2017-12-08:
@whitequark could you also please look at why the unittests in the windows VM are failing with similar symptoms?

whitequark, 2017-12-08:
Of course.

whitequark, 2017-12-27:
The Windows speed issue is a familiar sight:
[image]

whitequark, 2017-12-27:
Actually, no, it's far weirder than that. smoltcp is dropping lots of TCP packets from Windows because of...
... checksum errors.
```
[  1616.340586s] DEBUG(smoltcp::iface::ethernet): cannot process ingress packet: checksum error
[  1616.340699s] DEBUG(smoltcp::iface::ethernet): packet dump follows:
EthernetII src=00-16-3e-41-35-7d dst=02-cc-d9-b2-6a-95 type=IPv4
\ IPv4 src=192.168.1.125 dst=192.168.1.50 proto=TCP
 \ TCP src=52657 dst=1381 seq=1648219550 ack=2647738218 win=64174 len=1460 (checksum incorrect)
```

whitequark, 2017-12-27:
> @whitequark could you also please look at why the unittests in the windows VM are failing with similar symptoms?

Not related at all. Some issue with Xen and Windows caused TCP Large Send Offload to result in packets with invalid checksum being emitted, which smoltcp (correctly) rejected. After turning LSO off there is no problem anymore.

whitequark, 2017-12-27:
@cjbe This problem hard to conclusively assign to any cause without seeing a core log at a TRACE resolution. Please acquire such a log and I'll take a look. This does not seem to be related to smoltcp, but rather it is more likely a bad case in our allocator or something like that (since smoltcp's processing time per packet has a hard upper bound).

cjbe, 2017-12-29:
@whitequark I have not been able to reproduce this problem, even with the gateware version that I originally observed it on. It is possible that it only happens after the KC705 or switch has been power-cycled - I will test this next week when I will have physical access to the hardware again.

whitequark, 2018-01-13:
@cjbe did you have a chance to test this?

sbourdeauducq, 2018-01-23:
@cjbe ping

cjbe, 2018-01-23:
@sbourdeauducq I have not managed to reproduce this - I have tried all combinations of power cycling the KC705 and the switch.

closed 2018-01-23 by whitequark