m-labs/artiq #866: Installing artiq 3.1

philipkent, 2017-12-12:
Trying to install artiq 3.1 from the conda channel on Windows 7 gives the following error:

     conda install -c m-labs artiq-kc705-nist_qc2=3.1
     PackageNotFoundError: Packages missing in current channels:
         - artiq-kc705-nist_qc2 3.1* -> artiq 3.1 py_0+git4c2f25e8 -> llvmlite-artiq 0.12.0

sbourdeauducq, 2017-12-13:
Well the package is there:
https://anaconda.org/m-labs/llvmlite-artiq
Try installing it manually...

whitequark, 2017-12-13:
We only have 0.10.0dev for 32-bit Windows.

sbourdeauducq, 2017-12-13:
Is that 32-bit Windows? Conda + win32 + python3.5 is broken. Reintroducing win32 needs #652.

philipkent, 2017-12-13:
Hi.
Yes, that was on 32 bit windows.  Trying to install on Linux gives a dependency error for prettytable:
     Error: Dependency missing in current linux-64 channels:
          - artiq-kc705-nist_qc2 3.1* -> artiq 3.1 py_0+git4c2f25eq -> prettytable

The Linux version we're running is 64 bit Ubunutu 14.04.5 LTS.  I'll see if I can install the prettytable package manually.

We can also try building from source, we were just trying to avoid the extra compile time.

whitequark, 2017-12-13:
Are you perhaps missing the conda-forge channel?

philipkent, 2017-12-13:
It was missing the conda-forge channel.  That fixed it.

closed 2017-12-13 by jordens