m-labs/artiq #868: .bit build, file location expected by artiq_flash

jbqubit, 2017-12-13:
```~/artiq-dev/artiq$ python3 -m artiq.gateware.targets.sayma_amc_standalone```

produces binaries in 

```~/artiq-dev/artiq/misoc_saymaamcstandalone_sayma_amc```

but ```artiq_flash``` expects them to reside in

```~/artiq-dev/artiq/artiq/binaries/sayma/```

Please update build recipe to copy files to the location expected by ```artiq_flash```. Use of the ```-d``` option isn't practical as there are several directories.

sbourdeauducq, 2017-12-13:
artiq-flash is meant to be used in conjunction with the conda packages, not source builds (unless you reorganize files like the conda script does).

closed 2017-12-14 by sbourdeauducq

whitequark, 2017-12-14:
IMO conda package layout should not differ from source builds, this is just needless complexity. I have encountered this issue as well.

sbourdeauducq, 2017-12-14:
Conda and source builds layouts have pretty different requirements. But I'd be fine changing the conda layout if you insist.

jbqubit, 2017-12-14:
ARTIQ should be internally self consistent. Please change the layout.

sbourdeauducq, 2017-12-14:
What does this have to do with "self-consistency"?

whitequark, 2017-12-14:
I agree with @jbqubit that the filesystem layout of artiq_flash (which, only by coincidence, is what conda packages use) should match the output of `python -m artiq.gateware.targets.xxx`. This also makes it easier to do things like building everything but the bitstream locally, which unfortunately are also blocked because of similarly poor design decisions in regards to paths elsewhere in migen/misoc.

sbourdeauducq, 2017-12-14:
What design decisions exactly?

whitequark, 2017-12-14:
@sbourdeauducq using absolute paths, assuming that if a script is generated it needs to be run immediately, assuming that a generated file will be run on the same host. I was trying to make it possible to make misoc build process network transparent, but wasted nearly a week of time and gave up in the end, it would require too many changes in misoc. Maybe another time...

sbourdeauducq, 2017-12-14:
It was never intended to be a network build system. If we start doing complicated things like that, maybe there are existing build systems that can be reused?

sbourdeauducq, 2017-12-14:
Anyway, to go back to the original topic:
* with the source, building one output binary involves many intermediate files (object files, Vivado temporary files, reports and trash, etc.), that should to be placed in their own directory.
* with the package, there is no need for those directories, the binaries for one board should be placed together.

whitequark, 2017-12-14:
There are, but I wanted to hack together something that would mainly just let me test gateware changes without having Xilinx. Actually, maybe it is easier to just give up and install Xilinx junk on my laptop. Is it possible for me to borrow M-Labs' license using their flexlm crap?

jordens, 2017-12-14:
I'd be fine with packaging the binaries as `gateware/top.bit`, `software/runtime/runtime.fbi`, `software/bios/bios.bin`. The added directories are a minor inconvenience.

jbqubit, 2017-12-14:
Ok. I can better rephrase what I'm asking for now. I'd like to be able to use ```artiq_flash``` with both conda and locally generated binaries. Perhaps a flag to ```artiq_flash``` could indicate which directory contains the build results.

sbourdeauducq, 2017-12-15:
This is what the commit above does.

jbqubit, 2017-12-15:
Great! Thanks. :)

commit 71db953e (2017-12-14): artiq_flash: add srcbuild option to look into MiSoC source build trees. Closes #868