m-labs/artiq #872: Sayma_AMC artiq_flash :: Unknown flash device (ID 0x00000000)

jbqubit, 2017-12-14:
Occasionally I when running ```artiq_flash -t sayma``` I get an openocd error ```Error: Unknown flash device (ID 0x00000000)```. Usually, cycling the power to Sayma_AMC several times fixes the problem. I've not found a way to reproduce this reliably.

jordens, 2017-12-14:
I.e. always provide full logs...

jbqubit, 2017-12-14:
```
$ artiq_flash -t sayma
Design: top;COMPRESS=TRUE;UserID=0XFFFFFFFF;Version=2016.2
Part name: xcku040-ffva1156-1-c
Date: 2017/12/13
Time: 17:45:27
Bitstream payload length: 0x5928e4
Open On-Chip Debugger 0.10.0-00007-g977d9e4 (2017-10-23-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
loaded file /home/britton/anaconda3/envs/artiq-dev/share/bscan-spi-bitstreams/bscan_spi_xcku040-sayma.bit to pld device 1 in 3s 850847us
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unknown flash device (ID 0x00000000)

Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-dev/bin/artiq_flash", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_flash')()
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_flash.py", line 284, in main
    programmer.do()
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_flash.py", line 105, in do
    subprocess.check_call(cmdline)
  File "/home/britton/anaconda3/envs/artiq-dev/lib/python3.5/subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openocd', '-s', '/home/britton/anaconda3/envs/artiq-dev/share/openocd/scripts', '-c', 'interface ftdi; ftdi_device_desc "Quad RS232-HS"; ftdi_vid_pid 0x0403 0x6011; ftdi_channel 0; ftdi_layout_init 0x0098 0x008b; reset_config none; adapter_khz 5000; transport select jtag; source [find cpld/xilinx-xc7.cfg]; set CHIP XCKU040; source [find cpld/xilinx-xcu.cfg]; target create xcu.proxy testee -chain-position xcu.tap; set XILINX_USER1 0x02; set XILINX_USER2 0x03; flash bank xcu.spi0 jtagspi 0 0 0 0 xcu.proxy $XILINX_USER1; flash bank xcu.spi1 jtagspi 0 0 0 0 xcu.proxy $XILINX_USER2; init; pld load 1 {/home/britton/anaconda3/envs/artiq-dev/share/bscan-spi-bitstreams/bscan_spi_xcku040-sayma.bit}; reset halt; flash probe xcu.spi0; flash erase_sector 0 0 89; flash write_bank 0 {/tmp/tmpfzefuuol} 0x0; flash verify_bank 0 {/tmp/tmpfzefuuol} 0x0; flash probe xcu.spi1; flash erase_sector 1 0 0; flash write_bank 1 {/home/britton/artiq-dev/artiq/artiq/binaries/sayma/bios.bin} 0x0; flash verify_bank 1 {/home/britton/artiq-dev/artiq/artiq/binaries/sayma/bios.bin} 0x0; flash probe xcu.spi1; flash erase_sector 1 1 8; flash write_bank 1 {/home/britton/artiq-dev/artiq/artiq/binaries/sayma/runtime.fbi} 0x10000; flash verify_bank 1 {/home/britton/artiq-dev/artiq/artiq/binaries/sayma/runtime.fbi} 0x10000; xcu_program xcu.tap; exit']' returned non-zero exit status 1
```

A bit orthogonal comment... On seeing the hugeness of the monolithic openocd command. It's a bit hard to debug which part of the TCL is causing the trouble. Consider making several independent calls to openocd instead of one big one.

jbqubit, 2017-12-14:
Also, 

```
$ artiq_flash -t sayma start
Open On-Chip Debugger 0.10.0-00007-g977d9e4 (2017-10-23-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
```

jordens, 2017-12-14:
That would not work.

jordens, 2017-12-14:
From the symptoms this is the same as the 1.8V supply dying and the same as the FTDI chip locking up when bein accessed (or openocd ctrl-C-ed) when Sayma is not powered. We already have issues for those. 
https://github.com/m-labs/sinara/issues/358
https://github.com/m-labs/sinara/issues/255

closed 2017-12-14 by jordens

jbqubit, 2017-12-14:
@jordens don't summarily close Issues without discussion. 
- The 1.8V supply LED is on. Potential is 1.8 volts. 
- How can we talk about accessing the FTDI chip when there is no power supplied to Sayma? Does it also derive power from the USB port on the PC?  
- #255 is highly speculative about the cause of ```ID 0x00000000``` 
- Were you able to reproduce my bug?

sbourdeauducq, 2017-12-15:
> Were you able to reproduce my bug?

Sure. We regularly suffer from it, having to build a remotely-operated power-cycler for the boards and having to use it from time to time in the middle of debugging something else because the board failed again. This is a hardware or MMC firmware issue so, besides being a duplicate, it should not be in the artiq repository.

jordens, 2017-12-15:
Our workflow will continue this way. This is neither a democracy nor a therapeutic session. You need to figure out how to cope with duplicate, out-of-place issues being closed. Ask yourself whether your behavior makes ARTIQ or Sinara better.

jbqubit, 2017-12-15:
Ignoring that last [hyperbolic](https://www.vocabulary.com/dictionary/hyperbolic) statement, let's look at the Issues cited by @jordens. 

# https://github.com/m-labs/sinara/issues/255
Issue: review FTDI chip USB signals
- This Issue was created on August 18 and not touched until I ping'd developers yesterday
- M-Labs didn't engage after August 18 to further to investigate 

# https://github.com/m-labs/sinara/issues/358
Issue: Sayma loses 1.8V supply
-  This Issue was created on Nov 13
- @gkasprow responded with 3 concrete debugging paths on Nov 13
- M-Labs didn't engage with @gkasprow after Nov 13  

> Sure. We regularly suffer from it, having to build a remotely-operated power-cycler for the boards and having to use it from time to time in the middle of debugging something else because the board failed again. 

So this was sufficiently annoying that you had to implement a technical work around for your test system. And yet based on Issue tracking, M-Labs undertook no effort to promote discussion aimed at understanding and fixing the underlying problem. Months later the problem persists. 

> This is a hardware or MMC firmware issue so, besides being a duplicate, it should not be in the artiq repository.

Setting aside the question of who is supposed to fix this bug, I'm concerned that the problem has persisted unaddressed for so long. Please recall that M-Labs is contracted to take responsibility for "Project management for Sayma and Metlino testing and integration." I don't care if the Issues are in github's Sinara or ARTIQ Issue tracking system. Your responsibility as project manager is to stay on top of bugs and work with developers to squash them. I'm disappointed by your attentiveness to this Issue.

sbourdeauducq, 2017-12-15:
> @gkasprow responded with 3 concrete debugging paths on Nov 13
> M-Labs didn't engage with @gkasprow after Nov 13

So what? This issue is Greg's responsibility and he says he is confident debugging this kind of power supply issue. We have enough on our plate with RGMII and other insufferable bugs, and little time to play around with Arduino, Exar software, MMC firmware, and Windows.

jbqubit, 2017-12-15:
> We have enough on our plate

When negotiating FY17 contracts M-Labs requested project management responsibility. You were awarded that role as part of your contract. Project management is an ongoing process and not a task that can be deferred until it's convenient.

sbourdeauducq, 2017-12-15:
We filed the issue, Greg was informed and sounds very capable of fixing it, and until he does it we set up a rapid workaround to reduce its impact on our productivity. I don't see what the problem is here.

jbqubit, 2017-12-15:
@sbourdeauducq M-Labs is responsible for tracking the overall project and ensuring that dependent parties are giving due attention to Issues. In this case it's gone unattended since August. Please work with WUT to address it.

sbourdeauducq, 2017-12-15:
Again I disagree, there are higher priority tasks than a problem that:
* may not be a problem with Sayma at all and may just be some my USB cables being low-quality.
* only potentially affects people using JTAG or UART, which will not be normally used for regular Sayma operation.
* is easily worked around by using different USB cables. With the cables currently installed at M-Labs there are no USB issues visible on any of the 3 boards.

jbqubit, 2017-12-15:
> only potentially affects people using JTAG or UART, which will not be normally used for regular Sayma operation.

I see it while running ```artiq_flash``` which is an operation that users normally do. 

> is easily worked around by using different USB cables. With the cables currently installed at M-Labs there are no USB issues visible on any of the 3 boards.

That's news to me. Please add that observation to the relevant Issue. https://github.com/m-labs/sinara/issues/255