m-labs/artiq #873: Sayma_AMC: MMC doesn't init power ICs

jbqubit, 2017-12-14:
Set "M switches" on PCB "HS1" to 1234 -> 0111. 

I can flash the board using ```artiq_flash -t sayma``` and see on ttyUSB2 MiSoC BIOS messages. 

When I issue ```artiq_flash -t artiq start``` I see that the MiSoC BIOS messages again appear on ttyUSB2. 

When I remove power from Sayma, wait 10 seconds, then reapply power... 
- Only 12 V power LED on Sayma_AMC is illuminated

Cycle power again... remove power from Sayma, wait 10 seconds, then reapply power... 
- Only 12 V power LED on Sayma_AMC is illuminated

Cycle power again... remove power from Sayma, unplug micro USB cable from front of Sayma_AMC, wait 10 seconds, then reapply power... 
- All power LEDs are illuminated on Sayma_AMC

Cycle power again... remove power from Sayma, attach micro USB cable from front of Sayma_AMC, wait 10 seconds, then reapply power... 
- All power LEDs are illuminated on Sayma_AMC

I've seen this full sequence twice.

jbqubit, 2017-12-14:
@gkasprow

gkasprow, 2017-12-14:
@jbqubit does this happen on single board or all boards?
There is an issue that hot USB plugging resets the MMC.
It is caused by a glitch from bus-powered USB interface to the reset circuit of the MMC by the embedded MMC FLASH programmer.
But once the USB cable sits in its socket this should not happen.

jbqubit, 2017-12-14:
The following is reproducible....  

AMC on, serial cable attached but no serial communication initiated by PC -> power cycle AMC -> all power LEDS come on. Good.

AMC on, serial cable attached but no serial communication initiated by PC -> power cycle AMC -> all power LEDS come on. Good.

AMC on, serial cable attached -> open serial communication on PC, putty /dev/ttyUSB4 -> power cycle AMC -> only 12 V power LED on -> close putty window on PC (cable still attached) -> all power LEDs turn on

gkasprow, 2017-12-14:
@jbqubit this is correct behaviour. DRT line causes MMC reset. Please disable flow control. Or unsolder pin 3 of T23. I can add a DIP switch that enables USB firmware update...

gkasprow, 2017-12-14:
You can also short pins 1 and 2 of T24.

sbourdeauducq, 2017-12-15:
Please put issues like this into the sinara repository, the problem is clearly not coming from ARTIQ.

closed 2017-12-15 by sbourdeauducq

jbqubit, 2017-12-15:
Sure, I can see in this case that this is more likely a hardware related Issue. I appreciate that you'd prefer to not have it cluttering ARTIQ. However, closing miss-filed Issue without first confirming that they've been properly transferred to a different Issue system risks loosing track of bugs. Please don't do this.