m-labs/artiq #874: moninj: frequent updates to TTL input value break TCP connection

jonaskeller, 2017-12-15:
Using 3.1, nist_qc2, Windows 7 64bit.

Procedure to reproduce the error:
- Input reasonably fast square wave into a TTL input (here I used ca. 100kHz)
- Start dashboard

For a while, the FPGA just sends each packet twice. At some point (seems always shortly after the FPGA sends a fresh ARP request for the host's IP), both just keep sending each other the same acknowledgments repeatedly, at about 1kHz. This second part didn't occur with a 50kHz input signal; I didn't narrow it down any further.

Trying to start new experiments in that state results in a timeout (the FPGA doesn't even reply to the host's ARP request) and the only solution I found was to reboot it.

I'm attaching two logs.
The one recorded on the host was started before doing the following:
- FPGA reboot
- setting the core log level to TRACE
- starting the dashboard

The IP ending in "107" is the FPGA, "...117" the host.

[logs_TCP_issue.zip](https://github.com/m-labs/artiq/files/1561448/logs_TCP_issue.zip)

whitequark, 2017-12-15:
Right.  Here's what happens here:
[image]
[snip]
[image]

whitequark, 2017-12-15:
Looks like smoltcp got stuck in a keepalive loop.

whitequark, 2017-12-27:
> At some point (seems always shortly after the FPGA sends a fresh ARP request for the host's IP)

This should happen every 60 seconds, as per RFC.

> Trying to start new experiments in that state results in a timeout (the FPGA doesn't even reply to the host's ARP request) and the only solution I found was to reboot it.

This seems very odd; the symptoms match the core device crashing. Can you please take a look at the serial console?

In any case, I have ran moninj for a fairly long time and have not observed this issue. It also seems to match https://github.com/m-labs/smoltcp/issues/62.

jonaskeller, 2018-01-04:
I tried reproducing the error today. In 3.1 py_35+git3ba82cf1, only the duplicate TCP packets appeared, but it never entered the ACK loop. So maybe this is already fixed.

Reverting to 3.1 py_6+git231bf77b, it did enter the loop and prevented me from starting new experiments, but contrary to what I described above, there was no ARP request without response (maybe that was a coincidental occurrence of #877 ?) - sorry about the confusion.

Instead, I got two different errors this time:

    Terminating with exception (ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host)

The core device hadn't crashed, but just in case, here's its log (with two failed attempts to start an experiment at 156s and 174s):

	MiSoC BIOS
	(c) Copyright 2007-2017 M-Labs Limited
	Built Dec 20 2017 03:50:59

	BIOS CRC passed (e8b4b4b6)
	Initializing SDRAM...
	Write leveling: 15* 15* 13  14   9   8   5   6  completed
	Read bitslip: 7 6 5 4 3 2 
	Read delays: 7:02-12  6:02-13  5:05-15  4:06-16  3:11-22  2:11-21  1:01-10  0:01-10  completed
	Memtest OK
	Booting from serial...
	Press Q or ESC to abort boot completely.
	sL5DdSMmkekro
	Timeout
	Booting from flash...
	Loading 498872 bytes from flash...
	Executing booted program.
	[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
	[     0.003862s]  INFO(runtime): software version 3.1+6.g231bf77b
	[     0.009601s]  INFO(runtime): gateware version 3.1+6.g231bf77b
	[     0.015328s]  INFO(runtime): press 'e' to erase startup and idle kernels...
	[     1.015006s]  INFO(runtime): continuing boot
	[     1.018204s]  INFO(runtime): using MAC address 00-0a-35-03-18-d4
	[     1.023970s]  INFO(runtime): using IP address 132.163.182.107
	[     1.029724s] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry, using internal RTIO clock
	[     1.040169s]  INFO(runtime): UART log level set to INFO by default
	[     1.047247s]  INFO(runtime::mgmt): management interface active
	[     1.060564s]  INFO(runtime::session): accepting network sessions
	[     1.074582s]  INFO(runtime::session): running startup kernel
	[     1.079057s]  INFO(runtime::session): no startup kernel found
	[     1.084764s]  INFO(runtime::session): no connection, starting idle kernel
	[     1.091637s]  INFO(runtime::session): no idle kernel found
	[     2.321091s]  WARN(runtime): network error: truncated packet
	[    12.810583s]  INFO(runtime::mgmt): new connection from 132.163.182.117:57591
	[    12.816401s]  INFO(runtime::mgmt): changing UART log level to TRACE
	[    35.245136s]  INFO(runtime::mgmt): new connection from 132.163.182.117:57609
	[    37.381917s]  INFO(runtime::moninj): new connection from 132.163.182.117:57619
	[   137.935557s]  WARN(runtime): network error: truncated packet
	[   155.639772s]  WARN(runtime): network error: malformed packet
	[   155.981151s]  WARN(runtime): network error: malformed packet
	[   156.518012s]  WARN(runtime): network error: malformed packet
	[   156.804166s]  INFO(runtime::session): new connection from 132.163.182.117:57641
	[   157.764746s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   159.156445s]  INFO(runtime::session): no connection, starting idle kernel
	[   159.303466s]  INFO(runtime::session): no idle kernel found
	[   174.286356s]  INFO(runtime::session): new connection from 132.163.182.117:57648
	[   175.311155s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   177.773034s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   180.235128s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   182.696144s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   184.282354s]  WARN(runtime): network error: malformed packet
	[   185.108863s]  WARN(runtime): network error: malformed packet
	[   185.255453s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   185.605001s]  WARN(runtime): network error: malformed packet
	[   187.732887s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   190.260196s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   191.651338s]  INFO(runtime::session): no connection, starting idle kernel
	[   191.798339s]  INFO(runtime::session): no idle kernel found
	[   213.857144s]  WARN(runtime): network error: malformed packet
	[   215.303964s]  WARN(runtime): network error: malformed packet
[... 92 lines cut ...]
	[   658.575983s]  WARN(runtime): network error: malformed packet
	[   693.147519s]  WARN(runtime): network error: malformed packet
	[   693.152030s]  WARN(runtime): network error: malformed packet
	[   693.298014s]  WARN(runtime): network error: malformed packet
	[   722.853738s]  WARN(runtime): network error: malformed packet
	[   722.858249s]  WARN(runtime): network error: malformed packet
	[   723.133177s]  WARN(runtime): network error: malformed packet
	[   751.784116s]  WARN(runtime): network error: malformed packet
	[   751.993704s]  WARN(runtime): network error: malformed packet
	[   752.590256s]  WARN(runtime): network error: malformed packet
	[   820.664578s]  WARN(runtime): network error: malformed packet
	[   820.810114s]  WARN(runtime): network error: malformed packet
	[   821.607291s]  WARN(runtime): network error: malformed packet
	[   849.844352s]  WARN(runtime): network error: malformed packet
	[   850.510694s]  WARN(runtime): network error: malformed packet
	[   851.307817s]  WARN(runtime): network error: malformed packet
	[   879.679296s]  WARN(runtime): network error: malformed packet
	[   880.540543s]  WARN(runtime): network error: malformed packet
	[   880.686312s]  WARN(runtime): network error: malformed packet
	[   969.455465s]  INFO(runtime::session): new connection from 132.163.182.117:57936
	[   969.735345s]  WARN(runtime): network error: truncated packet
	[   970.078163s]  WARN(runtime): network error: checksum error
	[   971.090321s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   996.754694s]  INFO(runtime::session): no connection, starting idle kernel
	[   996.901461s]  INFO(runtime::session): no idle kernel found

whitequark, 2018-01-05:
> In 3.1 py_35+git3ba82cf1, only the duplicate TCP packets appeared, but it never entered the ACK loop.

By "only the duplicate TCP packets appeared", which packets do you mean specifically? And do you think these are problematic?

> Reverting to 3.1 py_6+git231bf77b,

That version uses smoltcp with known severe TCP bugs, so problems with it are expected and reporting them is not really useful for anything.

jonaskeller, 2018-01-05:
> By "only the duplicate TCP packets appeared", which packets do you mean specifically? And do you think these are problematic?

I'm referring to the upper part of the capture you show in your first reply to this issue. It's probably not an issue, I'm just mentioning the symptoms without making any judgement.

> That version uses smoltcp with known severe TCP bugs, so problems with it are expected and reporting them is not really useful for anything.

If this means that this issue has been caused by a known bug, I guess that resolves it. If that is also true for #877, please let me know so I don't waste any time. We currently can't switch to the newest 3.1 version, however, due to #885. Is there an intermediate version that includes the smoltcp fixes that you don't expect to have this bug? And if so, can you please let me know how to get your build bot to use a specific commit?

whitequark, 2018-01-05:
I had no idea #885 is a blocker for you, you should have mentioned it. It is of course very easy to fix, and I will do so.

Regarding the buildbot, `force build --revision=HASH ...` does the job.

jonaskeller, 2018-01-05:
Great, thanks! Until then, I'll stick with 3.1 py_6+git231bf77b and just not use moninj.

So since I couldn't reproduce this with the newest version, it looks like this issue can be closed.

closed 2018-01-05 by jonaskeller

whitequark, 2018-01-05:
Oh and please do try to reproduce #877; while it *probably* was caused by a known bug, it could have been caused by anything (unlike this issue) so the report would be useful, if any.

whitequark, 2018-01-09:
@jonaskeller FWIW #885 has been fixed.