m-labs/artiq #877: FPGA stops responding to ethernet after a few hours

jonaskeller, 2017-12-20:
Using artiq 3.1 py_6+git231bf77b nist_qc2, also observed with the 3.1 version from the "main" conda repository.

Several times a day, our FPGA stops responding via ethernet. Typically, it's the host's ARP request that never gets a response, but adding an entry to the ARP cache manually makes it get stuck when the host doesn't get a response to its TCP SYN request.

This seems to happen regardless of what the FPGA is doing at the time - I have seen it happen while an experiment was running as well as after just resetting the FPGA and not talking to it at all for few hours.

Here is the beginning of the log for the latter case, where the FPGA was reset shortly after t=0 and doesn't reply to the ARP requests coming from some other machine on our network after ~12ks:
[image]

I'm happy to provide further information on this if you let me know how.

whitequark, 2017-12-21:
@jonaskeller Could you please run flterm to see if there are any messages on the console? This seems like the runtime panicking, and seeing the panic message would be very helpful.

sbourdeauducq, 2017-12-21:
Do we have backtraces yet?

whitequark, 2017-12-21:
Not yet.

jonaskeller, 2017-12-23:
I'm afraid I can't do any tests right now, because I've already left for the holidays. Will do so once I'm back Jan 4th.

whitequark, 2017-12-27:
If there is no response to ARP requests it means the core device has crashed. Please acquire a backtrace from the serial console next time this happens.

sbourdeauducq, 2018-01-09:
@jonaskeller Please update to 3.2 to get those backtraces.

jonaskeller, 2018-01-13:
Just an update: I've been monitoring an idle FPGA every night this week and haven't seen it crash yet (using 3.2 py_1+gita4337944). Given that it used to happen multiple times a day when I opened this issue, it seems likely that the problem is fixed.

closed 2018-01-13 by whitequark