m-labs/artiq #878: IR generator bug on "" + str(1)

jordens, 2017-12-21:
```python
from artiq.experiment import *

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        raise ValueError("foo" + str(1))
```

```
(py35) rj@tito:~/.../artiq/examples/master (urukul-ad9912)$ artiq_run test_bug.py
Traceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "/home/rj/work/nist/artiq/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rj/work/nist/artiq/artiq/coredevice/core.py", line 124, in run
    self.compile(function, args, kwargs, set_result)
  File "/home/rj/work/nist/artiq/artiq/coredevice/core.py", line 107, in compile
    library = target.compile_and_link([module])
  File "/home/rj/work/nist/artiq/artiq/compiler/targets.py", line 180, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "/home/rj/work/nist/artiq/artiq/compiler/targets.py", line 180, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "/home/rj/work/nist/artiq/artiq/compiler/targets.py", line 134, in compile
    llmod = module.build_llvm_ir(self)
  File "/home/rj/work/nist/artiq/artiq/compiler/module.py", line 89, in build_llvm_ir
    attribute_writeback=self.attribute_writeback)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 491, in process
    self.process_function(func)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 653, in process_function
    llinsn = getattr(self, "process_" + type(insn).__name__)(insn)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 699, in process_Alloc
    llsize = self.map(insn.operands[0])
IndexError: list index out of range
```

klickverbot, 2017-12-21:
More minimal example:

```py
    @kernel
    def run(self):
        return str(1)
```

The call typechecks as the invocation of a user-defined constructor, but during ARTIQ IR generation, it is (correctly) recognized as a builtin and this odd-looking code path is taken: https://github.com/m-labs/artiq/blob/69d7e93e99eb0cfc3104e06efe7f6c78286d6d1b/artiq/compiler/transforms/artiq_ir_generator.py#L1744-L1745 (Note that this creates an invalid `Alloc` node, leading to the range error above – not sure when this path is even supposed to be hit.)

As far as I can see, we don't currently support string formatting on the core device, but this would be quite useful to have for diagnostic messages/… It seems like this should be easy to implement using `snprintf` and the existing polymorphic printing code – @whitequark?

whitequark, 2017-12-27:
> It seems like this should be easy to implement using snprintf and the existing polymorphic printing code – @whitequark?

I would much rather get rid of all C dependencies....

whitequark, 2017-12-27:
The miscompilation is now fixed, and I don't think str conversion should go into 3.2.

sbourdeauducq, 2017-12-27:
This issue is about the compiler crashing; string formatting is not supported and considering other projects that are funded and late (Sayma, DRTIO, etc.) we should not implement it for now. @klickverbot if you want it, please open a new issue.

closed 2017-12-27 by sbourdeauducq