m-labs/artiq #883: coredevice.test_performance.TransferTest error on windows

jordens, 2018-01-02:
```
test_device_to_host (coredevice.test_performance.TransferTest) ... ERROR
test_host_to_device (coredevice.test_performance.TransferTest) ... 2330157.608132362 B/s
ok

[...]

======================================================================
ERROR: test_device_to_host (coredevice.test_performance.TransferTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\slave64\artiq-win64-test\build\artiq\test\coredevice\test_performance.py", line 52, in test_device_to_host
    device_to_host_rate = exp.device_to_host()
  File "c:\slave64\artiq-win64-test\build\artiq\language\core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "c:\slave64\artiq-win64-test\build\artiq\coredevice\core.py", line 133, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "c:\slave64\artiq-win64-test\build\artiq\coredevice\comm_kernel.py", line 606, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "c:\slave64\artiq-win64-test\build\artiq\coredevice\comm_kernel.py", line 598, in _serve_exception
    raise python_exn
ZeroDivisionError: cannot divide by zero
```

http://buildbot.m-labs.hk/builders/artiq-win64-test/builds/670/steps/python_coverage/logs/stdio

jordens, 2018-01-03:
Also on linux
http://buildbot.m-labs.hk/builders/artiq/builds/1904/steps/python_coverage_1/logs/stdio

jordens, 2018-01-03:
```python
    @kernel
    def device_to_host(self):
        t0 = self.core.get_rtio_counter_mu()
        self.sink(self.data)
        t1 = self.core.get_rtio_counter_mu()
        return len(self.data)/self.core.mu_to_seconds(t1-t0)
```

@sbourdeauducq `rtio_counter` transfer not working ccorrectly?

sbourdeauducq, 2018-01-04:
Did you also see ``test_host_to_device`` fail?
Or this: https://github.com/m-labs/artiq/blob/c2cc29142dda7f23d531e2759f9ca9a80bb5569d/artiq/test/coredevice/test_rtio.py#L488
We can add another test specifically for the RTIO counter readback.

jordens, 2018-01-04:
No. You are seeing the same things I see.

sbourdeauducq, 2018-01-23:
> @sbourdeauducq rtio_counter transfer not working ccorrectly?

Not likely. Otherwise we would see failures in ``test_time_keeps_running`` as well.

sbourdeauducq, 2018-01-28:
``get_rtio_counter_mu`` seems to be working fine. Likely a runtime bug (parallel programming with async RPCs? core device memory corruption when dealing with the large array?) or miscompilation (though compilation is a pretty deterministic process and this bug is intermittent). @whitequark

whitequark, 2018-01-28:
Possibly a race condition with async RPCs...

sbourdeauducq, 2018-01-28:
If this is confirmed, it might make sense to switch to a single mailbox.

whitequark, 2018-01-28:
Or just fix the race condition.

sbourdeauducq, 2018-01-28:
I remember you saying that, if another concurrency bug was found in this code, you'd switch to a single mailbox. :)

whitequark, 2018-02-21:
I don't think this has anything to do with async RPCs. Even wrapping the offending line in `try`/`except` silences the bug (that is otherwise moderately easy to reproduce, it usually surfaces within ten iterations if running the test in a loop). Any attempts to observe t1/t2 also hide the bug. This seems to point to a miscompilation of some sort.

whitequark, 2018-02-21:
A very interesting question is why `test_device_to_host` sometimes fails but `test_host_to_device` never does...

whitequark, 2018-02-21:
On second look, this does have *something* to do with async RPCs, but not at all clear what. The following experiment reproduces the crash, often at the first iteration:
```
from artiq.experiment import *

class Transfer(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.data = b"\x00"*(10**3)

    @rpc(flags={"async"})
    def sink(self, data):
        assert data == self.data

    @rpc(flags={"async"})
    def print(self, t0, t1, dt):
        print(t0, t1, dt)

    @kernel
    def run_one(self):
        t0 = self.core.get_rtio_counter_mu()
        self.sink(self.data)
        t1 = self.core.get_rtio_counter_mu()
        self.print(t0,t1,self.core.mu_to_seconds(t1-t0))
        return len(self.data)/self.core.mu_to_seconds(t1-t0)

    @kernel
    def run(self):
        while True:
            print(".")
            self.run_one()
```
... but the values look all sane:
```
$ artiq_run test.py
.
1644554885800 1644555052560 0.00016676
```

whitequark, 2018-02-21:
We had a broken floating point comparison function, but *only* for the greater-or-equal operation.

closed 2018-02-21 by whitequark

whitequark, 2018-02-21:
@sbourdeauducq This should definitely go into 3.5.

commit 96f697ec (2018-02-21): firmware: update compiler_builtins to unbreak __gtdf2.

Fixes #883.