m-labs/artiq #885: core device backtrace zymbolization stumbles over `?` as line number.

jordens, 2018-01-03:
```
WARNING:artiq.coredevice.comm_kernel:Mismatch between gateware (4.0.dev+298.g64612b3e.dirty) and software (4.0.dev+307.ge3d66d28.dirty) versions
Traceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/rj/src/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/rj/src/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/rj/src/artiq/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "/home/rj/src/artiq/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rj/src/artiq/artiq/coredevice/core.py", line 133, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/home/rj/src/artiq/artiq/coredevice/comm_kernel.py", line 606, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/home/rj/src/artiq/artiq/coredevice/comm_kernel.py", line 587, in _serve_exception
    traceback = list(reversed(symbolizer(backtrace))) + \
  File "/home/rj/src/artiq/artiq/coredevice/core.py", line 111, in <lambda>
    lambda addresses: target.symbolize(library, addresses), \
  File "/home/rj/src/artiq/artiq/compiler/targets.py", line 220, in symbolize
    backtrace.append((filename, int(line), -1, function, address))
ValueError: invalid literal for int() with base 10: '?'
```

I have not found a way to reproduce this reliably.
The code is probably:

```python
    @kernel
      def init(self):
          self.write32(_AD9910_REG_CFR1, 0x00000002)
          delay(100*ns)
          self.cpld.io_update.pulse(100*ns)
          aux_dac = self.read32(_AD9910_REG_AUX_DAC)
          if aux_dac & 0xff != 0x7f:
               raise ValueError("Urukul AD9910 AUX_DAC mismatch")
 ```

whitequark, 2018-01-04:
Hmm, that's very strange, I wonder if it's a race condition? Ideally we would not fork for symbolication...

sbourdeauducq, 2018-01-04:
Fork?

whitequark, 2018-01-04:
We execve addr2line to get the symbolized backtrace instead of a list of addresses.

commit 34a54458 (2018-01-08): compiler: don't die if addr2line cannot extract line from backtrace.

Fixes #885.
commit 24562d23 (2018-01-08): compiler: don't die if addr2line cannot extract line from backtrace.

Fixes #885.
closed 2018-01-08 by whitequark