m-labs/artiq #886: rust sdram init code breaks support for hardcoded write leveling

jordens, 2018-01-04:
The old SDRAM code supported Spartan 6 and Artiz 7 memory PHYs that have hardcoded write leveling through the DQS clocks. The new rust sdram init code breaks this (and therefore Kasli).

https://github.com/m-labs/misoc/blob/master/misoc/software/bios/sdram.c#L435
https://github.com/m-labs/artiq/blob/master/artiq/firmware/libboard/sdram.rs#L261

jordens, 2018-01-04:
New init code also broken when handling the Kasli Ethernet PHY.

```
(py35) rj@murray:~/src/artiq (master)$ python -m artiq.gateware.targets.kasli --no-comp
ile-gateware
make: Entering directory '/home/rj/src/artiq/misoc_opticlock_kasli/software/bootloader'
 CARGO    cargo/or1k-unknown-none/debug/libbootloader.a
   Compiling board v0.0.0 (file:///home/rj/src/artiq/artiq/firmware/libboard)
error[E0433]: failed to resolve. Could not find `ethphy` in `csr`
  --> /home/rj/src/artiq/artiq/firmware/libboard/ethmac.rs:52:18
   |
52 |             csr::ethphy::crg_reset_write(1);
   |                  ^^^^^^ Could not find `ethphy` in `csr`

error[E0433]: failed to resolve. Could not find `ethphy` in `csr`
  --> /home/rj/src/artiq/artiq/firmware/libboard/ethmac.rs:54:18
   |
54 |             csr::ethphy::crg_reset_write(0);
   |                  ^^^^^^ Could not find `ethphy` in `csr`

error[E0425]: cannot find function `wlevel_en_write` in module `ddrphy`
  --> /home/rj/src/artiq/artiq/firmware/libboard/sdram.rs:15:17
   |
15 |         ddrphy::wlevel_en_write(enabled as u8);
   |                 ^^^^^^^^^^^^^^^ not found in `ddrphy`

error[E0425]: cannot find function `wdly_dq_rst_write` in module `ddrphy`
  --> /home/rj/src/artiq/artiq/firmware/libboard/sdram.rs:51:21
   |
51 |             ddrphy::wdly_dq_rst_write(1);
   |                     ^^^^^^^^^^^^^^^^^ did you mean `rdly_dq_rst_write`?
[...]
```

whitequark, 2018-01-04:
> The old SDRAM code supported Spartan 6 and Artiz 7 memory PHYs that have hardcoded write leveling through the DQS clocks. The new rust sdram init code breaks this (and therefore Kasli).

For the record, when I have ported the SDRAM init code, I have taken great care to port every code path. The code you're referring to *postdates* the Rust port (as [discussed](https://github.com/m-labs/misoc/commit/afa38d46d5968aa4f3597910f5797071a2b9c7ef#commitcomment-26548954)).

jordens, 2018-01-04:
Sure. Until the new code (in ARTIQ) replaces the old one (still current in MiSoC), forking means twice the work.

whitequark, 2018-01-09:
Fixed in 3bfb128a99909c932c9bf46a466c9651ce7d1284.

commit 267c6998 (2018-01-09): firmware: support DDRPHY without write leveling (fixes #886).

Currently, this is kasli.
closed 2018-01-09 by whitequark