m-labs/artiq #888: Installing artiq 3.1 from source fails during rust install

philipkent, 2018-01-10:
On Ubuntu 14.04.5 LTS 64 bit, trying to install ARTIQ 3.1 fails during the rust install with: 
I'm following the instructions on https://m-labs.hk/artiq/manual-release-3/developing.html#installing-artiq-from-source installation fails while trying to 'make install' rust.

There errors

whitequark, 2018-01-10:
Which errors do you get exactly?

philipkent, 2018-01-10:
cargo:warning=cc: error: ../../libcompiler_builtins/compiler-rt/lib/builtins/absvdi2.c: No such file or directory

I've searched for that file with find and I'm not seeing it, but maybe I'm missing something.

philipkent, 2018-01-10:
I do have three 'libcompiler_bultins' directories but no 'absvdi2.c' file:

    find . -name libcompiler_builtins
    ./rust/build/build/x86_64-unknown-linux-gnu/stage0-std/x86_64-unknown-linux-gnu/release/build/libcompiler_builtins
    ./rust/src/libcompiler_builtins
    ./rust/.git/modules/src/libcompiler_builtins

whitequark, 2018-01-10:
You forgot to run `git submodule update --init` in the Rust sources.

philipkent, 2018-01-10:
hmm, I'll try it again but I did update the rust submodules.

philipkent, 2018-01-10:
I tried a second time but I still get the same error.  git submodule update --init seemed to go smoothly.

This is everything I did.
```
maglab@687stylab:~/artiq-dev/artiq_3.1$ rm -fr rust
maglab@687stylab:~/artiq-dev/artiq_3.1$ git clone -b artiq-1.22.1 https://github.com/m-labs/rust
Cloning into 'rust'...
remote: Counting objects: 709966, done.
remote: Total 709966 (delta 0), reused 0 (delta 0), pack-reused 709965
Receiving objects: 100% (709966/709966), 323.81 MiB | 13.38 MiB/s, done.
Resolving deltas: 100% (573640/573640), done.
Checking connectivity... done.
maglab@687stylab:~/artiq-dev/artiq_3.1$ cd rust
maglab@687stylab:~/artiq-dev/artiq_3.1/rust$ git submodule update --init
Submodule 'book' (https://github.com/rust-lang/book.git) registered for path 'src/doc/book'
Submodule 'src/doc/nomicon' (https://github.com/rust-lang-nursery/nomicon.git) registered for path 'src/doc/nomicon'
Submodule 'reference' (https://github.com/rust-lang-nursery/reference.git) registered for path 'src/doc/reference'
Submodule 'src/jemalloc' (https://github.com/rust-lang/jemalloc.git) registered for path 'src/jemalloc'
Submodule 'src/libcompiler_builtins' (https://github.com/rust-lang-nursery/compiler-builtins) registered for path 'src/libcompiler_builtins'
Submodule 'src/liblibc' (https://github.com/rust-lang/libc.git) registered for path 'src/liblibc'
Submodule 'src/rt/hoedown' (https://github.com/rust-lang/hoedown.git) registered for path 'src/rt/hoedown'
Submodule 'src/tools/cargo' (https://github.com/rust-lang/cargo.git) registered for path 'src/tools/cargo'
Submodule 'src/tools/clippy' (https://github.com/rust-lang-nursery/rust-clippy.git) registered for path 'src/tools/clippy'
Submodule 'src/tools/miri' (https://github.com/solson/miri.git) registered for path 'src/tools/miri'
Submodule 'src/tools/rls' (https://github.com/rust-lang-nursery/rls.git) registered for path 'src/tools/rls'
Submodule 'src/rust-installer' (https://github.com/rust-lang/rust-installer.git) registered for path 'src/tools/rust-installer'
Submodule 'src/tools/rustfmt' (https://github.com/rust-lang-nursery/rustfmt.git) registered for path 'src/tools/rustfmt'
Cloning into 'src/doc/book'...
remote: Counting objects: 19001, done.
remote: Compressing objects: 100% (50/50), done.
remote: Total 19001 (delta 57), reused 59 (delta 38), pack-reused 18913
Receiving objects: 100% (19001/19001), 22.16 MiB | 9.05 MiB/s, done.
Resolving deltas: 100% (14835/14835), done.
Checking connectivity... done.
Submodule path 'src/doc/book': checked out '7db393dae740d84775b73f403123c866e94e3a5b'
Cloning into 'src/doc/nomicon'...
remote: Counting objects: 1074, done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 1074 (delta 4), reused 3 (delta 0), pack-reused 1060
Receiving objects: 100% (1074/1074), 918.69 KiB | 0 bytes/s, done.
Resolving deltas: 100% (705/705), done.
Checking connectivity... done.
Submodule path 'src/doc/nomicon': checked out '0ee3f7265e9d09746d901cef6f1f300baff1d923'
Cloning into 'src/doc/reference'...
remote: Counting objects: 2868, done.
remote: Total 2868 (delta 0), reused 0 (delta 0), pack-reused 2868
Receiving objects: 100% (2868/2868), 1.98 MiB | 0 bytes/s, done.
Resolving deltas: 100% (1980/1980), done.
Checking connectivity... done.
Submodule path 'src/doc/reference': checked out '36adc6ae504c6e0343ab5d7b3871f0a2a71236da'
Cloning into 'src/jemalloc'...
remote: Counting objects: 16722, done.
remote: Total 16722 (delta 0), reused 0 (delta 0), pack-reused 16722
Receiving objects: 100% (16722/16722), 7.39 MiB | 7.43 MiB/s, done.
Resolving deltas: 100% (11932/11932), done.
Checking connectivity... done.
Submodule path 'src/jemalloc': checked out '1f5a28755e301ac581e2048011e4e0ff3da482ef'
Cloning into 'src/libcompiler_builtins'...
remote: Counting objects: 2564, done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 2564 (delta 11), reused 21 (delta 7), pack-reused 2537
Receiving objects: 100% (2564/2564), 459.16 KiB | 0 bytes/s, done.
Resolving deltas: 100% (1597/1597), done.
Checking connectivity... done.
Submodule path 'src/libcompiler_builtins': checked out '0b9844764ea1f99ea11a7917a4f3ba7fd2db775c'
Cloning into 'src/liblibc'...
remote: Counting objects: 39963, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 39963 (delta 2), reused 9 (delta 0), pack-reused 39948
Receiving objects: 100% (39963/39963), 15.90 MiB | 12.11 MiB/s, done.
[... 201 lines cut ...]
CFLAGS_x86_64-unknown-linux-gnu = Some("-ffunction-sections -fdata-sections -fPIC -m64")
DEBUG = Some("false")
running: "cc" "-O2" "-ffunction-sections" "-fdata-sections" "-fPIC" "-ffunction-sections" "-fdata-sections" "-fPIC" "-m64" "-m64" "-fno-builtin" "-fvisibility=hidden" "-fomit-frame-pointer" "-ffreestanding" "-DVISIBILITY_HIDDEN" "-o" "/home/maglab/artiq-dev/artiq_3.1/rust/build/build/x86_64-unknown-linux-gnu/stage0-std/x86_64-unknown-linux-gnu/release/build/compiler_builtins-7bd5df6f54dc2a21/out/../../libcompiler_builtins/compiler-rt/lib/builtins/absvdi2.o" "-c" "../../libcompiler_builtins/compiler-rt/lib/builtins/absvdi2.c"
cargo:warning=cc: error: ../../libcompiler_builtins/compiler-rt/lib/builtins/absvdi2.c: No such file or directory
cargo:warning=cc: fatal error: no input files
cargo:warning=compilation terminated.
exit code: 4

--- stderr
thread 'main' panicked at '

[... cut ...] u" "--release" "--features" "panic-unwind jemalloc backtrace" "--manifest-path" "/home/maglab/artiq-dev/artiq_3.1/rust/src/libstd/Cargo.toml" "--message-format" "json"
expected success, got: exit code: 101', /home/maglab/artiq-dev/artiq_3.1/rust/src/bootstrap/compile.rs:883:8
note: Run with `RUST_BACKTRACE=1` for a backtrace.
failed to run: /home/maglab/artiq-dev/artiq_3.1/rust/build/build/bootstrap/debug/bootstrap install
Build completed unsuccessfully in 0:01:49
make: *** [install] Error 1
```

whitequark, 2018-01-10:
Oh! `git submodule update --init --recursive`.

philipkent, 2018-01-10:
Ah!  Will try that, thanks.

philipkent, 2018-01-10:
That fixed the 'absvdi2.c' error, thanks!.  Now it's failing for not being able to find llvmconfig.  Is that supposed to be installed as apart of the LLVM and Clang build?  There were a few directory permission issues on this machine earlier so it may be that llvmconfig is just not visible for my user.  I'm going to look into that more tomorrow.

whitequark, 2018-01-10:
> Is that supposed to be installed as apart of the LLVM and Clang build?

The `llvm-config` executable should be installed as a part of LLVM. Is it not present as `/usr/local/llvm-or1k/bin/llvm-config`?

philipkent, 2018-01-11:
It was present.  `llvm-config` did get installed to `/usr/local/llvm-or1k/bin`, it was just that `/usr/local/llvm-or1k/bin` wasn't on the system path.  After adding `/usr/local/llvm-or1k/bin` to my path variable `make install` ran without errors and I get the "Rust ready to roll" message at the end.

sbourdeauducq, 2018-01-11:
> trying to install ARTIQ 3.1

Any particular reason you're not using 3.2?

whitequark, 2018-01-11:
> After adding /usr/local/llvm-or1k/bin to my path variable make install ran without errors and I get the "Rust ready to roll" message at the end.

That's very odd, it should not be necessary. Exactly which component is not finding llvm-config? It would be helpful if you posted the actual error message in its context instead of paraphrasing it.

philipkent, 2018-01-11:
@sbourdeauducq 
> Any particular reason you're not using 3.2? Or the conda packages?

We need to build a modified version of the gateware with the output FIFO depth increased to 1024.  We did this for ARTIQ version 2 and it reduced the number of underflow errors we we're getting.  We tried running a few experiments using the gateware provided in the ARTIQ version 3 conda packages but that resulted in underflow errors that did not occur using the version 2.0 gateware with the FIFO output depth increased to 1024.  

I just noticed the mention of the `artiq-dev` package in the [instructions for version 3](https://m-labs.hk/artiq/manual-release-3/developing.html#installing-artiq-from-source)  I didn't realize those were available, would that be enough to build the gateware with an updated FIFO depth?

philipkent, 2018-01-11:
@whitequark 
`That's very odd, it should not be necessary. Exactly which component is not finding llvm-config? It would be helpful if you posted the actual error message in its context instead of paraphrasing it.`

I'll try running through the install process again tonight and post the specific error.  The [steps for installing from source ](https://m-labs.hk/artiq/manual-release-3/developing.html#installing-artiq-from-source) instruct you to add /usr/local/llvm-or1k/bin to your path with:

`export PATH=/usr/local/llvm-or1k/bin:$PATH`

_after_ installing rust.  Maybe that step just needs to be mentioned higher up in the instructions before the steps to build rust?

philipkent, 2018-01-12:
@whitequark This is the error I was getting when trying to install rust.

    maglab@687stylab:~/artiq-dev/artiq_3.1/rust/build$ make install
    Finished dev [unoptimized] target(s) in 0.0 secs
    Building stage0 std artifacts (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/divdc3.c:21:1: warning: conflicting types for built-in function â€˜__divdc3â€™ [enabled by default]
    warning:  __divdc3(double __a, double __b, double __c, double __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/divsc3.c:21:1: warning: conflicting types for built-in function â€˜__divsc3â€™ [enabled by default]
    warning:  __divsc3(float __a, float __b, float __c, float __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/divxc3.c:22:1: warning: conflicting types for built-in function â€˜__divxc3â€™ [enabled by default]
    warning:  __divxc3(long double __a, long double __b, long double __c, long double __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/muldc3.c:21:1: warning: conflicting types for built-in function â€˜__muldc3â€™ [enabled by default]
    warning:  __muldc3(double __a, double __b, double __c, double __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/mulsc3.c:21:1: warning: conflicting types for built-in function â€˜__mulsc3â€™ [enabled by default]
    warning:  __mulsc3(float __a, float __b, float __c, float __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/mulxc3.c:23:1: warning: conflicting types for built-in function â€˜__mulxc3â€™ [enabled by default]
    warning:  __mulxc3(long double __a, long double __b, long double __c, long double __d)
    warning:  ^
    Finished release [optimized] target(s) in 0.0 secs
    Copying stage0 std from stage0 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
    Building stage0 test artifacts (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
    Finished release [optimized] target(s) in 0.0 secs
    Copying stage0 test from stage0 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
    Building stage0 compiler artifacts (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
    Finished release [optimized] target(s) in 0.0 secs
    Copying stage0 rustc from stage0 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
    Assembling stage1 compiler (x86_64-unknown-linux-gnu)
    Building stage1 std artifacts (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/divdc3.c:21:1: warning: conflicting types for built-in function â€˜__divdc3â€™ [enabled by default]
    warning:  __divdc3(double __a, double __b, double __c, double __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/divsc3.c:21:1: warning: conflicting types for built-in function â€˜__divsc3â€™ [enabled by default]
    warning:  __divsc3(float __a, float __b, float __c, float __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/divxc3.c:22:1: warning: conflicting types for built-in function â€˜__divxc3â€™ [enabled by default]
    warning:  __divxc3(long double __a, long double __b, long double __c, long double __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/muldc3.c:21:1: warning: conflicting types for built-in function â€˜__muldc3â€™ [enabled by default]
    warning:  __muldc3(double __a, double __b, double __c, double __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/mulsc3.c:21:1: warning: conflicting types for built-in function â€˜__mulsc3â€™ [enabled by default]
    warning:  __mulsc3(float __a, float __b, float __c, float __d)
    warning:  ^
    warning: ../../libcompiler_builtins/compiler-rt/lib/builtins/mulxc3.c:23:1: warning: conflicting types for built-in function â€˜__mulxc3â€™ [enabled by default]
    warning:  __mulxc3(long double __a, long double __b, long double __c, long double __d)
    warning:  ^
        Finished release [optimized] target(s) in 0.0 secs
    Copying stage1 std from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
    Building stage1 test artifacts (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
        Finished release [optimized] target(s) in 0.0 secs
    Copying stage1 test from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
    Building stage1 compiler artifacts (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
        Finished release [optimized] target(s) in 0.0 secs
    Copying stage1 rustc from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
    Assembling stage2 compiler (x86_64-unknown-linux-gnu)
    Dist std stage2 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
    Uplifting stage1 std (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
    Copying stage2 std from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
    Uplifting stage1 test (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
    Copying stage2 test from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
[... cut ...]
    command did not execute successfully: "/home/maglab/artiq-dev/artiq_3.1/rust/build/build/x86_64-unknown-linux-gnu/stage0/bin/cargo" "build" "--target" "x86_64-unknown-linux-gnu" "--release" "--manifest-path" "/home/maglab/artiq-dev/artiq_3.1/rust/src/tools/rustdoc/Cargo.toml"
    expected success, got: exit code: 101
    
    
    failed to run: /home/maglab/artiq-dev/artiq_3.1/rust/build/build/bootstrap/debug/bootstrap install
    Build completed unsuccessfully in 0:01:42
    make: *** [install] Error 1

sbourdeauducq, 2018-01-12:
> I just noticed the mention of the artiq-dev package in the instructions for version 3 I didn't realize those were available, would that be enough to build the gateware with an updated FIFO depth?

Yes. And note that you do not need Rust if you are only making such a minor change to the gateware. You can skip the whole firmware compilation with ``--no-compile-software`` - but be careful, mismatches you introduce at the firmware/gateware interface can result in somewhat obscure bugs.

sbourdeauducq, 2018-01-12:
> We need to build a modified version of the gateware with the output FIFO depth increased to 1024. We did this for ARTIQ version 2 and it reduced the number of underflow errors we we're getting. We [...]

Which exactly is that FIFO you need expanding? Since you are the main users of those packages, we can also build them with the FIFO sizes you want.

jordens, 2018-01-12:
And the documentation also clearly states that that approach to installing from source is irrelevant to you. Just use the conda packages pulled in by artiq-dev.

whitequark, 2018-01-12:
Okay, this is a known rustc bug https://github.com/rust-lang/rust/issues/46995, I've updated our workaround for it in our rustc fork so you should no longer hit it.

closed 2018-01-12 by whitequark

dhslichter, 2018-01-23:
> Which exactly is that FIFO you need expanding? Since you are the main users of those packages, we can also build them with the FIFO sizes you want.

I have been very delayed on sending in an updated gateware description file for NIST QC 2, but we are going to expand the number of TTL channels in hardware and can make modifications to the FIFO depths at the same time. @philipkent is there something in particular you are looking for?  What are you doing so rapidly that you need such deep output FIFOs?

philipkent, 2018-01-29:
@dhslichter we needed to expanded the FIFO depth to prevent underflow errors from occurring during the Raman cooling sequence.  I'm working on more thorough tests using ARTIQ 3.1 and DMA but the expanded FIFO depth still seemed to be necessary on 3.1 after a first test, even when we used recorded DMA sequences for Raman cooling.  The FIFO depth of 1024 was the first modification that we made while on ARTIQ version 2 and we haven't changed that depth since.  A depth less than 1024 would probably be fine for our purposes, but we would need to test to find a reasonable lower limit.

dhslichter, 2018-01-30:
@philipkent ack.  We have had similar issues with our sideband cooling and went to deeper output FIFOs as a result.  Let's discuss offline and run some tests to come up with a good resource allocation.

philipkent, 2018-06-25:
@dhslichter We will be moving to ARTIQ 3 soon and we're planning on having to setup a development environment to compile gatewares with the expanded FIFO depths.  Would it make sense to update the FIFO depth's in the pre-compiled gatewares in the nist_qc2 conda package for ARTIQ 3 to be at least 1024 sense that seems to be what everyone is using?

dhslichter, 2018-06-25:
Yes, I think that sounds sensible.  There are separate input and output FIFOs, and I would suggest that we would make better use of the resources available if we make the input FIFOs relatively small (say 128 or 256) except on a few special channels, when we make them much deeper (16k or so).  Then we can hopefully fit 1024 on each output, which would be nice.  @r-srinivas @dtcallcock any strong feelings on optimal FIFO depths to shoot for?

r-srinivas, 2018-06-25:
I think 1024 is what we have now and that seems to work for most of our
experiments. This is what I have from our notes from almost 2 years ago.

Ofifo_depth for 39 ttl channels

block_ram_tile_util_percentage (%)

ramb36_fifo_util_percentage

(%)

ramb18_util_percentage (%)

64 (default)

34.38

23.82

10.56

1024

43.82

33.26

10.56

2048

61.35

50.79

10.56

4096

92.02

77.08

14.94

r-srinivas, 2018-06-25:
I think 1024 is what we have now and that seems to work for most of our
experiments. This is what I have from our notes from almost 2 years ago.

| ofifo depth for 39 ttl Channels | block_ram_tile_util_percentage (%) | ramb36_fifo_util_percentage (%) | ramb18_util_percentage (%) |
|---------------------------------|------------------------------------|---------------------------------|----------------------------|
| 64                              | 34.38                              | 23.82                           | 10.56                      |
| 1024                            | 43.82                              | 33.26                           | 10.56                      |
| 2048                            | 61.35                              | 50.79                           | 10.56                      |
| 4096                            | 92.02                              | 77.08                           | 14.94                      |