m-labs/artiq #890: naming of sayma classes don't follow artiq_flash convention

jbqubit, 2018-01-12:
```
$ find /home/britton/anaconda3/envs/artiq-dev/lib/python3.5/ -name "*.bit"
binaries/kc705-nist_clock/top.bit
binaries/sayma_amc-standalone/top.bit
binaries/sayma_rtm/top.bit
```
This naming scheme for the conda-distributed .bit breaks artiq_flash which assumes a hyphen... https://github.com/m-labs/artiq/blob/529033e016fffc69c20895348e037370faf78d20/artiq/frontend/artiq_flash.py#L219

Also, the name of the path is ```amc-standalone``` not ```standalone``` as in artiq_flash.py https://github.com/m-labs/artiq/blob/529033e016fffc69c20895348e037370faf78d20/artiq/frontend/artiq_flash.py#L204

sbourdeauducq, 2018-01-13:
> This naming scheme for the conda-distributed .bit breaks artiq_flash which assumes a hyphen... 
> Also, the name of the path is amc-standalone not standalone as in artiq_flash.py 

That's not what the problem is, but yes, there is a discrepancy in artiq_flash for the sayma names and paths.

hartytp, 2018-01-14:
Yes, I ran into this too. I have a PR prepared for this, just needs checking.

sbourdeauducq, 2018-01-14:
``artiq_flash`` will need a bit of refactoring to properly support Sayma and its RTM FPGA (we plan to put the bitstream in another flash partition like the runtime).

sbourdeauducq, 2018-01-17:
Fixed in https://github.com/m-labs/artiq/commit/f77aa9b78ff7d1a61417650cfd65dbc4635cca0f.

closed 2018-01-17 by sbourdeauducq