m-labs/artiq #893: artiq_session windows

TheCakeIsAPi, 2018-01-16:
Using Windows 10 artiq 3.2
```
>artiq_session
Traceback (most recent call last):
  File "C:\Users\monroe\Miniconda3\envs\artiq-dev\Scripts\artiq_session-script.py", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_session')()
  File "C:\Users\monroe\Miniconda3\envs\artiq-dev\lib\site-packages\pkg_resources\__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "C:\Users\monroe\Miniconda3\envs\artiq-dev\lib\site-packages\pkg_resources\__init__.py", line 2630, in load_entry_point
    raise ImportError("Entry point %r not found" % ((group, name),))
ImportError: Entry point ('console_scripts', 'artiq_session') not found
```

jordens, 2018-01-16:
Could you add the output of `conda list`?

TheCakeIsAPi, 2018-01-17:
Output of ```conda list``` in question.  Did I somehow end up with artiq 2.5?  I used "git checkout 3.2" on the cloned repository when creating my environment.

```
(artiq-dev) C:\Users\monroe>conda list
# packages in environment at C:\Users\monroe\Miniconda3\envs\artiq-dev:
#
aiohttp                   2.3.3            py35h22b64ad_0
alabaster                 0.7.10           py35h3a808de_0
artiq                     2.5             py_1+git44c78a5e    m-labs/label/dev
artiq-dev                 2.5             py_1+git44c78a5e    m-labs/label/dev
artiq-kc705-nist_clock    2.5             py_1+git44c78a5e    m-labs/label/dev
async-timeout             2.0.0            py35h1948f5c_0
asyncserial               0.1             py_13+git340e430    m-labs/label/main
babel                     2.5.1                    py35_0
binutils-or1k-linux       2.25.1                        4    m-labs/label/main
ca-certificates           2017.08.26           h94faf87_0
certifi                   2017.11.5        py35h456c6ae_0
cffi                      1.11.2           py35h74ad5ef_0
chardet                   3.0.4            py35h177e1b7_1
colorama                  0.3.9            py35h32a752f_0
coverage                  4.4.2            py35h1d76e2e_0
curl                      7.49.1                   vc14_2  [vc14]  conda-forge/label/main
dateutil                  2.4.2                      py_2    m-labs/label/main
docutils                  0.14             py35h8ccb97f_0
h5py                      2.7.1            py35hb2c3add_0
hdf5                      1.10.1                   vc14_1  [vc14]  conda-forge/label/main
icc_rt                    2017.0.4             h97af966_0
icu                       58.2                     vc14_0  [vc14]  conda-forge/label/main
imagesize                 0.7.1            py35h7f87e83_0
intel-openmp              2018.0.0             hd92c6cd_8
jinja2                    2.10             py35hdf652bb_0
jpeg                      9b                       vc14_2  [vc14]  conda-forge/label/main
levenshtein               0.12.0                   py35_1    m-labs/label/main
libgit2                   0.25.0                   vc14_0  [vc14]  conda-forge/label/main
libpng                    1.6.34                   vc14_0  [vc14]  conda-forge/label/main
libusb                    1.0.20                        0    m-labs/label/main
lit                       0.4.1                      py_9    m-labs/label/main
llvm-or1k                 4.0.1               hec1df13_23    m-labs/label/main
llvmlite-artiq            0.10.0.dev              py35_24    m-labs/label/main
markupsafe                1.0              py35hc253e08_1
migen                     0.4             py_1+gitb730caf    m-labs/label/main
migen                     0.4                       <pip>
misoc                     0.1                       <pip>
misoc                     0.3             py_0+git504775c    m-labs/label/main
mkl                       2018.0.1             h2108138_4
multidict                 3.3.2            py35ha61a431_0
numpy                     1.14.0           py35h4a99626_0
openocd                   0.10.0                        1    m-labs/label/main
openssl                   1.0.2n                   vc14_0  [vc14]  conda-forge/label/main
outputcheck               0.4.2                      py_7    m-labs/label/main
pip                       9.0.1            py35h691316f_4
prettytable               0.7.2                    py35_1    conda-forge/label/main
pycparser                 2.18             py35h15a15da_1
pygit2                    0.25.0              py35_vc14_0  [vc14]  conda-forge/label/main
pygments                  2.2.0            py35h24c0941_0
pyqt                      5.6.0            py35hd46907b_5
pyqtgraph                 0.10.0                   py35_2    conda-forge/label/main
pyqtgraph-pyqtgraph-0.9.10-115 g71bd3e7                  <pip>
pyserial                  3.4              py35he0af20a_0
python                    3.5.4               h1357f44_23
pythonparser              1.0                        py_8    m-labs/label/main
pytz                      2017.3           py35hb294b1a_0
qt                        5.6.2                    vc14_1  [vc14]  conda-forge/label/main
quamash                   0.5.5                      py_4    m-labs/label/main
regex                     2015.11.22               py35_1    m-labs/label/main
scipy                     1.0.0            py35h75710e8_0
setuptools                33.1.1                   py35_0    conda-forge/label/main
sip                       4.18.1           py35h01cbaa7_2
six                       1.11.0           py35hc1da2df_1
snowballstemmer           1.2.1            py35h4c55bfa_0
sphinx                    1.4.8                    py35_0
sphinx-argparse           0.1.13                     py_4    m-labs/label/main
sphinx_rtd_theme          0.2.4                    py35_0
sphinxcontrib-wavedrom    1.1.0                     <pip>
sphinxcontrib-wavedrom    1.1.0                      py_1    m-labs/label/main
vc                        14                   h0510ff6_3
vs2015_runtime            14.0.25123                    3
wheel                     0.30.0           py35h38a90bc_1
wincertstore              0.2              py35hfebbdb8_0
yarl                      0.14.2           py35h9136884_0
zlib                      1.2.11                   vc14_0  [vc14]  conda-forge/label/main
```

whitequark, 2018-01-17:
Conda installed packages and your git checkout are completely independent of each other. Also, artiq-dev 3+ is not available on Windows because we do not build the Rust components for it.

What are you trying to do exactly that you install `artiq-dev`?

TheCakeIsAPi, 2018-01-17:
Was trying to get around the artiq_flash issue #894 with the help of jbqubit by installing from source.

whitequark, 2018-01-17:
Just specify the binaries directory explicitly.

jbqubit, 2018-01-17:
Explicitly giving path to binaries directory didn't work either.

closed 2018-01-18 by whitequark