m-labs/artiq #894: binaries in wrong directory for artiq_flash

TheCakeIsAPi, 2018-01-16:
On Windows 10, Artiq 3.2:

```
artiq_flash -t kc705 -m nist_clock
Binaries directory 'C:\Users\monroe\Miniconda3\envs\artiq-dev\lib\site-packages\artiq\binaries\kc705-nist_clock' does not exist
```

The directory 'C:\Users\monroe\Miniconda3\envs\artiq-dev\lib\site-packages\artiq\binaries' does not exist at all.  Instead the files are in 'C:\Users\monroe\Miniconda3\envs\artiq-dev\Lib\python3.5\site-packages\artiq\binaries\kc705-nist_clock' and the command runs successfully as:

```
artiq_flash -t kc705 -m nist_clock -d C:\Users\monroe\Miniconda3\envs\artiq-dev\Lib\python3.5\site-packages\artiq\binaries\kc705-nist_clock
```

sbourdeauducq, 2018-01-17:
I don't see how this can happen and I cannot reproduce it. Are you sure the command you ran is just ``artiq_flash -t kc705 -m nist_clock``?

As you can see from the ``artiq_flash`` code, this is supposed to append ``kc705-nist_clock`` to the path:
https://github.com/m-labs/artiq/blob/release-3/artiq/frontend/artiq_flash.py#L77-L79

Can you add ``print`` statements in ``artiq_flash`` (e.g. to check that the body of the ``if opts.dir is None`` test is executed) and investigate this?

sbourdeauducq, 2018-01-17:
@TheCakeIsAPi What happened is conda installed ``artiq`` in ``C:\Users\monroe\Miniconda3\envs\artiq-dev\lib\site-packages\artiq`` and ``artiq-kc705-nist_clock`` a different directory, in ``C:\Users\monroe\Miniconda3\envs\artiq-dev\Lib\python3.5\site-packages\artiq``.

Any idea why?

whitequark, 2018-01-17:
@sbourdeauducq probably because artiq is `noarch: python` and artiq-$BOARD is `noarch: generic` in meta.yaml.

closed 2018-01-17 by whitequark

TheCakeIsAPi, 2018-01-17:
Not sure why.  I would say the bug is in the artiq-kc705-nist_clock package, because it is the only thing at all in ```Lib\python3.5```  Everything else is in ```Lib\site-packages```

joeshardow, 2018-01-26:
[image]
[image]
@TheCakeIsAPi I move the binary file into the conda searching location, but it is get a new error, see in the picture in attachment.  

The command and error is listed in the attachment picture.  The mainly error I suppose is no device found. I use Vivado driver for KC705 board( But the Vivado can flash binary file into KC705 correctly), The operation system is Win10 x64.  The anaconda version is 64x, and degrade to python 3.5.4

joeshardow, 2018-01-26:
Or I can use Vivado to flash the binary file into KC705?

sbourdeauducq, 2018-01-26:
> it is get a new error,

So, please open a new issue for it.

joeshardow, 2018-01-26:
I have figure out this problem. After replacing the USB driver with Zadig, that's work correctly.

sbourdeauducq, 2018-01-26:
This is explained in the ARTIQ manual:
http://m-labs.hk/artiq/manual-master/installing.html?highlight=zadig#configuring-openocd

commit 5c6276c7 (2018-01-17): conda: build artiq-$BOARD as noarch: python, like artiq itself.

Fixes #894.