m-labs/artiq #896: artiq 3+ not supported on windows

jbqubit, 2018-01-17:
As per #893 m-labs says they are no longer supporting Windows. I'm fine with that. @dhslichter? If Windows support is now deprecated please update the documentation indicating this.

http://m-labs.hk/artiq/manual-release-3/installing.html#installing-artiq

sbourdeauducq, 2018-01-17:
artiq != artiq-dev. Windows is still supported, what is not is conda packages that let you recompile the runtime etc. on Windows.

closed 2018-01-17 by sbourdeauducq

jbqubit, 2018-01-17:
OK. Please note that here

http://m-labs.hk/artiq/manual-release-3/developing.html#developing-artiq

dhslichter, 2018-01-17:
We still definitely require Windows support.

dhslichter, 2018-01-18:
@jbqubit we do not intend to do our own ARTIQ builds on Windows (can use a Linux box if this is really necessary), but we do require the ability to run ARTIQ on Windows.  This means maintaining CI for conda packages for Windows.

commit bdf41cd5 (2018-01-18): doc: mention that ARTIQ can only be developed on Linux. (fixes #896).