m-labs/artiq #898: bscan-spi-bitstreams not included in artiq-dev.yaml

jbqubit, 2018-01-19:
From https://github.com/m-labs/artiq/issues/856#issuecomment-358864414

```
$ artiq_flash --srcbuild ./misoc_standalone_sayma_amc -t sayma_rtm
proxy gateware bitstream bscan_spi_xcku040-sayma.bit not found
```

```
$ find /home/britton/anaconda3/envs/artiq-dev2 -name "*xcku040-sayma.bit"
(artiq-dev2) britton@britton1:~/artiq-dev2/artiq
$ 
```
>That's a different issue than serwb. Did you install our proxy bitstreams conda package?
>https://anaconda.org/m-labs/bscan-spi-bitstreams

I follow [these](http://m-labs.hk/artiq/manual-release-3/developing.html#artiq-anaconda-development-environment) instructions verbatim to create a new conda environment for every pull from master. Proxy bitstreams conda package is missing.

sbourdeauducq, 2018-01-19:
> I follow these instructions verbatim to create a new conda environment for every pull from master. Proxy bitstreams conda package is missing.

OK and does installing ``bscan-spi-bitstreams`` manually solve the problem?

jbqubit, 2018-01-19:
```
$ conda install bscan-spi-bitstreams
$
$ artiq_flash --srcbuild ./misoc_standalone_sayma_amc -t sayma_amcDesign: top;COMPRESS=TRUE;UserID=0XFFFFFFFF;Version=2016.2
Part name: xcku040-ffva1156-1-c
Date: 2018/01/18
Time: 18:37:01
Bitstream payload length: 0x77fea4
Open On-Chip Debugger 0.10.0 (2017-02-03-06:53)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Can't find cpld/xilinx-xcu.cfg
Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-dev2/bin/artiq_flash", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_flash')()
  File "/home/britton/artiq-dev2/artiq/artiq/frontend/artiq_flash.py", line 304, in main
    programmer.do()
  File "/home/britton/artiq-dev2/artiq/artiq/frontend/artiq_flash.py", line 106, in do
    subprocess.check_call(cmdline)
  File "/home/britton/anaconda3/envs/artiq-dev2/lib/python3.5/subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openocd', '-s', '/home/britton/anaconda3/envs/artiq-dev2/share/openocd/scripts', '-c', 'interface ftdi; ftdi_device_desc "Quad RS232-HS"; ftdi_vid_pid 0x0403 0x6011; ftdi_channel 0; ftdi_layout_init 0x0098 0x008b; reset_config none; adapter_khz 5000; transport select jtag; source [find cpld/xilinx-xc7.cfg]; set CHIP XCKU040; source [find cpld/xilinx-xcu.cfg]; target create xcu.proxy testee -chain-position xcu.tap; set XILINX_USER1 0x02; set XILINX_USER2 0x03; flash bank xcu.spi0 jtagspi 0 0 0 0 xcu.proxy $XILINX_USER1; flash bank xcu.spi1 jtagspi 0 0 0 0 xcu.proxy $XILINX_USER2; init; pld load 1 {/home/britton/anaconda3/envs/artiq-dev2/share/bscan-spi-bitstreams/bscan_spi_xcku040-sayma.bit}; reset halt; flash probe xcu.spi0; flash erase_sector 0 0 119; flash write_bank 0 {/tmp/tmpwk1b9z5m} 0x0; flash verify_bank 0 {/tmp/tmpwk1b9z5m} 0x0; flash probe xcu.spi1; flash erase_sector 1 0 1; flash write_bank 1 {./misoc_standalone_sayma_amc/software/bootloader/bootloader.bin} 0x0; flash verify_bank 1 {./misoc_standalone_sayma_amc/software/bootloader/bootloader.bin} 0x0; flash probe xcu.spi1; flash erase_sector 1 5 13; flash write_bank 1 {./misoc_standalone_sayma_amc/software/runtime/runtime.fbi} 0x50000; flash verify_bank 1 {./misoc_standalone_sayma_amc/software/runtime/runtime.fbi} 0x50000; xcu_program xcu.tap; exit']' returned non-zero exit status 1

$ find /home/britton/anaconda3/envs/artiq-dev2/ -name "xilinx-xcu.cfg"
$ 

```

sbourdeauducq, 2018-01-19:
Ah, yes. @enjoy-digital also saw that one but did not report it yet. #899

sbourdeauducq, 2018-01-19:
As for ``bscan-spi-bitstreams``, it is debatable whether it should be included in ``artiq-dev``, the board-specific packages, or ``openocd``. I'll let @jordens make that decision.

whitequark, 2018-01-19:
@jbqubit What is the output of `conda info`?

jordens, 2018-01-19:
That's an old openocd. The new one depends on the proxy bitstreams. whitequark needs to retighten the dependency.
#892

closed 2018-01-19 by jordens

jbqubit, 2018-01-19:
```
$ conda info
Current conda install:

               platform : linux-64
          conda version : 4.3.30
       conda is private : False
      conda-env version : 4.3.30
    conda-build version : 3.0.6
         python version : 3.5.4.final.0
       requests version : 2.18.1
       root environment : /home/britton/anaconda3  (writable)
    default environment : /home/britton/anaconda3/envs/artiq-dev2
       envs directories : /home/britton/anaconda3/envs
                          /home/britton/.conda/envs
          package cache : /home/britton/anaconda3/pkgs
                          /home/britton/.conda/pkgs
           channel URLs : https://repo.continuum.io/pkgs/main/linux-64
                          https://repo.continuum.io/pkgs/main/noarch
                          https://repo.continuum.io/pkgs/free/linux-64
                          https://repo.continuum.io/pkgs/free/noarch
                          https://repo.continuum.io/pkgs/r/linux-64
                          https://repo.continuum.io/pkgs/r/noarch
                          https://repo.continuum.io/pkgs/pro/linux-64
                          https://repo.continuum.io/pkgs/pro/noarch
                          http://conda.anaconda.org/conda-forge/label/main/linux-64
                          http://conda.anaconda.org/conda-forge/label/main/noarch
                          http://conda.anaconda.org/m-labs/label/dev/linux-64
                          http://conda.anaconda.org/m-labs/label/dev/noarch
                          http://conda.anaconda.org/m-labs/label/main/linux-64
                          http://conda.anaconda.org/m-labs/label/main/noarch
            config file : /home/britton/.condarc
             netrc file : None
           offline mode : False
             user-agent : conda/4.3.30 requests/2.18.1 CPython/3.5.4 Linux/3.19.0-80-generic debian/jessie/sid glibc/2.19    
                UID:GID : 1001:1001
```
So what should I do to test that this fixes artiq_flash?

jbqubit, 2018-01-19:
I followed these instructions this afternoon to create a new conda environment. It pulled in the new openocd in the process. 

```
~/artiq-dev2/artiq/artiq$ artiq_flash --srcbuild ./misoc_standalone_sayma_amc -t sayma_amc
Design: top;COMPRESS=TRUE;UserID=0XFFFFFFFF;Version=2016.2
Part name: xcku040-ffva1156-1-c
Date: 2018/01/19
Time: 12:01:10
Bitstream payload length: 0x6dc740
Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-dev2/bin/artiq_flash", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_flash')()
  File "/home/britton/artiq-dev2/artiq/artiq/frontend/artiq_flash.py", line 341, in main
    programmer.run()
  File "/home/britton/artiq-dev2/artiq/artiq/frontend/artiq_flash.py", line 117, in run
    cmdline += ["-c", "; ".join(script)]
NameError: name 'script' is not defined
```

whitequark, 2018-01-20:
This is a bug in artiq_flash that I've fixed long ago. Let me kick off a new build for you.

jbqubit, 2018-01-20:
I'm sure I'm running artiq_flash from master. I can add 1/0 to source and make it die. 

```
$ conda info

     active environment : artiq-dev2
    active env location : /home/britton/anaconda3/envs/artiq-dev2
            shell level : 1
       user config file : /home/britton/.condarc
 populated config files : /home/britton/.condarc
          conda version : 4.4.7
    conda-build version : 3.0.6
         python version : 3.5.4.final.0
       base environment : /home/britton/anaconda3  (writable)
           channel URLs : https://repo.continuum.io/pkgs/main/linux-64
                          https://repo.continuum.io/pkgs/main/noarch
                          https://repo.continuum.io/pkgs/free/linux-64
                          https://repo.continuum.io/pkgs/free/noarch
                          https://repo.continuum.io/pkgs/r/linux-64
                          https://repo.continuum.io/pkgs/r/noarch
                          https://repo.continuum.io/pkgs/pro/linux-64
                          https://repo.continuum.io/pkgs/pro/noarch
                          http://conda.anaconda.org/conda-forge/label/main/linux-64
                          http://conda.anaconda.org/conda-forge/label/main/noarch
                          http://conda.anaconda.org/m-labs/label/dev/linux-64
                          http://conda.anaconda.org/m-labs/label/dev/noarch
                          http://conda.anaconda.org/m-labs/label/main/linux-64
                          http://conda.anaconda.org/m-labs/label/main/noarch
          package cache : /home/britton/anaconda3/pkgs
                          /home/britton/.conda/pkgs
       envs directories : /home/britton/anaconda3/envs
                          /home/britton/.conda/envs
               platform : linux-64
             user-agent : conda/4.4.7 requests/2.18.1 CPython/3.5.4 Linux/3.19.0-80-generic ubuntu/14.04 glibc/2.19
                UID:GID : 1001:1001
             netrc file : None
           offline mode : False
```

```
$ pip list |grep artiq
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
artiq (4.0.dev0+450.g1c7cb737, /home/britton/artiq-dev2/artiq)
llvmlite-artiq (0.6.0+528.g7e5e95a)
```

```
$ conda list
# packages in environment at /home/britton/anaconda3/envs/artiq-dev2:
#
aiohttp                   2.3.3            py35h1b2d56f_0  
alabaster                 0.7.10           py35h6fb19ab_0  
artiq                     4.0.dev0+450.g1c7cb737           <pip>
artiq-dev                 4.0.dev         py_447+git9eb13aba    m-labs/label/dev
async-timeout             2.0.0            py35h12a94dc_0  
asyncserial               0.1             py_13+git340e430    m-labs/label/main
babel                     2.5.1                    py35_0  
binutils-or1k-linux       2.27                          5    m-labs/label/main
bscan-spi-bitstreams      0.10.0+git                    1    m-labs/label/main
ca-certificates           2017.08.26           h1d4fec5_0  
certifi                   2017.11.5        py35h9749603_0  
cffi                      1.11.2           py35hc7b2db7_0  
chardet                   3.0.4            py35hb6e9ddf_1  
colorama                  0.3.9            py35h81e2b6c_0  
coverage                  4.4.2            py35h8fc71f1_0  
dbus                      1.10.22              h3b5a359_0  
docutils                  0.14             py35hd11081d_0  
expat                     2.2.5                he0dffb1_0  
fontconfig                2.12.4               h88586e7_1  
freetype                  2.8                  hab7d2ae_1  
[... 51 lines cut ...]
pytz                      2017.3           py35hb13c558_0  
qt                        5.6.2               h974d657_12  
quamash                   0.5.5                      py_4    m-labs/label/main
readline                  7.0                  ha6073c6_4  
regex                     2015.11.22               py35_1    m-labs/label/main
rust-core-or1k            1.23.0                       19    m-labs/label/main
rustc                     1.23.0                       18    m-labs/label/main
scipy                     1.0.0            py35hcbbe4a2_0  
setuptools                33.1.1                   py35_0    conda-forge/label/main
sip                       4.18.1           py35h9eaea60_2  
six                       1.11.0           py35h423b573_1  
snowballstemmer           1.2.1            py35h5435977_0  
sphinx                    1.4.8                    py35_0  
sphinx-argparse           0.1.13                     py_4    m-labs/label/main
sphinx_rtd_theme          0.2.4                    py35_0  
sphinxcontrib-wavedrom    1.1.0                     <pip>
sphinxcontrib-wavedrom    1.1.0                      py_1    m-labs/label/main
sqlite                    3.20.1               hb898158_2  
system                    5.8                           2  
tk                        8.6.7                hc745277_3  
wheel                     0.30.0           py35hd3883cf_1  
xz                        5.2.3                h55aa19d_2  
yarl                      0.14.2           py35h31c3c03_0  
zlib                      1.2.11               ha838bed_2  
```

whitequark, 2018-01-20:
Sorry, I forgot to push. ebd02c4f4

jbqubit, 2018-01-20:
Thanks @whitequark. I can now flash using artiq_flash.

jbqubit, 2018-01-20:
Spoke too soon... AMC works but RTM does not. You got the path wrong for location of RTM .bit.

```
$ python -m artiq.gateware.targets.sayma_rtm
$ find . -name "*.bit"
./misoc_standalone_sayma_amc/gateware/top.bit
./artiq_sayma_rtm/top.bit
$ artiq_flash --srcbuild ./artiq_sayma_rtm -t sayma_rtm
Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-dev2/bin/artiq_flash", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_flash')()
  File "/home/britton/artiq-dev2/artiq/artiq/frontend/artiq_flash.py", line 305, in main
    with open(bit, "rb") as f, open(bin_handle, "wb") as g:
FileNotFoundError: [Errno 2] No such file or directory: './artiq_sayma_rtm/gateware/top.bit'
```

Also, folder names for builds are inconsistent. 
- ./misoc_standalone_sayma_amc
- ./artiq_sayma_rtm

whitequark, 2018-01-20:
@jbqubit RTM FPGA bitstream is not loaded from flash anyway right now, you have to invoke openocd manually. This will be fixed when the bitstream load is implemented.

whitequark, 2018-01-20:
@jbqubit I've fixed artiq_flash for RTM FPGA, but only for --srcbuild for now, since I'm working on that code. You can use this as a stopgap.
```
artiq_flash --srcbuild ./artiq_sayma_rtm -t sayma_rtm load
```

jbqubit, 2018-01-22:
@whitequark Build output for RTM doesn't follow same directory hierarchy other designs.  So, ```artiq_flash --srcbuild ./artiq_sayma_rtm -t sayma_rtm load``` fails line after this one: 
https://github.com/m-labs/artiq/blob/472840f16bb1275a96f495800646a0d3e6bbf6c6/artiq/frontend/artiq_flash.py#L300

jbqubit, 2018-01-22:
Then ```programmer_flash_binary``` fails since ```'artiq_sayma_rtm/software/bootloader/bootloader.bin'``` doesn't exist.

jbqubit, 2018-01-22:
Default action for RTM shouldn't be default="proxy gateware bootloader firmware start" since there is not bootloader. User ought not have to know this.

jbqubit, 2018-01-22:
And more bugs... Did you test this before pushing? 

Also, now playing with artiq_flash it occurs to me that there should just be a single -t sayma. If RTM isn't present report its absence and program AMC alone. For those who'd like to debug the system by, say, repeatedly re-flashing RTM use openocd.

sbourdeauducq, 2018-01-23:
AMC without RTM is not supported.

jbqubit, 2018-01-25:
> Accessing AMC hardware without the RTM connected is not supported.

That's fine. And why I suggested having a single ```-t sayma``` option for ```artiq_flash```.

whitequark, 2018-01-25:
There will be one option after my last round of `artiq_flash` changes.

jbqubit, 2018-01-25:
@whitequark Did you try building .bit from scratch and then using ```artiq_flash```? From status 3 days ago things were far from seamless.

whitequark, 2018-01-25:
No. It doesn't really matter because of the rewrite I mentioned.

jbqubit, 2018-01-26:
Please make an Issue for your rewrite. Please discuss adding it to ARTIQ 4.0 with @sbourdeauducq since this is a Sayma 1 dependency.

whitequark, 2018-01-26:
It's #813.

whitequark, 2018-01-27:
@jbqubit 
> @whitequark Build output for RTM doesn't follow same directory hierarchy other designs. So, artiq_flash --srcbuild ./artiq_sayma_rtm -t sayma_rtm load fails line after this one:
artiq/artiq/frontend/artiq_flash.py
> gateware_bit = artifact_path("gateware", "top.bit") 

By the way, this is your error, not mine. The line you are highlighting is from the "gateware" action, not "load" action, so it could not have possibly be executed after running the command you mention in the comment. And in fact that comment works, I just verified it.

jbqubit, 2018-01-29:
> gateware_bit = artifact_path("gateware", "top.bit")

This line appears three times: once in "gateware" and twice in "load" action. https://github.com/m-labs/artiq/blob/472840f16bb1275a96f495800646a0d3e6bbf6c6/artiq/frontend/artiq_flash.py#L325

jbqubit, 2018-01-31:
> You linked to line 300 specifically.

whitequark, 2018-01-31:
Yes, I already said that I did and it works...