m-labs/artiq #902: FPGA panics

jonaskeller, 2018-01-20:
Using 3.2 py_1+gita4337944, kc705, nist_qc2 gateware, Windows 7 64bit

The symptoms are the same as #877, but since it doesn't happen by itself anymore, I'm making this a new issue.

It happened every few 10min while I was running various experiments. I thought I had it narrowed down to running the following experiment (first thing after resetting the FPGA and starting the master+dashboard):

    @kernel
    def run(self):
        self.core.reset()
        while not self.scheduler.check_pause():
            self.core.break_realtime()
            delay(100*us)

This triggered the issue after ~0.5...10min on a few consecutive tries, until it suddenly stopped happening, so there must be another parameter involved.

I'm attaching the corelog for the crash caused by the above experiment, in the hope that you can give me a hint what to look for in order to reproduce it reliably.

[uart_corelog.txt](https://github.com/m-labs/artiq/files/1648422/uart_corelog.txt)

whitequark, 2018-01-20:
Unfortunately there is no backtrace so there's not much I can do.

sbourdeauducq, 2018-01-23:
Ran the experiment for 15 minutes with master and ``artiq_run``, no crash.

whitequark, 2018-01-26:
I've fixed the backtraces in d58393a1 (master) and aa64b8ad (release-3), can you please reproduce again with a newer firmware? You can decode the backtrace with `addr2line` if you feed the addresses and `runtime.elf` to it.

whitequark, 2018-01-26:
@sbourdeauducq I think we need to cut another 3.x release... Not sure how backtraces fell through the cracks here.

jonaskeller, 2018-01-30:
Below is the backtrace for one instance of it crashing (this is with 3.2+8.gaa64b8ad). Since I wasn't sure what causes it and was working on something else, I just let one of our experiments run in a loop - it's far from minimal, I'm afraid. If you need me to test something simpler, please let me know.

	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517012401014/work/artiq/firmware/libboard/config.rs:91
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/panicking.rs:71
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/slice/mod.rs:745
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/slice/mod.rs:876

whitequark, 2018-01-30:
No, this is enough. Actually, the crash has nothing to do with your *experiment*. Do you have an idle kernel set up?

jonaskeller, 2018-01-30:
No (we usually do, but I didn't set it up this time).

whitequark, 2018-01-30:
Do you still have the orignial backtrace (with addresses)? Looks like you didn't pass the `-i` option and the inlined frames are skipped.

jonaskeller, 2018-01-30:
Oh, sorry. Here's the original:

	panic at src/libcore/slice/mod.rs:745: index 60 out of range for slice of length 46
	backtrace for software version 3.2+8.gaa64b8ad:
	0x40021af4
	0x4003d248
	0x4003d394
	0x4001eff8
	0x40021868
	halting.

whitequark, 2018-01-30:
Can you please rerun it with `-i`?

jonaskeller, 2018-01-30:
/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517012401014/work/artiq/firmware/libboard/config.rs:91
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517012401014/work/artiq/firmware/libboard/config.rs:106
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517012401014/work/artiq/firmware/runtime/lib.rs:207
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/panicking.rs:71
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/slice/mod.rs:745
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/slice/mod.rs:876
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/slice/mod.rs:727
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/888b1d2/src/wire/ipv4.rs:325
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/888b1d2/src/iface/ethernet.rs:566
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/888b1d2/src/iface/ethernet.rs:493
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/888b1d2/src/iface/ethernet.rs:354
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517012401014/work/artiq/firmware/runtime/ethmac.rs:85
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/888b1d2/src/iface/ethernet.rs:353
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/888b1d2/src/iface/ethernet.rs:302
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517012401014/work/artiq/firmware/runtime/lib.rs:154
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/ops/function.rs:223
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517012401014/work/artiq/firmware/liblogger_artiq/lib.rs:43
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517012401014/work/artiq/firmware/runtime/lib.rs:178

whitequark, 2018-01-30:
That... makes no sense. The only way to interpret that backtrace is that a packet arrived, passed the length verification in `smoltcp::Ipv4Packet::new_checked`, and subsequently failed the exact same length check it in `smoltcp::Ipv4Packet::payload()`.

whitequark, 2018-01-30:
Ah, it fails a *different* length check.

sbourdeauducq, 2018-01-30:
@jonaskeller How problematic is this bug for you? Should we do a 3.4 release soon with the fix?

jonaskeller, 2018-01-30:
Thanks for fixing this.

At the moment, I'm still setting up the system, but I hope to integrate it into the experiment around next week. At that point, this bug would become quite problematic, since it occurs at a rate >1/h.

sbourdeauducq, 2018-01-30:
OK, we will cut 3.4 next Monday.

jonaskeller, 2018-01-30:
Great, thanks!

sbourdeauducq, 2018-01-30:
Your package is built now, do you confirm the issue is resolved?

jonaskeller, 2018-01-30:
I'm writing from home - will check tomorrow and let you know.

jonaskeller, 2018-02-07:
I'm afraid I just checked with 3.3+3.gb92b00a1 (can't test with 3.4 yet due to #915) and still saw a panic after <1h, with a pretty similar looking backtrace:

	...
	[  1773.725254s]  WARN(runtime): network error: malformed packet
	[  1801.498077s]  WARN(runtime): network error: malformed packet
	[  1802.334523s]  WARN(runtime): network error: malformed packet
	[  1803.020813s]  WARN(runtime): network error: malformed packet
	[  1829.824930s]  INFO(runtime::session): no connection, starting idle kernel
	[  1829.989399s]  INFO(runtime::session): no idle kernel found
	[  1830.373260s]  WARN(runtime): network error: malformed packet
	[  1830.915350s]  INFO(runtime::session): new connection from 132.163.182.117:51117
	[  1831.550079s]  WARN(runtime): network error: malformed packet
	[  1832.974991s]  WARN(runtime): network error: malformed packet
	[  1836.867849s]  INFO(runtime::kern_hwreq): resetting RTIO
	panic at src/libcore/slice/mod.rs:745: index 143 out of range for slice of length 46
	backtrace for software version 3.3+3.gb92b00a1:
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517288134406/work/artiq/firmware/libboard/config.rs:91
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517288134406/work/artiq/firmware/libboard/config.rs:106
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517288134406/work/artiq/firmware/runtime/lib.rs:207
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/panicking.rs:71
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/slice/mod.rs:745
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/slice/mod.rs:876
	/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/slice/mod.rs:727
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/181083f/src/wire/ipv4.rs:329
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/181083f/src/iface/ethernet.rs:642
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/181083f/src/iface/ethernet.rs:496
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/181083f/src/iface/ethernet.rs:357
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517288134406/work/artiq/firmware/runtime/ethmac.rs:85
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/181083f/src/iface/ethernet.rs:356
	/var/lib/buildbot/.cargo/git/checkouts/smoltcp-ebf9e93b1271bd34/181083f/src/iface/ethernet.rs:305
	/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_qc2_1517288134406/work/artiq/firmware/runtime/lib.rs:154
	halting.
	use `artiq_coreconfig write -s panic_reboot 1` to reboot instead

whitequark, 2018-02-08:
I'm afraid that I will need a packet capture to fix this now, because the fuzzer comes up empty.

jonaskeller, 2018-02-08:
Here you are: [panic_logs_artiq33.zip](https://github.com/m-labs/artiq/files/1707975/panic_logs_artiq33.zip)

This is the packet capture during the panic mentioned above.

Core device IP is `132.163.182.107`, host is `132.163.182.117`

The capture filter `host 132.163.182.107` was used for that one. I also have a capture from today where I added `or broadcast`, but I'm not sure how happy IT would be with me making that available publicly.

If you can't find what you need in there, please let me know.

whitequark, 2018-02-08:
I'll need the `or broadcast` capture since the panics are caused by malformed IP packets and those aren't in the core device communication.

whitequark, 2018-02-09:
I've looked at your packet capture. This is very weird; I was not able to crash smoltcp locally. I will add an option to ARTIQ to print a packet trace.

whitequark, 2018-02-14:
Can you please:
  * flash the firmware built from the `release-3` branch;
  * run `artiq_coreconfig write -s net_trace 1`
  * run `artiq_coreboot reboot`
  * wait for the offending packet
  * send me the pcap together with the core log
I can't seem to be able to reproduce the crash using tcpreplay for reasons I don't understand, so I need to know which exact packet causes it.

whitequark, 2018-02-15:
The build is done now.

sbourdeauducq, 2018-02-15:
Try this MiSoC patch:
```
--- a/misoc/cores/liteeth_mini/mac/sram.py
+++ b/misoc/cores/liteeth_mini/mac/sram.py
@@ -86,27 +86,25 @@ class LiteEthMACSRAMWriter(Module, AutoCSR):
                 ),
                 If(sink.eop,
                     If((sink.error & sink.last_be) != 0,
-                        NextState("DISCARD")
+                        counter_reset.eq(1),
+                        NextState("IDLE")
                     ).Else(
-                        NextState("TERMINATE")
+                        NextState("COMPLETE")
                     )
                 )
             )
         )
-        fsm.act("DISCARD",
-            counter_reset.eq(1),
-            NextState("IDLE")
-        )
         fsm.act("DISCARD_REMAINING",
+            counter_reset.eq(1),
             If(sink.stb & sink.eop,
-                NextState("TERMINATE")
+                NextState("IDLE")
             )
         )
         self.comb += [
             fifo.sink.slot.eq(slot),
             fifo.sink.length.eq(counter)
         ]
-        fsm.act("TERMINATE",
+        fsm.act("COMPLETE",
             counter_reset.eq(1),
             slot_ce.eq(1),
             fifo.sink.stb.eq(1),
```

sbourdeauducq, 2018-02-16:
@enjoy-digital Does this patch look good?

enjoy-digital, 2018-02-16:
@sbourdeauducq: we added some features to drop packets and code seems indeed inconsistent.

In your case, you are missing the case where the packet exceeds the mtu and we just want to discard the remaining.

This should be better:
```
--- a/misoc/cores/liteeth_mini/mac/sram.py
+++ b/misoc/cores/liteeth_mini/mac/sram.py
@@ -41,14 +41,6 @@ class LiteEthMACSRAMWriter(Module, AutoCSR):
                 increment.eq(4)
             )
         counter = Signal(lengthbits)
-        counter_reset = Signal()
-        counter_ce = Signal()
-        self.sync += \
-            If(counter_reset,
-                counter.eq(0)
-            ).Elif(counter_ce,
-                counter.eq(counter + increment)
-            )
 
         # slot computation
         slot = Signal(slotbits)
@@ -69,46 +61,39 @@ class LiteEthMACSRAMWriter(Module, AutoCSR):
             If(sink.stb,
                 If(fifo.sink.ack,
                     ongoing.eq(1),
-                    counter_ce.eq(1),
+                    NextValue(counter, increment),
                     NextState("WRITE")
                 ).Else(
                     NextValue(self.errors.status, self.errors.status + 1),
-                    NextState("DISCARD_REMAINING")
+                    NextState("DISCARD")
                 )
             )
         )
+        fsm.act("DISCARD",
+            If(sink.stb & sink.eop,
+                NextState("IDLE")
+            )
+        )
         fsm.act("WRITE",
             If(sink.stb,
-                If(counter == eth_mtu,
-                    NextState("DISCARD_REMAINING")
-                ).Else(
-                    counter_ce.eq(1),
+                If(counter != eth_mtu,
+                    NextValue(counter, counter + increment),
                     ongoing.eq(1)
                 ),
                 If(sink.eop,
                     If((sink.error & sink.last_be) != 0,
-                        NextState("DISCARD")
+                        NextState("IDLE")
                     ).Else(
-                        NextState("TERMINATE")
+                        NextState("COMPLETE")
                     )
                 )
             )
         )
-        fsm.act("DISCARD",
-            counter_reset.eq(1),
-            NextState("IDLE")
-        )
-        fsm.act("DISCARD_REMAINING",
-            If(sink.stb & sink.eop,
-                NextState("TERMINATE")
-            )
-        )
         self.comb += [
             fifo.sink.slot.eq(slot),
             fifo.sink.length.eq(counter)
         ]
-        fsm.act("TERMINATE",
-            counter_reset.eq(1),
+        fsm.act("COMPLETE",
             slot_ce.eq(1),
             fifo.sink.stb.eq(1),
             NextState("IDLE")
```

sbourdeauducq, 2018-02-16:
Yes, my code completely discards oversized packets on purpose. If you truncate them, they will fail FCS and get discarded later - why do you want to keep this behavior?

whitequark, 2018-02-16:
@jonaskeller via e-mail:
> Also, it seems worth mentioning that enabling the net_trace option
> dramatically increases the rate of panics. In all my attempts, it took
> less than a minute between FPGA reset and panic. Due to the risk of
> this being a separate bug, I've disabled the option for this capture,
> which is why it took about 20 minutes.

This identifies the bug as gateware bug with virtual certainty; every IP packet undergoes the same processing regardless of the net_trace option, and increasing load on the core device triggering the race condition more often is exactly what I would expect.

jonaskeller, 2018-02-17:
Just out of curiosity: Does that still mean that it takes a corrupted packet to trigger it? And if so, have you been able to reproduce it with the latest capture / found a culprit?

As for your suggestion to check whether it persists in 4.0.dev: conda refuses to install anything beyond `4.0.dev-py_206+git70130847` because it can't satisfy the `openocd 0.10.0 4` dependence. Is it worth trying with that one or should I look into getting that openocd version?

whitequark, 2018-02-17:
> Just out of curiosity: Does that still mean that it takes a corrupted packet to trigger it? And if so, have you been able to reproduce it with the latest capture / found a culprit?

There are no corrupted packets in your logs as far as I can tell. It looks like the bug where smoltcp could be crashed by a corrupted packet was completely independent and you never triggered it; it was pure chance that the code path triggering that bug and the backtrace you've collected intersected. At least, that's my current understanding, I may still be wrong.

whitequark, 2018-02-17:
> Is it worth trying with that one or should I look into getting that openocd version?

No, you need the very last commit on master.

jonaskeller, 2018-02-17:
So what's the quickest way to fulfill the openocd version requirement? I.e. do I need to build it myself or is there a way around that?

whitequark, 2018-02-17:
I'm actually not sure why you're hitting that error--are you flashing on Windows perchance?

jonaskeller, 2018-02-17:
Yes, it's on windows (not my choice...). Looks to me like the win64 and linux64 packages for openocd are the same version in the conda repository, though.
Still, if moving to linux fixes the problem, I'll go find a linux machine next week and retry.

sbourdeauducq, 2018-02-17:
Try it on windows again with the latest artiq (07a31f8d or newer) - I just fixed some openocd conda issues.

sbourdeauducq, 2018-02-20:
@jonaskeller How are things (flashing and panic) with a recent artiq-4 package from master? If everything works I'll backport the changes to release-3 and make a new release.

jonaskeller, 2018-02-20:
Conda still defaults to `4.0.dev-py_206+git70130847` for the `artiq-kc705-nist_qc2` package. If I try to force either `py_553+git07a31f8d` or `py_582+gitf060d6e1`, it complains about not being able to fulfill the `openocd 0.10.0 4` dependency.

sbourdeauducq, 2018-02-21:
Sorry, I had missed the openocd 4 dependency in one place. Does it work with the lastest master now?

jonaskeller, 2018-02-21:
Installation of the packages works now (I'm using `4.0.dev-py_583+giteed64a6d`), but I can't flash the core device.

I wasn't sure how to pass what was formerly the `-m` parameter, so I tried both `-V nist_qc2`, and `-d "C:\Anaconda3\envs\artiqdev\Lib\site-packages\artiq\binaries\kc705-nist_qc2"`. An update to the documentation would be helpful here.

In either case, I get the same error, which looks like these paths could do with some escape characters:

	(artiqdev) c:\artiq_cryo1\repository>artiq_flash -t kc705 -d "C:\Anaconda3\envs\artiqdev\Lib\site-packages\artiq\binaries\kc705-nist_qc2"
	Design: top;UserID=0XFFFFFFFF;COMPRESS=TRUE;Version=2017.4
	Part name: 7k325tffg900
	Date: 2018/02/21
	Time: 11:23:10
	Bitstream payload length: 0x49e17c
	Open On-Chip Debugger 0.10.0-00013-gbb7bedad (2018-02-17-05:04)
	Licensed under GNU GPL v2
	For bug reports, read
			http://openocd.org/doc/doxygen/bugs.html
	none separate
	Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
	adapter speed: 25000 kHz
	Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
	Info : clock speed 25000 kHz
	Info : JTAG tap: xc7.tap tap/device found: 0x43651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x4)
	Info : gdb server disabled
	TEMP 35.94 C
	VCCINT 1.005 V
	VCCAUX 1.826 V
	VCCBRAM 1.008 V
	VPVN 0.000 V
	VREFP 0.000 V
	VREFN 0.000 V
	VCCPINT 0.000 V
	VCCPAUX 0.000 V
	VCCODDR 0.000 V
	Error: couldn't stat() C:Anaconda3envsrtiqdevsharscan-spi-bitstreamscan_spi_xc7k325t.bit: No such file or directory
	failed loading file C:Anaconda3envsrtiqdevsharscan-spi-bitstreamscan_spi_xc7k325t.bit to pld device 0

	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiqdev\Scripts\artiq_flash-script.py", line 5, in <module>
		exit(main())
	  File "C:\Anaconda3\envs\artiqdev\lib\site-packages\artiq\frontend\artiq_flash.py", line 357, in main
		programmer.run()
	  File "C:\Anaconda3\envs\artiqdev\lib\site-packages\artiq\frontend\artiq_flash.py", line 188, in run
		self._client.run_command(cmdline)
	  File "C:\Anaconda3\envs\artiqdev\lib\site-packages\artiq\remoting.py", line 56, in run_command
		subprocess.check_call([arg.format(tmp=self._tmp, **kws) for arg in cmd])
	  File "C:\Anaconda3\envs\artiqdev\lib\subprocess.py", line 271, in check_call
		raise CalledProcessError(retcode, cmd)
	subprocess.CalledProcessError: Command '['openocd', '-s', 'C:\\Anaconda3\\envs\\artiqdev\\Library\\share\\openocd\\scripts', '-c', 'source [find board/kc705.cfg]; target create xc7.spi0.proxy testee -chain-position xc7.tap; flash bank spi0 jtagspi 0 0 0 0 xc7.spi0.proxy 0x2; gdb_port disabled; tcl_port disabled; telnet_port disabled; init; xadc_report xc7.tap; pld load 0 C:\\Anaconda3\\envs\\artiqdev\\share\\bscan-spi-bitstreams\\bscan_spi_xc7k325t.bit; flash probe spi0; flash erase_sector spi0 0 73; flash write_bank spi0 C:\\Users\\rabi\\AppData\\Local\\Temp\\1\\tmpa1v_bx7v 0x0; flash verify_bank spi0 C:\\Users\\rabi\\AppData\\Local\\Temp\\1\\tmpa1v_bx7v 0x0; flash probe spi0; flash erase_sector spi0 175 176; flash write_bank spi0 C:\\Anaconda3\\envs\\artiqdev\\Lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\bootloader.bin 0xaf0000; flash verify_bank spi0 C:\\Anaconda3\\envs\\artiqdev\\Lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\bootloader.bin 0xaf0000; flash probe spi0; flash erase_sector spi0 180 188; flash write_bank spi0 C:\\Anaconda3\\envs\\artiqdev\\Lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\runtime.fbi 0xb40000; flash verify_bank spi0 C:\\Anaconda3\\envs\\artiqdev\\Lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\runtime.fbi 0xb40000; xc7_program xc7.tap; exit']' returned non-zero exit status 1

sbourdeauducq, 2018-02-21:
> An update to the documentation would be helpful here.

It's in the release notes already, though the options are called by their long names (``--...``) there. ``-d`` should not be required.

jonaskeller, 2018-02-21:
Oh, ok - I missed that, sorry.
So this means that one of my attempts already used the right command. Still, there seems to be a problem with the paths to these other bitstream files.

sbourdeauducq, 2018-02-22:
Can you try with 3.4-py_4+gitcb605cf0?

jonaskeller, 2018-02-23:
I don't have time at the moment to look into building it myself, so I'd rather wait for your bot for now. Once I have a compiled version, I'll give it a try.

sbourdeauducq, 2018-02-23:
As I mentioned to you on IRC there are binaries. https://anaconda.org/m-labs/artiq-kc705-nist_qc2/files

jonaskeller, 2018-02-24:
So far, it's looking good. I left it idle for 6h and didn't see it crash, then set `net_trace` to 1 and left an experiment running in a loop to stress the core device. It's been running in that mode for over 2h now, which is a lot considering that I never saw it survive for more than a minute before. I'll leave it running over the weekend and let you know Monday how it went.

jonaskeller, 2018-02-27:
After running all weekend, it seemed fine when I got here this morning. So, in any case, this is a major improvement. I can't yet tell whether it's entirely solved, however: Since then, I've seen two instances of the coredevice not responding anymore until I manually reset the FPGA, but didn't catch any `panic` entries in the UART log (as described here https://github.com/m-labs/misoc/issues/66 , flterm needs to be manually restarted every few hours). I'll keep watching and will try to determine whether it's the same issue or not.

jonaskeller, 2018-03-01:
I've been continuously monitoring since the last post and haven't seen a single coredevice panic. Occasionally (~2 times a day), connections are refused for a brief period of time, and that might have been what I described above (just being too impatient and hitting reset before it resolved itself). These are quite different symptoms though, so it seems very likely that the original issue has been fixed. If the other one keeps occurring, I'll open a new issue with a more detailed description.

closed 2018-03-01 by jonaskeller

sbourdeauducq, 2018-03-01:
OK, thanks for your testing and patience!

commit 807eb115 (2018-01-30): Update smoltcp.

Fixes #902.
commit b92b00a1 (2018-01-30): Update smoltcp.

Fixes #902.