m-labs/artiq #903: Binaries missing  in artiq 3.x on Windows

KaifengC, 2018-01-22:
Trying to upgrade artiq to 3.2 version using `conda create -n artiq_32 artiq-kc705-nist_clock=3.2`.
The installation is successful but then I tried `artiq_flash -t kc705 -m nist_clock` but got `Binaries directory...... does not exist`

I checked both nist_clock and nist_qc2 hardwares on windows. It seems the binaries files are lost after the installation.

sbourdeauducq, 2018-01-22:
https://github.com/m-labs/artiq/issues/894 ?

jbqubit, 2018-01-22:
@sbourdeauducq Please try installing 3.2 on a Windows 10 machine. You will see several hiccups in addition to those reported by @TheCakeIsAPi and @KaifengC.

sbourdeauducq, 2018-01-23:
What exactly? We do have Windows CI for many things so that's surprising.

KaifengC, 2018-01-23:
Yes, it's the same with #894 . Sorry I didn't see that before.
I installed several versions on Windows. This bug firstly appears on 3.0. It works well on version 2.5

sbourdeauducq, 2018-01-23:
Yes, it's due to switching to new-style conda noarch package that happened in 2.4, since after an update of conda the old-style noarch packages began to have various bugs and problems. Maintaining conda takes constant work...

sbourdeauducq, 2018-01-23:
For now, can you put the binaries manually where artiq_flash expects them?

closed 2018-01-23 by sbourdeauducq