m-labs/artiq #904: jesd204 sc1 support does not compile

sbourdeauducq, 2018-01-23:
```
$ ./artiq/gateware/targets/sayma_amc.py 
Traceback (most recent call last):
  File "./artiq/gateware/targets/sayma_amc.py", line 485, in <module>
    main()
  File "./artiq/gateware/targets/sayma_amc.py", line 481, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/sb/artiq_drtio/artiq/build_soc.py", line 25, in build_artiq_soc
    builder.build()
  File "/home/sb/misoc/misoc/integration/builder.py", line 176, in build
    self.soc.finalize()
  File "/home/sb/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/sb/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/sb/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/sb/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/sb/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/sb/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/sb/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/sb/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/sb/migen/migen/fhdl/decorators.py", line 40, in get_fragment
    f = orig_get_fragment()
  File "/home/sb/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/sb/migen/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/home/sb/jesd204/jesd204b/core.py", line 113, in do_finalize
    assert hasattr(self, "jref_registered")
AssertionError
```

enjoy-digital, 2018-01-23:
We need to un-comment this: https://github.com/m-labs/artiq/blob/master/artiq/gateware/targets/sayma_amc.py#L101

sbourdeauducq, 2018-01-23:
I see. Well, Sayma is a development system at the moment and building from source is the primary way, so for now the priority is to get source builds working correctly (not conda).

sbourdeauducq, 2018-01-23:
@enjoy-digital Let's make a new jesd204 release when sc1 is validated - this should be soon?

commit 4b4374f7 (2018-01-23): sayma: register_jref for JESD204. Closes #904
closed 2018-01-23 by sbourdeauducq