m-labs/artiq #906: JESD core conda update

jbqubit, 2018-01-25:
cf #854

>Get the JESD core from git (https://github.com/m-labs/jesd204b), not the conda package. Considering how many people run into this issue, I should probably update the conda package...
You can also strip all the JESD stuff since you don't need it for Ethernet.

>It's just 3 commands:
conda remove jesd204b
git clone https://github.com/m-labs/jesd204b
export PYTHONPATH=$PYTHONPATH:`pwd`/jesd204b

closed 2018-01-25 by sbourdeauducq

jbqubit, 2018-01-26:
@sbourdeauducq is this closed because you're annoyed or because jesd204b conda is updated? I didn't see a commit.

sbourdeauducq, 2018-01-26:
The conda package will be built when JESD204 SC1 is done, at the latest. We don't need an issue for every single thing. And, in the present state of affairs, if you use Sayma you are generally expected to build from source and use git.

jbqubit, 2018-01-26:
[This](http://m-labs.hk/artiq/manual-master/developing.html#artiq-anaconda-development-environment) is how I prepare a conda environment for building .bit. Your build system also uses the same artiq-dev.yaml. Please make it easy for us all to use the same build environment.

sbourdeauducq, 2018-01-26:
For now just install the JESD core from git using the instructions I posted elsewhere.

jbqubit, 2018-01-26:
```
conda remove jesd204b
git clone https://github.com/m-labs/jesd204b
export PYTHONPATH=$PYTHONPATH:`pwd`/jesd204b
```