m-labs/artiq #907: KC705 Set(flash) the MAC and IP address

joeshardow, 2018-01-26:
When I try to follow the installation step from artiq manual to flash the MAC and IP address by using the commands, I meet some problems: 

$ artiq_mkfs flash_storage.img -s mac xx:xx:xx:xx:xx:xx -s ip xx.xx.xx.xx           
(This command works, and generate the flash_storage.img file in the current file folder. I suppose that the "xx:xx:xx:xx:xx:xx" means KC705 MAC, it is printed on hardware board, and the ip "xx.xx.xx.xx" means the IP KC705 will get.)

$ artiq_flash -t [board] -m [adapter] -f flash_storage.img proxy storage start
(This command is not work, gives me an KeyError: 'board'.)

The command and the result is shows below:

(artiq-main) C:\Users\joewong>artiq_flash -t board -m adapter -f flash_storage.img proxy storage start
Traceback (most recent call last):
  File "C:\Users\joewong\Anaconda3\envs\artiq-main\Scripts\artiq_flash-script.py", line 9, in <module>
    sys.exit(main())
  File "C:\Users\joewong\Anaconda3\envs\artiq-main\lib\site-packages\artiq\frontend\artiq_flash.py", line 75, in main
    }[opts.target]
KeyError: 'board'

what's the problem with it? Where can I find the artiq command relating information list. Maybe the manual of artiq has, but I didn't know how to find?

[image]

sbourdeauducq, 2018-01-26:
As I told you before, you need to replace things between brackets with the appropriate values for your situation, in your case kc705 instead of [board].

closed 2018-01-26 by sbourdeauducq

joeshardow, 2018-01-26:
What's the meaning of adapter? What should be replaced with the place of '[adapter]'?

joeshardow, 2018-01-26:
What's the meaning of adapter? What should it be replaced by with the place of '[adapter]'? There is not an option example in the manual.