m-labs/artiq #908: Sayma MiSoC memory test failed

jbqubit, 2018-01-26:
Building .bit from source using
```
commit 440e19b8f9c8ebfce80402a519796cee7fdd6b06
```
I see... 

```
$ flterm /dev/ttyUSB2

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017 M-Labs Limited

Bootloader CRC passed
Initializing SDRAM...
Write leveling: 32 59 47 69 59 69 40 48 done
Read delays: 7:00-116 6:00-138 5:27-172 4:38-54 3:57-76 2:67-83 1:98-116 0:109-125 done
SDRAM initialized
Memory test failed (384482/1114624 words incorrect)
Halting.

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017 M-Labs Limited

Bootloader CRC passed
Initializing SDRAM...
Write leveling: 34 56 45 65 55 67 40 47 done
Read delays: 7:00-112 6:01-127 5:15-33 4:26-44 3:65-84 2:78-94 1:90-123 0:103-123 done
SDRAM initialized
Memory test failed (412138/1114624 words incorrect)
Halting.

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017 M-Labs Limited

Bootloader CRC passed
Initializing SDRAM...
Write leveling: 33 55 45 64 55 72 40 47 done
Read delays: 7:00-115 6:00-136 5:10-26 4:36-53 3:61-79 2:76-97 1:88-108 0:96-113 done
SDRAM initialized
Memory test failed (499333/1114624 words incorrect)
Halting.

|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017 M-Labs Limited

Bootloader CRC passed
Initializing SDRAM...
Write leveling: 34 57 48 69 59 73 41 50 done
Read delays: 7:00-117 6:00-134 5:18-40 4:31-47 3:70-86 2:76-96 1:103-120 0:93-109 done
SDRAM initialized
Memory test failed (473026/1114624 words incorrect)
Halting.

```

whitequark, 2018-01-26:
Possibly caused by 7429ee4fb63316b05da07407d6802670ebdb80fd?

cjbe, 2018-01-26:
The typical valid read region seems to be ~170 LSB on my board, so I don't think that commit (increasing the initial step from 8 LSB to 16 LSB) caused this.

This also looks different from the problem that commit solved for me, where the size of the read window was always the size of the initial step. Here the gaps vary from 16 to 20.

sbourdeauducq, 2018-01-27:
What Vivado version?

jbqubit, 2018-01-29:
I'm using 2016.2. Will upgrade and try again.

marmeladapk, 2018-01-29:
I'm using 2017.4 and I also got this issue, though with build from 25.01. I'm currently building against 0edc34a, will update when it finishes.

marmeladapk, 2018-01-29:
```
MiSoC Bootloader
Copyright (c) 2017 M-Labs Limited

Bootloader CRC passed
Initializing SDRAM...
Write leveling: 58 89 78 88 56 85 47 48 done
Read delays: 7:00-19 6:07-23 5:53-74 4:60-76 3:111-132 2:117-133 1:125-141 0:133-151 done
SDRAM initialized
Memory test failed (522120/1114624 words incorrect)
Halting.
```

sbourdeauducq, 2018-01-30:
Here is everything I built from the current master (with RTM bridge, RTIO and other things disabled to save compilation time):
http://dl.free.fr/lAFdh3oQV
With those binaries, I verified that SDRAM works fine on both Florent's board and Sayma-1.
Can you try those binaries on your boards?
@marmeladapk You can use the Ethernet TX clock phase adjustement script I posted in the RGMII issue on those binaries.
@marmeladapk If the problem persists, can you use Sayma-2 that I shipped to you to debug Ethernet, since I didn't have SDRAM problems on that one?

enjoy-digital, 2018-01-30:
thanks @sbourdeauducq. I'll look at that. The read leveling procedure is probably still not robust enough.

marmeladapk, 2018-01-30:
@sbourdeauducq I loaded it to check if memory tests are passed:
```
Bootloader CRC passed
Initializing SDRAM...
Write leveling: 95 130 118 133 96 125 87 88 done
Read delays: 7:37-249 6:61-267 5:110-316 4:131-336 3:167-369 2:173-377 1:195-40e
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000005s]  INFO(runtime): ARTIQ runtime starting...
[     0.003864s]  INFO(runtime): software version 4.0.dev+516.g0edc34a9
[     0.010126s]  INFO(runtime): gateware version 4.0.dev+516.g0edc34a9.dirty
[     0.016910s]  INFO(runtime): log level set to INFO by default
[     0.022630s]  INFO(runtime): UART log level set to INFO by default
[     0.028790s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1.028006s]  INFO(runtime): continuing boot
[     1.030975s]  WARN(runtime): using default MAC address 02-00-00-00-76-01; ct
[     1.039568s]  INFO(runtime): using default IP address 192.168.1.60
[     1.054501s]  INFO(runtime::session): accepting network sessions
[     1.059438s]  INFO(runtime::session): running startup kernel
[     1.064959s]  INFO(runtime::session): no startup kernel found
[     1.070665s]  INFO(runtime::session): no connection, starting idle kernel
[     1.077527s]  INFO(runtime::session): no idle kernel found
[     1.084122s]  INFO(runtime::mgmt): management interface active
[     6.274350s]  WARN(runtime): ethernet mac: rx preamble errors: 2
[     7.357698s]  WARN(runtime): ethernet mac: rx preamble errors: 3
[    19.820658s]  WARN(runtime): ethernet mac: rx preamble errors: 4
[    20.128752s]  WARN(runtime): ethernet mac: rx preamble errors: 5
[    20.888642s]  WARN(runtime): ethernet mac: rx preamble errors: 6
```
So it works. I'll try the script you mentioned later.

sbourdeauducq, 2018-01-30:
Good. You do however seem to get a large number of Ethernet RX corrupted packets (preamble errors). Is the PHY correctly set in RGMII mode? You can change the RX phase as well by using this script command instead: ``set_property CLKOUT0_PHASE <phase> [get_cells crg_ethrx_mmcm]``

marmeladapk, 2018-01-30:
@sbourdeauducq Should I change it in xdc in artiq/artiq_sayma/gateware/top.xdc and rebuild? Will latest artiq pass memory tests?

sbourdeauducq, 2018-01-30:
No. Please follow the instruction in my comment: https://github.com/m-labs/artiq/issues/854#issuecomment-360497764 - you just save the script as ``edit_pll.tcl`` and run the mentioned vivado command.
There is no bitstream rebuilding and it is a rather quick process. Nothing in the design other than the PLL phase will be changed, the routes etc. will be exactly as before, so yes memory test should be unaffected.

sbourdeauducq, 2018-02-03:
I also see the problem with the default build (including SAWG) on ARTIQ 4c22d64ee438d8b65ba728829794698191719181, migen e554f072905ceeb27c9c179c8c7b785acd1676bc, misoc cb8e314c7515eade46f5bcde4e48903d7ec92490
```
Initializing SDRAM...
Write leveling: 43 66 49 68 35 56 34 25 done
Read delays: 7:00-121 6:00-141 5:39-55 4:44-60 3:67-85 2:76-92 1:105-121 0:113-129 done
SDRAM initialized
Memory test failed (356593/1114624 words incorrect)
```

When disabling the SAWG (``--without-sawg``), the system boots correctly.
@enjoy-digital Can you move forward with JESD SC1 by disabling SAWG (which you want to do anyway to reduce compilation time)? I cannot reproduce the "no output on UART" bug.

enjoy-digital, 2018-02-03:
@sbourdeauducq: yes i'll continue on Monday.

sbourdeauducq, 2018-02-06:
@hartytp Are you looking into this?
This definitely worked when I did the SAWG test and posted the scope screenshot. So it should be possible to isolate what code change exactly caused this problem, maybe with the help of tools like ``git-bisect``.
But I suspect this is due to the non-determinism of Vivado compilation, or to plain Vivado bugs. In the first case, this is normally solvable by adding appropriate timing constraints. In the second case, considering how Xilinx technical support has been degrading for the past years, the first option is basically to apply somewhat random non-functional changes to the code and hope that *ça tombe en marche*, or try various Vivado synthesis options. (Xilinx's answer to the bug invasion is pretty much [the usual](https://www.bunniestudios.com/blog/?p=5127))
@whitequark's addition of RTM loading gateware is a good suspect for the triggering of this kind of Vivado misbehavior.

hartytp, 2018-02-06:
> @hartytp Are you looking into this?

I wasn't planning to, no. In general, I'm trying to prioritize things like the HMC830 on Sayma, which seem to be (at least in part) hardware issues. In contrast, the mem test thing is just firmware/gateware, isn't it? As such, it seemed like the standard yak shaving required to get a new board up and running, and not something particular to Sayma. So, I figured that you guys were probably best placed to look into it.

I have a busy week lined up this week, but I might have some time to look into it.

Side note: we've had Sayma for quite a while now, but the ARTIQ tool chain still feels quite hacked and fragile. It would be great to get to the point where Artiq flash can do the RTM as well, the package includes the correct version of JESD204B, etc.

hartytp, 2018-02-06:
Anyway to be clear, in case I do find time to look into this, your plan is basically to dig through the git history, building various versions of Sayma gateware/firmware with SAWG (at a few hours per build) until we find the point where it stopped working? IIRC, that's a bit complicated by the fact that the tools to build Sayma have changed a bit over time, so it's not always the same instructions to build/flash it, and by the fact that the package doesn't include the right version of JESD204B etc, so one needs to track the history of several projects to make sure that each build uses the correct version of each. Doesn't sound like fun.

sbourdeauducq, 2018-02-06:
> Doesn't sound like fun.

Yep, standard fare.

hartytp, 2018-02-06:
Well, as I said, as this seems like standard yak shaving for getting a board up and running, rather than a particular hardware/design issue with Sayma. So, do you mind taking a look at it first -- it's likely to be quicker for you since you've probably kept a closer eye on the changes that have been made to ARTIQ over the past weeks.

sbourdeauducq, 2018-02-07:
That's what I thought - the patch below works around the problem.
```
diff --git a/artiq/gateware/targets/sayma_amc.py b/artiq/gateware/targets/sayma_amc.py
index c45f8d37a..f6c5b95f6 100755
--- a/artiq/gateware/targets/sayma_amc.py
+++ b/artiq/gateware/targets/sayma_amc.py
@@ -160,9 +160,9 @@ class Standalone(MiniSoC, AMPSoC):
         ]
 
         # RTM bitstream upload
-        rtm_fpga_cfg = platform.request("rtm_fpga_cfg")
-        self.submodules.rtm_fpga_cfg = SlaveFPGA(rtm_fpga_cfg)
-        self.csr_devices.append("rtm_fpga_cfg")
+        #rtm_fpga_cfg = platform.request("rtm_fpga_cfg")
+        #self.submodules.rtm_fpga_cfg = SlaveFPGA(rtm_fpga_cfg)
+        #self.csr_devices.append("rtm_fpga_cfg")
 
         # AMC/RTM serwb
         serwb_pll = serwb.phy.SERWBPLL(125e6, 625e6, vco_div=2)
```
@whitequark What about using GPIO and bit-banging instead? Hopefully the Vivado trash will behave then.

marmeladapk, 2018-02-08:
@sbourdeauducq With latest commit (2d4a134) when I compile `python3 -m artiq.gateware.targets.sayma_amc --without-sawg` I still get memory test failed.

sbourdeauducq, 2018-02-08:
Still? it always worked for me when using --without-sawg.
Anyway for Ethernet debugging you can use the binaries.

sbourdeauducq, 2018-02-08:
And with sawg?

marmeladapk, 2018-02-08:
I wanted to insert probes.

> And with sawg?

Will check now.

sbourdeauducq, 2018-02-08:
> I wanted to insert probes.

For debugging RX? The Ethernet core actually doesn't need SDRAM and will continue to receive frames even with a dead CPU system. Or you can try with a super-minimal design that will also have the advantage of reducing the compilation time. Some people don't like me saying that, but Sayma (and Ultrascale) is a trash fire, and the only way to make any progress is to cling to whatever still works...

marmeladapk, 2018-02-08:
@sbourdeauducq I still get this error.

sbourdeauducq, 2018-02-08:
Those binaries are from ARTIQ 4.0.dev+521.g4c22d64e with the RTM loading gateware commented out. I tested that SDRAM works on the board when flashing them (and then booting from flash).
http://dl.free.fr/mffzh7lVw

whitequark, 2018-02-08:
@sbourdeauducq Since this bug is clearly not caused with my gateware based on this failure I'm not going to waste time rewriting this in some other way.

sbourdeauducq, 2018-02-08:
Whatever the path of least resistance is. If GPIO bit-banging works around the problem, it's easier than tracking down the original problem which is likely a Vivado bug or something equally obscure.

marmeladapk, 2018-02-08:
@sbourdeauducq With your binaries I still get memory test error.

hartytp, 2018-02-14:
@sbourdeauducq Was going to have a look at synchronisation. Built the binaries with the latest ARTIQ, but still see

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017 M-Labs Limited

Bootloader CRC passed
Initializing SDRAM...
Write leveling: 25 62 57 74 52 61 14 26 done
Read delays: 7:00-113 6:00-112 5:20-39 4:27-43 3:59-76 2:46-62 1:86-103 0:96-112 done
SDRAM initialized
Memory test failed (571059/1114624 words incorrect)
Halting.
Traceback (most recent call last):

sbourdeauducq, 2018-02-14:
> I assume that none of this is an issue. How do you normally process Vivado's output to separate interesting warnings from the usual noise?

I don't have a solution for this nor the SDRAM problem. Among its many flaws, Vivado regularly spits out loads of spurious warnings, including when compiling Xilinx's own cores. The problem is not specific to Vivado; ISE and Altera/Intel Quartus also suffer from it.

Without SAWG, you still have JESD204, but only generating ramps.

sbourdeauducq, 2018-02-14:
Are the software-programmed delays the same for the same board when the SDRAM works and when it does not?

hartytp, 2018-02-14:
> Are the software-programmed delays the same for the same board when the SDRAM works and when it does not?

See my posts above.

With SAWG:
- Read delays: 7:00-113 6:00-112 5:20-39 4:27-43 3:59-76 2:46-62 1:86-103 0:96-112 done

Without SAWG:
- Read delays: 7:00-164 6:04-175 5:49-216 4:61-229 3:98-244 2:97-250 1:122-273 0:125-278 done

So, no, not the same.

> Without SAWG, you still have JESD204, but only generating ramps.
Okay, will look at that next

hartytp, 2018-02-14:
Rebuilt the gateware *with* SAWG and rebooted a few times looking at the read delays...

`Read delays: 7:00-71 6:00-78 5:01-32 4:05-22 3:31-47 2:31-51 1:48-64 0:59-80 done`
`Read delays: 7:00-70 6:00-84 5:04-20 4:11-43 3:29-45 2:35-51 1:47-67 0:64-83 done`
`Read delays: 7:00-73 6:00-76 5:04-20 4:06-23 3:24-41 2:33-51 1:40-56 0:50-67 done`
`Read delays: 7:00-52 6:00-66 5:04-20 4:04-20 3:30-48 2:33-52 1:43-60 0:44-60 done`

The previous time I built it (posted above) I got:
`Read delays: 7:00-113 6:00-112 5:20-39 4:27-43 3:59-76 2:46-62 1:86-103 0:96-112 done`

Not sure if that's interesting data.

sbourdeauducq, 2018-02-14:
The "write leveling" figures are also delays, and we don't know if it's the reads or writes that are failing.

hartytp, 2018-02-14:
Updated.

> The "write leveling" figures are also delays, and we don't know if it's the reads or writes that are failing.

Worth adding some more diagnostic info?

hartytp, 2018-02-14:
And, for comparison, rebuilt *without* SAWG and saw(g):

`Write leveling: 40 80 73 86 66 75 31 41 done`
`Read delays: 7:00-150 6:02-168 5:45-210 4:57-225 3:91-237 2:86-242 1:112-262 0:119-270 done`

The previous time I built it (posted above) I got:

`Write leveling: 42 81 77 89 71 78 35 45 done`
`Read delays: 7:00-164 6:04-175 5:49-216 4:61-229 3:98-244 2:97-250 1:122-273 0:125-278 `

Not sure if that's interesting data.

hartytp, 2018-02-14:
So, does seem to be a reasonable difference between the two.

jbqubit, 2018-02-14:
It's challenging to build a memory controller using Xilinx tools. Xilinx's own DDR3 core has a long list of bug (and fixes).  https://www.xilinx.com/support/answers/69036.html

One route that may resolve this Issue is to let Xilinx handle these sort of subtitles. @sbourdeauducq Have you tried using the Xilinx DDR3 IP? https://www.xilinx.com/products/intellectual-property/ddr3.html

sbourdeauducq, 2018-02-15:
> Have you tried turning off write-leveling?

Yes. I even went one step further and turned off the DDR3 completely: problem solved!

enjoy-digital, 2018-02-15:
I need to have a closer look at that. I'll try to look at that this evening.

hartytp, 2018-02-15:
The read/write delays should be the same for both gateware versions, right (w/wo SAWG)? As a temporary hack, would it make sense to just hard code the numbers to the values found without the SAWG?

sbourdeauducq, 2018-02-15:
> The read/write delays should be the same for both gateware versions, right (w/wo SAWG)?

In theory, yes.

> As a temporary hack, would it make sense to just hard code the numbers to the values found without the SAWG?

Can try, but it's a shot in the dark - the root cause is not necessarily a failure of the timing calibration program (I think it's even unlikely to be the problem).

hartytp, 2018-02-15:
Okay.

hartytp, 2018-02-17:
@sbourdeauducq what's the plan for this? I'm itching to try Sayma with the SAWG and AFAICT this is currently the only thing stopping me.

sbourdeauducq, 2018-02-17:
I don't know what is going on.
Did you try those binaries on your board? They work on mine:
https://github.com/m-labs/artiq/issues/908#issuecomment-364132304
Maybe try with the other DDR bank (the 32-bit one) or try using fewer bits on the main one.

sbourdeauducq, 2018-02-17:
It could be this: https://github.com/m-labs/misoc/issues/75

enjoy-digital, 2018-02-17:
@hartytp, @sbourdeauducq: i still need to have a look. You can do what @sbourdeauducq is suggesting.
I'm also wondering if we don't have a problem on write_leveling. I remembered during my tests that commenting write_leveling was making things work, you can eventually give a try until i have a look at that: https://github.com/m-labs/artiq/blob/7429ee4fb63316b05da07407d6802670ebdb80fd/artiq/firmware/libboard/sdram.rs#L248

sbourdeauducq, 2018-02-17:
We do seem to have problem with IOSERDES clocking though, which produces the warning and the error in the timing report, and seems to be a plausible explanation for those symptoms. I don't understand why, though: CLK is sys4x and CLKDIV is sys, which are generated by the same PLL and without phase offset. Or am I missing something?

enjoy-digital, 2018-02-17:
@sbourdeauducq: I think Vivado generates this warning since we are applying a constraint on clocks at the output of the PLL. Vivado is then no longer propagating constraints from the input of the PLL to the outputs (ie that sys4x and sys are generated from the same PLL) and generates this warning.

hartytp, 2018-02-17:
> Did you try those binaries on your board? They work on mine: #908 (comment)

Not yet, but happy to do that next week. Remind me what the difference between them and the ones I'm generating from the latest ARTIQ commit is?

Still, I'd rather not get into a situation where I'm reliant on asking for magic binaries to be able to do anything.

> Maybe try with the other DDR bank (the 32-bit one) or try using fewer bits on the main one (simply removing DQ/DQS/DM pins in Migen does it).

Do you have a patch to do that? I'm not up to speed on that part of the code.

In any case, we should try to get both banks working fully before moving on to Sayma v2.0 in case we decide that hardware changes are needed.

> We do seem to have a problem with IOSERDES clocking though, which produces the warning and the error in the timing report, and is a plausible explanation for those symptoms. I don't understand why, though: CLK is sys4x and CLKDIV is sys, which are generated by the same PLL and without phase offset. Or am I missing something?

Yes, fixing the timing warnings/errors sounds like a good starting point.

sbourdeauducq, 2018-02-17:
Confirmed that this is the source of the warning. Thanks.
https://github.com/m-labs/misoc/commit/d1489edfb8ebba57320fae9157c44aa0e1c1b783
There is a similar issue with serwb.

enjoy-digital, 2018-02-17:
@hartytp: if you want to use the 32 bits sdram, use `sdram="ddram_32"` here:
https://github.com/m-labs/misoc/blob/master/misoc/targets/sayma_amc.py#L119

hartytp, 2018-02-17:
@enjoy-digital thanks, I'll try that next week. I'd still like to see a proper fix of this at some point though.

hartytp, 2018-02-17:
Can we now program the RTM FPGA from the AMC via artiq_flash?

sbourdeauducq, 2018-02-17:
No, and this is unrelated to the SDRAM.

hartytp, 2018-02-17:
> No, and this is unrelated to the SDRAM.

Sorry, this isn't the right place to ask about this.

> Edit: Okay, I see why you're asking - it's because of the commit I reverted. It only seems to be "fixing" the problem on the HKG board, so it doesn't make much sense to keep that core disable

Right. Anyway, I'd lost track of how far you'd got on RTM programming, and wasn't sure how much that core did.

sbourdeauducq, 2018-02-17:
For what it's worth, SDRAM is now working on the HKG board with SAWG, using the current ARTIQ master (41adbef9) which includes the RTM loading core and the clock constraint fix in MiSoC. It used not to work with recent gateware when the SAWG and the loading core were both present. But since this is so random, we cannot conclude yet that the clock constraint fix solved it.

sbourdeauducq, 2018-02-17:
And, on the other hand, serwb is now broken. Again this doesn't make any sense, since Vivado is deriving itself the constraints I removed:
```
clk50                                                     {0.000 10.000}       20.000          50.000          
  standalone_pll_clk200                                   {0.000 2.500}        5.000           200.000         
  standalone_pll_eth_txclk                                {2.000 6.000}        8.000           125.000         
  standalone_pll_fb                                       {0.000 10.000}       20.000          50.000          
  standalone_pll_sys                                      {0.000 4.000}        8.000           125.000         
    serwb_pll_pll_fb                                      {0.000 8.000}        16.000          62.500          
    serwb_pll_pll_serwb_serdes_20x_clk                    {0.000 1.600}        3.200           312.500         
      serwb_pll_pll_serwb_serdes_20x_clk_INTERNAL_DIVCLK  {0.000 6.400}        12.800          78.125          
    serwb_pll_pll_serwb_serdes_5x_clk                     {0.000 6.400}        12.800          78.125          
    serwb_pll_pll_serwb_serdes_clk                        {0.000 32.000}       64.000          15.625          
  standalone_pll_sys4x                                    {0.000 1.000}        2.000           500.000         
    standalone_pll_sys4x_INTERNAL_DIVCLK                  {0.000 4.000}        8.000           125.000  
```

[275 more replies]

commit 2d4a1340 (2018-02-07): sayma_amc: remove RTM bitstream upload core. Closes #908
commit 287d5334 (2018-02-17): Revert "sayma_amc: remove RTM bitstream upload core. Closes #908"

This reverts commit 2d4a1340ea3a25dbaaa2200b7e86d2f42e7a19b5.
closed 2018-03-27 by sbourdeauducq