m-labs/artiq #910: vivado version check

jbqubit, 2018-01-29:
ARTIQ documentation does not specify supported version of Vivado. cf #908 Consider: 
- Add Vivado version ++ to [RELEASE_NOTES.rst](https://github.com/m-labs/artiq/blob/master/RELEASE_NOTES.rst) so end users know to upgrade. 
- Add Vivado version check (and warning banner) to artiq/gateware/targets build scripts.

sbourdeauducq, 2018-01-29:
So what should this say, exactly?

jbqubit, 2018-01-29:
Add Vivado version information to some file in ARTIQ codebase (like meta.yaml for conda). Scripts in artiq/gateware/targets halt if wrong version of vivado is used. A script flag permits build with outdated vivado.

jbqubit, 2018-01-29:
[Instructions](http://www.m-labs.hk/artiq/manual-master/developing.html#) for installing Xilinx build tool need more specificity. 

> Install the FPGA vendor tools (i.e. Vivado):

Download Xilinx __Vivado HLx 2017.4: WebPACK and Editions__. Building .bit for the FPGAs used in ARTIQ systems (7 Series, UltraScale) requires a license for Vivado HL Design Edition.

jbqubit, 2018-02-06:
Please just fix the docs and close. Somebody else can pay for the Vivado check.

whitequark, 2018-02-06:
> ARTIQ documentation does not specify supported version of Vivado. cf #908

It doesn't help to use 2017.4. I've built the bitstream using 2017.4 myself and I still had #908. So the utility of this check seems doubtful.

sbourdeauducq, 2018-02-06:
Yes, it's a rather complicated situation.

sbourdeauducq, 2018-02-08:
And the vivado version we use may change, to work around bugs etc.
So maintaining the docs etc. takes work, and is of limited benefit. You can look at the buildbot logs to determine the Vivado version used for each build.

closed 2018-02-08 by sbourdeauducq

jbqubit, 2018-02-14:
> And the vivado version we use may change, to work around bugs etc.

If you've positively identified a bug that is remedied by using a particular version of Vivado, users need to know about it. Else we encounter the same bug, right? @hartytp, @dhslichter

hartytp, 2018-02-14:
This one I'm not too fussed about. The problem with documenting this kind of thing is that the docs can easily go stale and do more harm than good. I'm pretty happy bugging people on IRC for this kind of thing.

dhslichter, 2018-02-14:
I think it would be sufficient for the documentation to point out that one should look at the buildbot logs to find out information about things like vivado versions, etc.  I agree with @hartytp that this sort of thing goes stale in a hurry.

jbqubit, 2018-02-14:
OK. Let's include a link in the documentation to the right [build bot](http://buildbot.m-labs.hk/builders/artiq-board). And instructions to choose a recent build and examine ```stdio``` for ```conda_build build artiq-kc705-nist_clock ```. The Vivado version used by M-Labs is reported as 

```
****** Vivado v2017.4 (64-bit)
  **** SW Build 2086221 on Fri Dec 15 20:54:30 MST 2017
  **** IP Build 2085800 on Fri Dec 15 22:25:07 MST 2017
```

commit 0ef33dd0 (2018-02-15): manual: add note about the "correct" vivado version

close #910