m-labs/artiq #911: building sayma_amc: no rule for libunwind-bare.a

jbqubit, 2018-01-29:
I built a new development machine this afternoon based on Ubuntu 16.04. It has Xilinx Vivado 2017.4. 

```
$ python artiq/artiq/gateware/targets/sayma_amc.py 
make: Entering directory '/home/britton/artiq-dev/artiq_sayma/software/bootloader'
 CARGO    cargo/or1k-unknown-none/debug/libbootloader.a
cargo: /home/britton/anaconda3/envs/artiq-dev/bin/../lib/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
cargo: /home/britton/anaconda3/envs/artiq-dev/bin/../lib/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
cargo: /home/britton/anaconda3/envs/artiq-dev/bin/../lib/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
cargo: /home/britton/anaconda3/envs/artiq-dev/bin/../lib/libcrypto.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
    Finished dev [unoptimized + debuginfo] target(s) in 0.0 secs
 LD       bootloader.elf
 OBJCOPY  bootloader.bin
 MSCIMG   bootloader.bin
make: Leaving directory '/home/britton/artiq-dev/artiq_sayma/software/bootloader'
make: Entering directory '/home/britton/artiq-dev/artiq_sayma/software/libunwind'
make: *** No rule to make target 'libunwind-bare.a', needed by 'all'.  Stop.
make: Leaving directory '/home/britton/artiq-dev/artiq_sayma/software/libunwind'
Command make -C /home/britton/artiq-dev/artiq_sayma/software/libunwind failed
```

```
$ git log 
commit 0edc34a9e5d3790a6692af33e16dff45924e2c5b
Author: whitequark <whitequark@whitequark.org>
Date:   Sun Jan 28 14:57:43 2018 +0000

    artiq_devtool: the proxy artiq_flash action doesn't exist anymore.
```

whitequark, 2018-01-29:
You need to update misoc.

jbqubit, 2018-01-31:
I did the following again this afternoon. Same error.

```
$ git clone --recursive https://www.github.com/m-labs/artiq
$ cd artiq
$ conda env create -f conda/artiq-dev.yaml
$ pip install -e .
$ cd /github/m-labs/jesd204b
$ python setup.py develop 
$ cd ~/artiq-dev
$ python artiq/artiq/gateware/targets/sayma_rtm.py
$ python artiq/artiq/gateware/targets/sayma_amc.py

make: Entering directory '/home/britton/artiq-dev/artiq_sayma/software/bootloader'
 CARGO    cargo/or1k-unknown-none/debug/libbootloader.a
cargo: /home/britton/anaconda3/envs/artiq-dev/bin/../lib/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
cargo: /home/britton/anaconda3/envs/artiq-dev/bin/../lib/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
cargo: /home/britton/anaconda3/envs/artiq-dev/bin/../lib/libssl.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
cargo: /home/britton/anaconda3/envs/artiq-dev/bin/../lib/libcrypto.so.1.0.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libcurl.so.4)
    Finished dev [unoptimized + debuginfo] target(s) in 0.0 secs
 LD       bootloader.elf
 OBJCOPY  bootloader.bin
 MSCIMG   bootloader.bin
make: Leaving directory '/home/britton/artiq-dev/artiq_sayma/software/bootloader'
make: Entering directory '/home/britton/artiq-dev/artiq_sayma/software/libunwind'
make: *** No rule to make target 'libunwind-bare.a', needed by 'all'.  Stop.
make: Leaving directory '/home/britton/artiq-dev/artiq_sayma/software/libunwind'
Command make -C /home/britton/artiq-dev/artiq_sayma/software/libunwind failed
```

Then when installing misoc... https://github.com/m-labs/artiq/issues/914

sbourdeauducq, 2018-02-01:
why not use conda?

jbqubit, 2018-02-02:
So I updated Misoc and the error is still there...

By using the artiq-dev.yaml I'm recreating the same build environment that your build bot uses. I create a new environment from scratch for every build. This has reduced the incidence of wrong-version dependency problems I've encountered in the last several months. Shouldn't this work?

sbourdeauducq, 2018-02-02:
Use the latest misoc from git (conda package probably available still, or you can trigger a build on Irc) we're not doing ci and full conda support for Sayma yet.

whitequark, 2018-02-02:
@jbqubit Did you run `git submodule update --init` in the misoc checkout?

jbqubit, 2018-02-02:
> we're not doing ci and full conda support for Sayma yet.

Is this something @whitequark can work on in parallel?

@whitequark I cloned with --recursive.

sbourdeauducq, 2018-02-02:
Try using the conda packages. https://anaconda.org/m-labs/misoc/files is up to date.

whitequark, 2018-02-02:
> @whitequark I cloned with --recursive.

Are there any files in `misoc/misoc/software/unwinder`?

sbourdeauducq, 2018-02-10:
@hartytp You also encountered this problem, did you resolve it with a correct recursive clone?

jordens, 2018-02-13:
AFAICT this is a hidden dependency of cargo on libcurl. See the output.

hartytp, 2018-02-14:
This is currently working fine for me.
- Cloned the latest ARTIQ using `git clone --recursive https://github.com/m-labs/artiq ~/scratch/artiq-sayma/artiq`
- `cd scratch/artiq-sayma/artiq`
- `conda env create -f conda/artiq-dev.yaml --name artiq-sayma`
- `. activate artiq-sayma`
- `conda uninstall jesd204b --force`
- `git clone --recursive https://github.com/m-labs/jesd204b ~/scratch/jesd204b`
- `export PYTHONPATH=$PYTHONPATH:/home/$USER/scratch/jesd204b`
- 'cd ~/scratch/artiq-sayma'
- `python -m artiq.gateware.targets.sayma_rtm'
- 'python -m artiq.gateware.targets.sayma_amc'

hartytp, 2018-02-14:
PS like the new diagnostics in artiq_flash.

closed 2018-02-14 by sbourdeauducq

whitequark, 2018-02-14:
> AFAICT this is a hidden dependency of cargo on libcurl. See the output.

No, libunwind is not written in Rust.

jordens, 2018-02-14:
It's still a bug. Maybe not the underlying one here.

jordens, 2018-03-08:
Anecdotal solutions are either `conda install libcurl` or checking for proper recursive git checkouts.