m-labs/artiq #913: artiq_flash exits with "Error: Unknown flash device (ID 0x00ffffff)"

jonaskeller, 2018-01-30:
Trying to install the nist_qc2 gateware 3.3 py_3+gitb92b00a1, I get the above error message (see below for full output), while trying with an older release (3.2 py_8+gitaa64b8ad) works and outputs this in the same place:

    Info : Found flash device 'micron n25q128' (ID 0x0018ba20)

I've also tried having the latest commit built, but that failed:
http://buildbot.m-labs.hk/builders/artiq-board/builds/1177

Full output of `artiq_flash -t kc705 -m nist_qc2`:

	Design name: b'top;UserID=0XFFFFFFFF;COMPRESS=TRUE;Version=2017.4\x00'
	Partname b'7k325tffg900\x00'
	Date b'2018/01/30\x00'
	Time b'13:16:35\x00'
	found binary data length: 5581604
	Open On-Chip Debugger 0.10.0-00010-g9f8c46a (2018-01-30-14:41)
	Licensed under GNU GPL v2
	For bug reports, read
			http://openocd.org/doc/doxygen/bugs.html
	none separate
	Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
	adapter speed: 25000 kHz
	Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
	Info : clock speed 25000 kHz
	Info : JTAG tap: xc7.tap tap/device found: 0x43651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x4)
	loaded file C:\Anaconda3\envs\artiq3.3\lib\site-packages\artiq\binaries\kc705-nist_qc2\bscan_spi_xc7k325t.bit to pld device 0 in 0s 610025us
	Info : JTAG tap: xc7.tap tap/device found: 0x43651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x4)
	Error: Unknown flash device (ID 0x00ffffff)

	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.3\Scripts\artiq_flash-script.py", line 5, in <module>
		exit(main())
	  File "C:\Anaconda3\envs\artiq3.3\lib\site-packages\artiq\frontend\artiq_flash.py", line 138, in main
		"-c", "; ".join(prog),
	  File "C:\Anaconda3\envs\artiq3.3\lib\subprocess.py", line 271, in check_call
		raise CalledProcessError(retcode, cmd)
	subprocess.CalledProcessError: Command '['openocd', '-s', 'C:\\Anaconda3\\envs\\artiq3.3\\Library\\share\\openocd\\scripts', '-f', 'board\\kc705.cfg', '-c', 'init; jtagspi_init 0 {C:\\Anaconda3\\envs\\artiq3.3\\lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\bscan_spi_xc7k325t.bit}; jtagspi_program {C:\\Users\\rabi\\AppData\\Local\\Temp\\1\\tmpvdu8vlvs} 0x0; jtagspi_program {C:\\Anaconda3\\envs\\artiq3.3\\lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\bios.bin} 0xaf0000; jtagspi_program {C:\\Anaconda3\\envs\\artiq3.3\\lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\runtime.fbi} 0xb40000; xc7_program xc7.tap; exit']' returned non-zero exit status 1

whitequark, 2018-01-30:
Do you have several boards plugged in?

jonaskeller, 2018-01-30:
No, just one.

whitequark, 2018-01-30:
@sbourdeauducq Any ideas? This was before any of my artiq_flash rewrite changes...

sbourdeauducq, 2018-01-31:
Any difference in openocd versions?

jordens, 2018-01-31:
You need the last commit.
@whitequark could you look at the list failure?

jordens, 2018-04-05:
Iirc you didn't have this problem before. That code hasn't changed over the last months. What's different? Is that a new board? Has it worked before? What's your bscan-spi-bitstreams version?

jordens, 2018-04-05:
And could you file a new issue? This one was specific to 3.x.

marmeladapk, 2018-04-05:
@jordens I'll do it after I verify for 100% that we don't have problems with this board.

marmeladapk, 2018-04-05:
After a random `artiq_flash (...) load` all subsequent flashes work (altough still can't start ARTIQ, silence on tty1).