m-labs/artiq #914: misoc asyncserial

jbqubit, 2018-01-31:
```
$ git clone --recursive https://github.com/m-labs/misoc
$ cd misoc
$ python setup.py develop
/home/britton/anaconda3/envs/artiq-dev/lib/python3.5/site-packages/setuptools/dist.py:333: UserWarning: Normalizing '0.6.dev' to '0.6.dev0'
  normalized_version,
running develop
running egg_info
writing top-level names to misoc.egg-info/top_level.txt
writing misoc.egg-info/PKG-INFO
writing dependency_links to misoc.egg-info/dependency_links.txt
writing entry points to misoc.egg-info/entry_points.txt
writing requirements to misoc.egg-info/requires.txt
reading manifest file 'misoc.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'misoc.egg-info/SOURCES.txt'
running build_ext
Creating /home/britton/anaconda3/envs/artiq-dev/lib/python3.5/site-packages/misoc.egg-link (link to .)
Removing misoc 0.6.dev0 from easy-install.pth file
Adding misoc 0.6.dev0 to easy-install.pth file
Installing mkmscimg script to /home/britton/anaconda3/envs/artiq-dev/bin
Installing flterm script to /home/britton/anaconda3/envs/artiq-dev/bin

Installed /home/britton/artiq-dev/misoc
Processing dependencies for misoc==0.6.dev0
Searching for asyncserial
Reading https://pypi.python.org/simple/asyncserial/
Couldn't find index page for 'asyncserial' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
No local packages or working download links found for asyncserial
error: Could not find suitable distribution for Requirement.parse('asyncserial')
```

```
$ conda info
Current conda install:

               platform : linux-64
          conda version : 4.3.30
       conda is private : False
      conda-env version : 4.3.30
    conda-build version : 3.0.27
         python version : 3.6.3.final.0
       requests version : 2.18.4
       root environment : /home/britton/anaconda3  (writable)
    default environment : /home/britton/anaconda3/envs/artiq-dev
       envs directories : /home/britton/anaconda3/envs
                          /home/britton/.conda/envs
          package cache : /home/britton/anaconda3/pkgs
                          /home/britton/.conda/pkgs
           channel URLs : http://conda.anaconda.org/m-labs/label/dev/linux-64
                          http://conda.anaconda.org/m-labs/label/dev/noarch
                          http://conda.anaconda.org/conda-forge/label/dev/linux-64
                          http://conda.anaconda.org/conda-forge/label/dev/noarch
                          https://repo.continuum.io/pkgs/main/linux-64
                          https://repo.continuum.io/pkgs/main/noarch
                          https://repo.continuum.io/pkgs/free/linux-64
                          https://repo.continuum.io/pkgs/free/noarch
                          https://repo.continuum.io/pkgs/r/linux-64
                          https://repo.continuum.io/pkgs/r/noarch
                          https://repo.continuum.io/pkgs/pro/linux-64
                          https://repo.continuum.io/pkgs/pro/noarch
            config file : /home/britton/.condarc
             netrc file : None
           offline mode : False
             user-agent : conda/4.3.30 requests/2.18.4 CPython/3.6.3 Linux/4.13.0-32-generic debian/stretch/sid glibc/2.23    
                UID:GID : 1000:1000
```

sbourdeauducq, 2018-02-01:
Yes, install asyncserial manually if you want to install misoc this way.

closed 2018-02-01 by sbourdeauducq

whitequark, 2018-02-01:
@sbourdeauducq So why wouldn't we publish asyncserial on pypi?..

sbourdeauducq, 2018-02-01:
If we put packages on pypi, we have to ensure that they are maintained and properly versioned there as well, otherwise there will be incorrect versions installed silently, and corresponding support issues. And pypi/setuptools don't have good ways of dealing with version dependencies, unlike e.g. cargo. So I think it is better to only support conda for packaging, which is enough of a PITA already.

whitequark, 2018-02-01:
We already have pythonparser on pypi.