m-labs/artiq #915: Artiq3.4: binutils-or1k-linux dependency issue on windows

jonaskeller, 2018-02-05:
Trying to install artiq 3.4 (py_0+git9db30ce8) on windows 7 64bit fails at the dependency

    binutils-or1k-linux >=2.27

The most recent windows binaries in the conda repository are v 2.25.1.

I could look into building it myself, but since you'll probably want it in your repository anyway, that doesn't seem worth the effort.

jordens, 2018-02-05:
@whitequark I tightened that dependency to match `artiq-dev`. Are we/should we be using different binutils?

whitequark, 2018-02-05:
@jordens I don't remember the exact diff between 2.27 and 2.25.1. I think 2.27 is just 2.25.1 plus our patches. The Windows build requires Cygwin and that isn't currently automated.

sbourdeauducq, 2018-02-06:
Why are the automatic tests not catching that?

whitequark, 2018-02-06:
Catching what?

sbourdeauducq, 2018-02-06:
The ARTIQ conda package not installing correctly on Windows, either on the release-3 branch or master; both specify binutils-or1k-linux >=2.27.

sbourdeauducq, 2018-02-06:
http://buildbot.m-labs.hk/builders/artiq/builds/2026 was for https://github.com/m-labs/artiq/commit/4c22d64ee438d8b65ba728829794698191719181.

jordens, 2018-02-06:
In that build build 520 was installed on windows while the parent build was 521.

jordens, 2018-02-07:
The windows test is not installing the artiq version that the build is running for http://buildbot.m-labs.hk/builders/artiq-win64-test/builds/718/steps/conda_create/logs/stdio
@whitequark what should we do? different binutils versions on windows (kernel build) and linux (firmware build)? ignore the version?

sbourdeauducq, 2018-02-07:
I believe the clean solution is a) fix the buildbot b) do the build of binutils-or1k-linux 2.27 that wasn't done for Windows.

sbourdeauducq, 2018-02-08:
> do the build of binutils-or1k-linux 2.27 that wasn't done for Windows.

Done.

closed 2018-02-08 by sbourdeauducq