m-labs/artiq #920: Error:  'device_db.py'can not be found

mingshenli, 2018-02-11:
with artiq3.5,run`artiq_compile idle.py` ,
 I can find  'device_db.py'   in  
 C:\ProgramData\Anaconda3\envs\artiq-3.5\Lib\site-packages\artiq\master,
and I move it to
 C:\ProgramData\Anaconda3\pkgs\python-3.6.3-h9e2ca53_1\Lib\idlelib
[image]
but
[image]
the same things happened when I run `artiq_run led.py`  In accordance with the instructions of the instructions on http://m-labs.hk/artiq/manual-release-3/getting_started_core.html

sbourdeauducq, 2018-02-11:
Those programs look for ``device_db.py`` in the current directory by default. Use the ``cd`` command to go to ``C:\ProgramData\Anaconda3\pkgs\python-3.6.3-h9e2ca53_1\Lib\idlelib`` (though you may want to put those in a more appropriate folder)

mingshenli, 2018-02-11:
the `device_db.py` has been copy to the folder where `idle.py` in(the first picture), from `C:\ProgramData\Anaconda3\envs\artiq-3.5\Lib\site-packages\artiq\master,`

sbourdeauducq, 2018-02-11:
https://en.wikipedia.org/wiki/Working_directory

sbourdeauducq, 2018-02-11:
Your current directory is ``c:\windows\system32``. It is looking for ``c:\windows\system32\device_db.py``, which doesn't exist.

mingshenli, 2018-02-11:
can moving the `device_db.py` to `c:\windows\system32\device_db.py` solve the problem?

sbourdeauducq, 2018-02-11:
``c:\windows\system32`` is not meant to hold the ARTIQ device database, use the ``cd`` command to change the current directory instead.

mingshenli, 2018-02-11:
ok, I will try it. thanks.

closed 2018-02-13 by mingshenli