m-labs/artiq #921: serial over IP for Sayma

jbqubit, 2018-02-14:
@whitequark How does IP over serial sounds as a near-term approach to interactive operation of Sayma? This was formerly done via PPP. Assume no support for Windows machines.

whitequark, 2018-02-14:
@jbqubit There's no problem with that at all, it just needs a small adjustment to smoltcp so that it supports other physical layers other than Ethernet. I would prefer to use either SLIP or a custom protocol on the host side. Windows is also doable using [these](https://community.openvpn.net/openvpn/wiki/ManagingWindowsTAPDrivers) drivers, if you want it.

jbqubit, 2018-02-14:
The aim is to get an interactive session working over USB between a Linux ARTIQ Master and Sayma as core device. Even if Ethernet is squared away it will be helpful to test Sayma and SAWG without the struggle of applying white wire patches to a bunch of v1 boards. 

SLIP is well supported by Linux. No need for something custom or Windows support.

hartytp, 2018-10-02:
AFAICT, this is no longer needed since ethernet on Sayma works fine now

closed 2018-10-02 by hartytp