m-labs/artiq #923: autogenerate board conda recipes

sbourdeauducq, 2018-02-21:
There is a lot of redundancy in the board conda recipes and this will only get worse as more boards/variants are added.

It woud be better to have one database file with entries containing only:
* package name
* package description
* target file and variant
* master/satellite (as the firmware is named differently)

Or the package building program could look into the target files to get this information.

sbourdeauducq, 2018-02-21:
Or maybe this: https://conda.io/docs/user-guide/tasks/build-packages/variants.html

sbourdeauducq, 2018-02-21:
And Sayma (with the RTM) will need special treatment.

jordens, 2018-02-21:
Conda variants won't do the trick AFACT since they are mutually exclusive in an environment.

commit 10c02afd (2018-02-22): conda: merge all board packages except sayma_rtm into one.

Fixes #923.
closed 2018-02-22 by whitequark