m-labs/artiq #925: integer inference issue when importing

jordens, 2018-02-21:
* Calling the function in the imported module alone errors.
* Calling the imported function alone works.
* Calling the imported function before the one in the imported module "heals" (1).

```
py35) rj@murray:~/src/artiq (master)$ artiq_run test_func_infer.py
test_func_infer.py:19:9-19:23: error: the type of this expression cannot be fully inferred
        test_mod.f2(1) + 1
        ^^^^^^^^^^^^^^
test_func_infer.py:19:9-19:23: note: the expression has type numpy.int?
        test_mod.f2(1) + 1
        ^^^^^^^^^^^^^^
```

`test_func_infer.py`:
```python
from artiq.experiment import *

import test_mod
from test_mod import f2

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.core.reset()

        # OK:
        # f2(1) + 1

        # not OK:
        test_mod.f2(1) + 1
```

`test_mod.py`:
```python
from artiq.experiment import *

@portable
def f2(x):
    return x & 1
```

whitequark, 2018-02-21:
I don't think that should go into 3.5. It's not a bug but an artefact of how imported names are represented (they're basically just locals), and it's not trivial to change.

jordens, 2018-02-21:
Ok. But can you explain (3) to me?

whitequark, 2018-02-21:
It's the same as (2). The compiler can see that the two functions are one and same, and types inferenced for the local name propagate to the non-local name.

jordens, 2018-02-21:
And the take home message is that the compiler is unable to infer stuff about attributes of modules?
Do we fix it for 4.0?

whitequark, 2018-02-21:
> And the take home message is that the compiler is unable to infer stuff about attributes of modules?

Only that global type inference interacts badly with the polymorphic integer literals, and these two need hacks to work together.

> Do we fix it for 4.0?

We should.

sbourdeauducq, 2018-03-21:
I suggest focusing 4.0 on supporting the new hardware - there are enough issues with it already and this compiler change doesn't seem simple. Let's move it to 5.0?