m-labs/artiq #929: artiq_flash: bail if scan chain is wrong

jordens, 2018-02-23:
Instead of continuing blindly, `artiq_flash` should have `openocd` error out if the JTAG scan chain is not as expected.

```

Design: top;COMPRESS=TRUE;UserID=0XFFFFFFFF;Version=2017.4
Part name: xcku040-ffva1156-1-c
Date: 2018/02/23
Time: 22:23:16
Bitstream payload length: 0x6d4708
Open On-Chip Debugger 0.10.0-00010-g9f8c46a (2018-01-30-14:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : JTAG tap: xcu.tap tap/device found: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Warn : JTAG tap: xcu.tap       UNEXPECTED: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Error: JTAG tap: xcu.tap  expected 1 of 1: 0x03822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x0)
Error: Trying to use configured scan chain anyway...
Error: xcu.tap: IR capture error; saw 0x3f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : gdb server disabled
RTM FPGA XADC:
TEMP -273.15 C
VCCINT 0.000 V
VCCAUX 0.000 V
VCCBRAM 0.000 V

...
```

whitequark, 2018-02-23:
I am not aware of a simple enough way to implement this. @jordens?

jordens, 2018-02-23:
I'll look into it.

jordens, 2018-02-23:
Didn't find anything so far. We might have to manually check the IDCODEs again after `init`...

closed 2022-07-26 by sbourdeauducq