m-labs/artiq #932: "Unknown flash device" when running artiq_flash

philipkent, 2018-02-23:
On Windows 7 64-bit, Artiq version 2.4.  When I try to run `artiq_flash -t kc705 -m nist_qc2` it fails with an "Unknown flash device" error:

    (artiq-2.4) c:\Users\pdk1>artiq_flash -t kc705 -m nist_qc2
    Design name: b'top;UserID=0XFFFFFFFF;COMPRESS=TRUE;Version=2016.2\x00'
    Partname b'7k325tffg900\x00'
    Date b'2017/04/23\x00'
    Time b'23:47:20\x00'
    found binary data length: 4749552
    Open On-Chip Debugger 0.10.0-00013-gbb7bedad (2018-02-17-04:39)
    Licensed under GNU GPL v2
    For bug reports, read
            http://openocd.org/doc/doxygen/bugs.html
    none separate
    Info : auto-selecting first available session transport "jtag". To override use 'transport select <trans
    port>'.
    adapter speed: 25000 kHz
    Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge
    falling"
    Info : clock speed 25000 kHz
    Info : JTAG tap: xc7.tap tap/device found: 0x43651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x4)
    loaded file C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-2.4\lib\site-packages\artiq\binar
    ies\kc705-nist_qc2\bscan_spi_xc7k325t.bit to pld device 0 in 0s 593804us
    Info : JTAG tap: xc7.tap tap/device found: 0x43651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x4)
    Error: Unknown flash device (ID 0x00ffffff)
    
    Traceback (most recent call last):
      File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-2.4\Scripts\artiq_flash-script.py", l
    ine 9, in <module>
        sys.exit(main())
      File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-2.4\lib\site-packages\artiq\frontend\
    artiq_flash.py", line 146, in main
        "-c", "; ".join(prog),
      File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-2.4\lib\subprocess.py", line 271, in
    check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['openocd', '-s', 'C:\\Users\\pdk1\\AppData\\Local\\Continuum\\a
    naconda3\\envs\\artiq-2.4\\Library\\share\\openocd\\scripts', '-f', 'board\\kc705.cfg', '-c', 'init; jta
    gspi_init 0 {C:\\Users\\pdk1\\AppData\\Local\\Continuum\\anaconda3\\envs\\artiq-2.4\\lib\\site-packages\
    \artiq\\binaries\\kc705-nist_qc2\\bscan_spi_xc7k325t.bit}; jtagspi_program {C:\\Users\\pdk1\\AppData\\Lo
    cal\\Temp\\1\\tmp1zclk_7j} 0x0; jtagspi_program {C:\\Users\\pdk1\\AppData\\Local\\Continuum\\anaconda3\\
    envs\\artiq-2.4\\lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\bios.bin} 0xaf0000; jtagspi_progra
    m {C:\\Users\\pdk1\\AppData\\Local\\Continuum\\anaconda3\\envs\\artiq-2.4\\lib\\site-packages\\artiq\\bi
    naries\\kc705-nist_qc2\\runtime.fbi} 0xb00000; xc7_program xc7.tap; exit']' returned non-zero exit statu
    s 1

jordens, 2018-02-24:
You'll need to make ually downgrade openocd to a version that was current when 2.4 was released

sbourdeauducq, 2018-02-24:
@jordens can you delete (or move to the dev channel) the buggy openocd versions? If course they are the ones that get installed as per Murphy's law...

jordens, 2018-02-24:
@sbourdeauducq It's not buggy openocd versions but incompatibility between the proxy bitstream shipped in the old board packages and the new openocd versions. Moving openocd won't help.
@whitequark suggested that the old versioning scheme was causing problems with conda so I changed it and I'm now using the build number to pin openocd. Before (and certainly not in 2.x) there was little constraint on the openocd version. Ergo the old releases won't work as long as any newer openocd gets installed automatically.

philipkent, 2018-02-27:
Downgrading openocd to version 0.10.0 build 1 fixed the unknown flash device error.  Running `artiq_flash -t kc705 -m nist_qc2` using artiq version 2.4 now complains that it cannot find the binaries directory though:

     (artiq-2.4) C:\Users\pdk1>artiq_flash -t kc705 -m nist_qc2
     Binaries directory 'C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-2.4\lib\site-packages\artiq\binaries\kc705-nist_qc2' does not exist

Is it possible to build the binaries and add them to the version 2.4 conda repository?  I'm going to try copying the binaries from the 2.3 environment to the 2.4 environment.

philipkent, 2018-02-27:
Copying the binaries directory over from 2.3 worked.  Are there any differences between the 2.3 binaries and artiq 2.4 that might cause problems doing it this way?

whitequark, 2018-02-27:
There are no gateware or firmware changes between 2.3 and 2.4, so no.

philipkent, 2018-02-27:
Trying to run an experiment using version 2.4 on windows 7 64 bit now fails with the error:

     or1k-linux-id.exe: The program can't start because cygwin1.dll is missing from your computer.  Try reinstalling the program to fix this problem

whitequark, 2018-02-27:
Working on it.

philipkent, 2018-02-28:
Thanks.  I made separate issue #935.  Just looks like or1k-linux-id needs to be pinned at version 2.25.1.

sbourdeauducq, 2018-03-07:
@philipkent Are there still flashing problems?

closed 2018-03-07 by sbourdeauducq