m-labs/artiq #933: DRTIO multilink broken on Ultrascale/Sayma

sbourdeauducq, 2018-02-27:
Patch sayma_amc as below:
```
diff --git a/artiq/gateware/targets/sayma_amc.py b/artiq/gateware/targets/sayma_amc.py
index 30060fe6a..70fe666a2 100755
--- a/artiq/gateware/targets/sayma_amc.py
+++ b/artiq/gateware/targets/sayma_amc.py
@@ -292,15 +292,18 @@ class Master(MiniSoC, AMPSoC):
         self.config["RTIO_FREQUENCY"] = str(rtio_clk_freq/1e6)
 
         self.comb += platform.request("sfp_tx_disable", 0).eq(0)
+        self.comb += platform.request("sfp_tx_disable", 1).eq(0)
         self.submodules.drtio_transceiver = gth_ultrascale.GTH(
             clock_pads=platform.request("si5324_clkout"),
-            data_pads=[platform.request("sfp", 0)],
+            data_pads=[platform.request("sfp", i) for i in range(2)],
             sys_clk_freq=self.clk_freq,
             rtio_clk_freq=rtio_clk_freq)
         self.csr_devices.append("drtio_transceiver")
 
         self.submodules.drtio0 = ClockDomainsRenamer({"rtio_rx": "rtio_rx0"})(
             DRTIOMaster(self.drtio_transceiver.channels[0]))
+        self.submodules.drtio1 = ClockDomainsRenamer({"rtio_rx": "rtio_rx1"})(
+            DRTIOMaster(self.drtio_transceiver.channels[1]))
         self.csr_devices.append("drtio0")
         self.add_wb_slave(self.mem_map["drtio_aux"], 0x800,
                           self.drtio0.aux_controller.bus)
```

DRTIO link 0 stops working (no link detected anywhere, no ClockAligner).

When commenting out those two lines:
```
       self.submodules.drtio1 = ClockDomainsRenamer({"rtio_rx": "rtio_rx1"})(
           DRTIOMaster(self.drtio_transceiver.channels[1]))
```
functionality of channel 0 is fully restored.

sbourdeauducq, 2018-02-27:
And when connecting the rest of the ``drtio1`` properly, the breakage persists.

sbourdeauducq, 2018-02-27:
> When commenting out those two lines:
> 
>        self.submodules.drtio1 = ClockDomainsRenamer({"rtio_rx": "rtio_rx1"})(
>            DRTIOMaster(self.drtio_transceiver.channels[1]))
> 
> functionality of channel 0 is fully restored.

Er, scratch that - it worked one time by chance, but it is still broken.

So, simply adding a second channel to the GTH breaks the first channel.

enjoy-digital, 2018-02-28:
https://github.com/m-labs/artiq/commit/2896dc619bf3aa350595d04f3e1841c47349fbb8 should fix that. (I tested with Master and Satellite with 2 data lanes)

sbourdeauducq, 2018-03-07:
Confirmed that enabling multilink no longer breaks channel 0.

sbourdeauducq, 2018-03-07:
The other channel (second SFP) is also OK, including the DRTIO protocol.

closed 2018-03-07 by sbourdeauducq