m-labs/artiq #935: Artiq 2.4: Experiment fails to run on Windows because cygwin1.dll can't be found

philipkent, 2018-02-28:
Running an experiment using version 2.4 on windows 7 64 bit from the conda repository fails with the following error:

      or1k-linux-id.exe: The program can't start because cygwin1.dll is missing from your computer.  Try reinstalling the program to fix this problem

The `cygwin1.dll` dll can't be found.

This can be fixed by downgrading the `binutils-or1k-linux` package to version 2.25.1-4 with:

    conda install --name <environment name> binutils-or1k-linux=2.25.1

whitequark, 2018-02-28:
No, it needs to be rebuilt at version 2.25.1-5 using msys instead of cygwin.

whitequark, 2018-02-28:
Also, it's or1k-linux-ld, not -id.

whitequark, 2018-02-28:
Package rebuilt.

whitequark, 2018-02-28:
No, actually, it doesn't work.

whitequark, 2018-02-28:
The package does work now.

sbourdeauducq, 2018-03-15:
No, it doesn't.

sbourdeauducq, 2018-03-15:
Still happens with binutils-or1k-linux 2.27 5.

whitequark, 2018-03-15:
No, it doesn't. I have provisioned a completely fresh Windows machine just now and installed only the binutils-or1k-linux package, and it doesn't crash because of `cygwin1.dll`:
[image]

closed 2018-03-15 by whitequark

whitequark, 2018-03-15:
And I can actually `artiq_compile` code with these binutils.

sbourdeauducq, 2018-03-16:
Well @mingshenli does have a Windows machine on which this package is installed, and ``ld`` fails due to missing cygwin1.dll. Interestingly, this only happens when the experiment is run by the master, ``artiq_run`` and ``artiq_compile`` are working correctly. I thought ``ld`` was now completely independent from cygwin?

sbourdeauducq, 2018-03-16:
Though that behavior may be due to the ``cygwin1.dll`` he put in the device db directory as a workaround.

sbourdeauducq, 2018-03-16:
@mingshenli To double-check: can you activate the artiq conda environment, go to ``d:\artiq1`` and run ``or1k-linux-ld --version``?

whitequark, 2018-03-16:
It is completely independent from cygwin. It is compiled as a MinGW application, which means it's windows native. What you're describing seems like the result of master being run from a different conda environment with older binutils package.

mingshenli, 2018-03-16:
yes, I find the or1k-linux-ld.exe and run it, but failed because " the cygwin1.ddl can not be found."

whitequark, 2018-03-16:
This must be an old conda package. If all else fails, you can remove your entire conda install and start over.

sbourdeauducq, 2018-03-16:
@whitequark same conda environment that makes artiq_run work.

whitequark, 2018-03-16:
Is there another miniconda installation in PATH?

mingshenli, 2018-03-16:
Now it seems work.

sbourdeauducq, 2018-03-16:
@mingshenli Still, what is the output of ``or1k-linux-ld --version``?

mingshenli, 2018-03-16:
nothing happened after I run `start C:\ProgramData\Anaconda3\envs\artiq-3.5\bin\or1k-linux-ld.exe `  except a cmd window flash.
I only have those on  my computer.
[image]

sbourdeauducq, 2018-03-16:
Please follow the instructions carefully:

> activate the artiq conda environment, go to d:\artiq1 and run or1k-linux-ld --version

Do this in an already open cmd window.

mingshenli, 2018-03-16:
do as  follow?

 `activate artiq-3.5`     
`cd D:\artiq1`
`or1k-linux-ld --version`

sbourdeauducq, 2018-03-16:
Yes.

mingshenli, 2018-03-16:
but it didn't work because 'or1k-linux-ld --version' is not a command or file

sbourdeauducq, 2018-03-16:
Strange, if you see this screenshot it works:
https://github.com/m-labs/artiq/issues/935#issuecomment-373520539

mingshenli, 2018-03-16:
is it have any influence on  running  `artiq_run` or other command uploading the python file to the kc-705 board?