m-labs/artiq #936: ProEM HS 512BX3 camera driver

whitequark, 2018-03-01:
Product page: http://www.princetoninstruments.com/products/ProEM-EMCCD
Driver download: ftp://ftp.princetoninstruments.com/Public/Software/Official/PICam
Test server: `ssh -J murray.ber.quartiq.de 10.0.18.3 -N`, camera at `10.32.4.139`

sbourdeauducq, 2018-03-02:
Note: due to the dependency on the proprietary driver, this probably should be out-of-tree (https://github.com/m-labs/artiq/issues/887).

whitequark, 2018-03-02:
See https://github.com/m-labs/picam.

jordens, 2018-03-24:
Lets try to figure out aravis. I suspect it will work fine. Will build and install the Debian package and then we can give the gobject stuff a spin.
https://ftp-master.debian.org/new/aravis_0.5.11-1.html

jordens, 2018-06-21:
https://github.com/quartiq/picam

closed 2018-06-21 by jordens