m-labs/artiq #938: Positive slack RTIO underflows with external clock

cjbe, 2018-03-02:
I see frequent RTIO underflows with a significant (>10us) *positive* slack reported, only when using the external clock. 

For example, the below experiment runs without error using the internal clock, but gives this error within a few seconds of running on the external clock:
```
Core Device Traceback (most recent call last):
  File "repo/test_underflow.py", line 17, in artiq_run_test_underflow.PulseRate.run(..., ...) (RA=+0x528)
    self.ttl.pulse(10*us)
  File "<artiq>/coredevice/ttl.py", line 91, in ... artiq.coredevice.ttl.TTLOut.pulse<artiq.coredevice.ttl.TTLOut>(...) (inlined)
    self.off()
  File "<artiq>/coredevice/ttl.py", line 71, in ... artiq.coredevice.ttl.TTLOut.off<artiq.coredevice.ttl.TTLOut>(...) (inlined)
    self.set_o(False)
  File "<artiq>/coredevice/ttl.py", line 47, in ... artiq.coredevice.ttl.TTLOut.set_o<artiq.coredevice.ttl.TTLOut>(...) (inlined)
    rtio_output(now_mu(), self.channel, 0, 1 if o else 0)
  File "<artiq>/firmware/ksupport/rtio.rs", line 46, column 13, in (Rust function)
    raise!("RTIOUnderflow",
    ^
artiq.coredevice.exceptions.RTIOUnderflow(0): RTIO underflow at 4360040696 mu, channel 4, slack 2158080 mu
Traceback (most recent call last):
  File "/home/ion/anaconda3/envs/artiq-cjb/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/ion/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/ion/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/ion/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "/home/ion/artiq-dev/artiq/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/ion/artiq-dev/artiq/artiq/coredevice/core.py", line 133, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/home/ion/artiq-dev/artiq/artiq/coredevice/comm_kernel.py", line 606, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/home/ion/artiq-dev/artiq/artiq/coredevice/comm_kernel.py", line 598, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.RTIOUnderflow: RTIO underflow at 4360040696 mu, channel 4, slack 2158080 mu
``` 
```python
class PulseRate(EnvExperiment):
    def build(self):
        self.setattr_device("core")

        self.ttl = self.get_device("ttl4")

    @kernel
    def run(self):
        self.core.reset()

        while True:
            delay(10*us)
            self.ttl.pulse(10*us)
```

I am switching from internal to external clock via the "external_clock" option to the core device in the device_db. 

My understanding is that the external clock on the Kasli is fed from the Si5324. As expected, on the internal clock RTIO events precess in phase relative to the Si5324 outputs on the MMCX, whereas on the external clock there is a clear phase relation between the RTIO events and the Si5324 clock - so there does seem to be a valid RTIO clock present after the switch.

I am using a Kasli with a (slightly modified) Opticlock gateware (Artiq 2f9d012).

cjbe, 2018-03-02:
I also see this issue using the Kasli DRTIO (master+satellite) gateware

jordens, 2018-03-06:
I can reproduce this with the current opticlock variant (right now, in that variant, `external_clock=True` means using the si5324 in free_run mode).

jordens, 2018-03-07:
a6d1b030c1f1697755a081f671c571d734a96187

```python
from artiq.experiment import *

class PulseRate(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.core.external_clock = True
        self.ttl = self.get_device("ttl4")

    @kernel
    def run(self):
        self.core.reset()
        # self.core.break_realtime()

        min_slack = 10000000
        for i in range(100000000):
            delay(1*us)
            slack = now_mu() - self.core.get_rtio_counter_mu()
            if slack < min_slack:
                min_slack = slack
            try:
                self.ttl.pulse(1*us)
            except:
                print(i)  # anywhere from a few 1e3 to a few 1e5
                print(slack)   # ~1.5e5 mu matches the error slack to within 500 mu
                print(min_slack)   # ~1.2e5 mu, somewhat smaller than current slack
                raise
```

closed 2018-03-07 by jordens