m-labs/artiq #940: Urukul AD9910 AUX_DAC mismatch on init

cjbe, 2018-03-02:
If a kernel ends abnormally (force terminated, RTIO underflow etc) it can leave the Urukul in a error state, such that a subsequent init() gives an "Urukul AD9910 AUX_DAC mismatch" error. A second init after this succeeds. 

I would expect the init to succeed regardless of the previous command history. I guess this may be due to the AD9910 SPI state machine being left in an odd state, and an IO_RESET may help.

jordens, 2018-03-02:
An IO_RST pulse is already done on `Urukul.init()`:

https://github.com/m-labs/artiq/blob/master/artiq/coredevice/urukul.py#L172

jordens, 2018-03-07:
@cjbe ping. What do you mean by `init()`?

cjbe, 2018-03-07:
The init() on the AD9910 class fails. My Urukul initialisation kernel does:
```python
self.urukul0_cpld.init()
self.urukul0_ch0.init()
self.urukul0_ch1.init()
self.urukul0_ch2.init()
self.urukul0_ch3.init()
```
The error is (sometimes) thrown in one of the AD9910 channel init() calls, if the previous kernel does not end gracefully.

cjbe, 2018-03-13:
With the below experiment (which runs for a while and then underflows) I see the error once every 5 or so runs. There is no startup or idle kernel, and there are no errors in the core log. I am using Artiq 2edf65f57b86fa9797d5f9cb300191cec31e50ba.

```python
class Hammertime(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_cpld")
        self.setattr_device("urukul0_ch0")

    @kernel
    def run(self):
        self.core.reset()

        self.urukul0_cpld.init()
        self.urukul0_ch0.init()
        self.urukul0_ch0.sw.on()

        self.core.break_realtime()
        while True:
            self.urukul0_ch0.set_mu(10000000)
            delay_mu(6000)
```

cjbe, 2018-03-20:
@jordens have you managed to reproduce this?

jordens, 2018-03-20:
Yes. I added a workaround. The MASTER_RESET pulse leads to the AD9910 not responding every other time that experiment is run (and errors). It is related to the activity after the cpld and dds init and it is also related to the underflow error. The IO_RST pulse is benign but also does not help. I don't understand yet why the AD9910 gets messed up if and only if there is the master reset.

closed 2018-03-20 by jordens

sbourdeauducq, 2018-03-20:
The AD9914 on the NIST hardware also gets messed up if you reset it (with the hardware reset pin). Related?

jordens, 2018-03-20:
Don't know.